MarlinFirmware/Marlin #25528: compile error multiple definition of `Wire'

GuttalBug, 2023-03-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to use soft_i2c_eprom, I use the following declaration:
#if NO_EEPROM_SELECTED
 //#define EEPROM_SETTINGS
 #define I2C_EEPROM
 #define SOFT_I2C_EEPROM
 #undef I2C_SDA_PIN
 #define I2C_SDA_PIN                           70
 #undef I2C_SCL_PIN
 #define I2C_SCL_PIN                           71
 #define MARLIN_EEPROM_SIZE                    0x7FFF  // 32K (25LC256)
 #undef NO_EEPROM_SELECTED
#endif
but wen I compile with Visual Studio Code I get the duplicate "Wire" definition error. If I remove the #define I2C_EEPROM statment I can compile but Marlin use eeprom emulation.
[Marlin configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10993773/Marlin.configuration.zip)

### Version of Marlin Firmware

02010200

### Printer model

custom corexy

### Electronics

board with arduino due

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-03-17:
Since BOARD_CABER is totally custom,we have no idea what platform this is on and absolutely now way to even start looking at this 

You need to provide sufficient information so we can replicate the issue. 

No Boards pin.h file 
No build environment details
Not even a error log.

GuttalBug, 2023-03-18:
Thanks for the answer, I include the the pin definition for BOARD_CABER. 
[pins_CABER.zip](https://github.com/MarlinFirmware/Marlin/files/11007888/pins_CABER.zip)
The solution that I have found is to modify eeprom_if_i2c.cpp renaming "wire" to "wiregg".
[eeprom_if_i2c.zip](https://github.com/MarlinFirmware/Marlin/files/11007897/eeprom_if_i2c.zip)
The solution work perfectly but I don't know if there is a more elegant solution

GuttalBug, 2023-03-18:
sorry I closed for mistake

selahattinerdal, 2023-03-29:
Same problem. I have changed in eeprom_if_i2c.cpp

#if ENABLED(SOFT_I2C_EEPROM)
  #include <SlowSoftWire.h>
  SlowSoftWire Wire = SlowSoftWire(I2C_SDA_PIN, I2C_SCL_PIN, true);
#else
  #include <Wire.h>
#endif

to

  #include <Wire.h>

and its worked.

GuttalBug, 2023-04-02:
with your solution you still use hardware I2c and not the SlowSoftWire that alluw to use any I/O pin

thinkyhead, 2023-04-03:
The pins files define (or don't define) `SOFT_I2C_EEPROM`. So the simpler solution is to not define it in your pins file.

thisiskeithb, 2023-04-27:
This was probably fixed in https://github.com/MarlinFirmware/Marlin/pull/25707.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

GuttalBug, 2023-05-02:
I test with the last build and the problem is fixed. Thanks to all

closed 2023-05-02 by GuttalBug