MarlinFirmware/Marlin #25539: [BUG/OVERSIGHT] (Park disabled for switching nozzle +  editable angles )

studiodyne, 2023-03-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With switching nozzle :
- Park is disabled : Why ? Because of switching nozzle, we don't need to purge in an area or bucket ? Switching nozzle is just to avoid scratching the object and not a new miraculous machine that don't need to be primed as others
- Park is included in toolchange process, then make priming/swapping/purging , do you think switching nozzles must be ejected of this process ?
- Don't forget to lower the nozzle when parked , no lowering code in park code
- Editable angles not avalaible for switching nozzle ' 2 servos ', this settings is important for 3dprinter builder or optimisation of servos positionning. We can't upload each time for 1 degrees more !

Not bugs , but not serious and for all machines ' lulzbolt/raise3d' or any with dual servo switching , this new release is not functionnal, and must be tweaked

Suggestions : ( to simplify your job , i have one dual servo printer and I use it )

-editable angles
`  inline void _move_nozzle_servo(const uint8_t e, const uint8_t angle_index) {
      constexpr int8_t  sns_index[2] = { SWITCHING_NOZZLE_SERVO_NR, SWITCHING_NOZZLE_E1_SERVO_NR };
      /*constexpr int16_t sns_angles[2] = SWITCHING_NOZZLE_SERVO_ANGLES;*/
      #if ENABLED(EDITABLE_SERVO_ANGLES)
        const uint16_t sns_angles[2] = {servo_angles[SWITCHING_NOZZLE_SERVO_NR][0], servo_angles[SWITCHING_NOZZLE_SERVO_NR][1]};
      #else
        constexpr int16_t sns_angles[2] = SWITCHING_NOZZLE_SERVO_ANGLES;
      #endif`

And for park , you need to allow preprocessing in toolchange.cpp + lower the nozzle

`    #if HAS_SWITCHING_NOZZLE
        // Move back down. (Including when the new tool is higher.)
        if (!should_move)
          do_blocking_move_to_z(destination.z, planner.settings.max_feedrate_mm_s[Z_AXIS]);
        TERN_(SWITCHING_NOZZLE_TWO_SERVOS, lower_nozzle(new_tool));`

Please fix this

### Version of Marlin Firmware

bf2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

sjasonsmith, 2023-04-01:
I'm not sure what is going on with this bug report, but if this is just a list being constantly modified please do not do that. Each issue should be tracked in a separate bug report so they can be fixed and closed out individually.

I'm going to close this, but feel free to file individual reports for one problem at a time. If you already have solutions for them you can go ahead and post pull requests for those.

closed 2023-04-01 by sjasonsmith

studiodyne, 2023-04-01:
Sorry.
Can you tell me why in a rumba32  , with zmax PA4 ' output for coolant_mist ' + PD10 'free gpio' for coolant_flood output , when in insert SDCARD , it makes an M7 ' and PA4 turns ON.
I have any spare board , i can make a test kit with lcd + motherboard , but , you have experience of this board , and i ask you
Thks