MarlinFirmware/Marlin #25543: M73 Progress Plugin in Ocotoprint bargraph not displaying

ProtomakerSprint, 2023-03-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

M73 Progress Plugin in Ocotoprint bargraph not displaying it was showing the bargraph on a previous version of Marlin

Marlin-bugfix-2.1.2

#define CONFIGURATION_ADV_H_VERSION 02010300

OctoPrint Plugin M73 Progress  OctoPrint-M73Progress
https://github.com/cesarvandevelde/OctoPrint-M73Progress

in Configuration_adv.h

// Add the G-code 'M73' to set / report the current job progress
#define SET_REMAINING_TIME true  // Protomaker Sprint SET_PROGRESS_PERCENT, SET_REMAINING_TIME, SET_INTERACTION_TIME to be enabled."

#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
  //#define SET_PROGRESS_PERCENT          // Add 'P' parameter to set percentage done, otherwise use Marlin's estimate
  //#define SET_REMAINING_TIME            // Add 'R' parameter to set remaining time, otherwise use Marlin's estimate
  //#define SET_INTERACTION_TIME          // Add 'C' parameter to set time until next filament change or other user interaction
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  //#define M73_REPORT                    // Report progress to host with 'M73'
#endif

### Bug Timeline

Since : Marlin-bugfix-2.1.2

### Expected behavior

I expected the Bargraph to be displayed

### Actual behavior

No bargraph to is displayed

### Steps to Reproduce

Compile Marlin-bugfix-2.1.2 with only changes required to compile it

Flash to 3D Printer Creality Ender 5 Plus with Octopus v1.1 control-board
RE: Bigtree Tech Octopus V1.1 STM32F446ZET6 chip

Display Bigtree Tech TF35 Dual display used in LCD12864 display option

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11014199/Configurations.zip)

### Version of Marlin Firmware

Marlin-bugfix-2.1.2

### Printer model

Creality Ender 5 Plus

### Electronics

igtree Tech Octopus V1.1 STM32F446ZET6 chip

### Add-ons

Bigtree Tech TF35 Dual display, BLTouch, POWER_LOSS_RECOVERY, Servo Nozzle wiper arm 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-03-20:
`#define SET_REMAINING_TIME true`
is not the same as 
`#define SET_REMAINING_TIME`

you say "SET_INTERACTION_TIME to be enabled" but your config has thisi disabled...

closed 2023-03-20 by ellensp

ProtomakerSprint, 2023-03-21:
Doing a 1 day plus print at this time; I will try this later **THANKING YOU 2000%**

ProtomakerSprint, 2023-03-22:
A Shot Video of the M73 Bargraph working
https://youtube.com/shorts/A1-cIO4NQMg?feature=share

ProtomakerSprint, 2023-03-24:
It stopped workig so I change the code, still to ne tested to:-
// Add the G-code 'M73' to set / report the current job progress
// M73 Progress Bar graph working now
// A Massive THANK YOU to [ellensp](https://github.com/ellensp) 
#define SET_REMAINING_TIME   // Protomaker Sprint SET_PROGRESS_PERCENT, SET_REMAINING_TIME, SET_INTERACTION_TIME to be enabled.". I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543

#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
  #define SET_PROGRESS_PERCENT            //  Protomaker Sprint Add 'P' parameter to set percentage done, otherwise use Marlin's estimate.Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #define SET_REMAINING_TIME              //  Protomaker Sprint Add 'R' parameter to set remaining time, otherwise use Marlin's estimate. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #define SET_INTERACTION_TIME            // Add 'C' parameter to set time until next filament change or other user interaction. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  #define M73_REPORT                    // Report progress to host with 'M73'  Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
#endif

ProtomakerSprint, 2023-03-25:
What happened M73 displayed for one 3D print then with no change to firmware then it stopped displaying?

**_[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11069015/Configurations.zip)_**:=
[image]

// Add the G-code 'M73' to set / report the current job progress
// M73 Progress Bar graph working now
// A Massive THANK YOU to [ellensp](https://github.com/ellensp) 
#define SET_REMAINING_TIME   // Protomaker Sprint SET_PROGRESS_PERCENT, SET_REMAINING_TIME, SET_INTERACTION_TIME to be enabled.". I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543

#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
  #define SET_PROGRESS_PERCENT            //  Protomaker Sprint Add 'P' parameter to set percentage done, otherwise use Marlin's estimate.Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #define SET_REMAINING_TIME              //  Protomaker Sprint Add 'R' parameter to set remaining time, otherwise use Marlin's estimate. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #define SET_INTERACTION_TIME            // Add 'C' parameter to set time until next filament change or other user interaction. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  #define M73_REPORT                    // Report progress to host with 'M73'  Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
#endif

ellensp, 2023-03-25:
These only displays what you put in gcode 
If the gcodes does not contain the correct code, nothing is displayed

ProtomakerSprint, 2023-03-25:
I've ONLY just checked an OctoPrint does **_NOT_** now have the M73 Progress plugin;

_This is Very very, very odd as **I know** it was added before._

I will re-add he M73 Progress plugin after the 3D print I am doing now :0

ProtomakerSprint, 2023-03-26:
I just re-add the M73 Progress plugin and it is functioning as expedited.

Paragraph, R, R times and Percentage are now displayed.

Odd that the Plugin seemed to vanish. About two weeks ago the "Fullscreen Plugin" also seemed to vanish and I had to reinstall it;

Other Plugins where still in use, very odd to me.

I have confirm uninstall turned on for plugins.

Very odd to me>

Here is the code section as it is now within _Configuration_adv.h_  :-

// Add the G-code 'M73' to set / report the current job progress
// M73 Progress Bar graph working now
// A Massive THANK YOU to [ellensp](https://github.com/ellensp) 
#define SET_REMAINING_TIME   // Protomaker Sprint SET_PROGRESS_PERCENT, SET_REMAINING_TIME, SET_INTERACTION_TIME to be enabled.". I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543

#define SET_PROGRESS_MANUALLY  // Protomaker Sprint Show Progress using OctoPrint and M73 Plugin
#if ENABLED(SET_PROGRESS_MANUALLY)
  #define SET_PROGRESS_PERCENT            //  Protomaker Sprint Add 'P' parameter to set percentage done, otherwise use Marlin's estimate. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #define SET_REMAINING_TIME              //  Protomaker Sprint Add 'R' parameter to set remaining time, otherwise use Marlin's estimate. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
 // #define SET_INTERACTION_TIME            // Add 'C' parameter to set time until next filament change or other user interaction. Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  #define M73_REPORT                    // Report progress to host with 'M73'  Protomaker Sprint I had Bar Graph not displaying https://github.com/MarlinFirmware/Marlin/issues/25543
#endif

**THANKING YOU**