MarlinFirmware/Marlin #25545: [BUG] Filament runout reported at print start

livius2, 2023-03-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.1.2

Filament runout reported on print start even if M119 raport open.
I have checked M119 if it changed when it is out or in and all is reported ok.
I even build marlin with commented out `//#define FILAMENT_RUNOUT_SENSOR`
and it still report filament runout on print start...

I have confirmed that `//#define FILAMENT_RUNOUT_SENSOR` is working as now M119 do not show me Filament sensor status.

It is for me evident marlin bug, not the hardware as i have disabled sensor at all in the build process.

### Expected behavior

When filament is in and M119 report open it should see filament is in.

### Actual behavior

No matter of M119 report open or triggered always reported filament runnout.

### Version of Marlin Firmware

TFT screen

### Printer model

custom

### Electronics

SKR 3 EZ

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/11017333/config.zip)

ellensp, 2023-03-20:
Some TFT displays have their own (independent of marlin) runout sensors

eg the btt tft https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini#L714-L739

livius2, 2023-03-20:
@ellensp this was tip of a day. I didn't know that there are 2 different independent/conflict places of filament sensor software, Marlin and  Bigtreetech. Thank you for so fast response.

closed 2023-03-20 by thisiskeithb