MarlinFirmware/Marlin #25547: [BUG] TronXY Chitu V6 F103 board incorrect Z_STOP_PIN

lukasradek, 2023-03-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bed leveling probe does not work as Z_MIN switch.
There is a wrong pin assigned to Z_STOP_PIN in `pins_CHITU3D_V6.h`.

The change came with [this PR](https://github.com/MarlinFirmware/Marlin/pull/25227).
Before that, the value from `pins_CHITU3D_common.h` was used. (pin `PG9`)
After the change the value is assigned in `pins_CHITU3D_V6.h` and is different (pin `PA14`).

I have a late 2020 TronXY X5SA-500 Pro with Chitu V6 F103.

The PR says it fixed the issue on some other board, but it broke mine and I actually cannot find any other configuration file with the correct pin value... therefore I assume that there are either more board variants than is currently supported or the PR fix was incorrect and he should have used a different board (both V5 and V9 also use the `PA14` pin that the PR introduced).

It works fine by changing the pin back to `PG9`.

### Bug Timeline

It started with this pull request https://github.com/MarlinFirmware/Marlin/pull/25227

### Expected behavior

Z probe supplies values to the firmware.

### Actual behavior

Z probe is not recognized because of incorrect pin.
M119 reports it as `open` even though the sensor is triggered.
The sensor itself is working.

### Steps to Reproduce

1. Compile Marlin with Chitu V6 motherboard and f103 env.
2. Flash firmware to TronXY CXY-V6 board (or at least to the version that I have).

### Version of Marlin Firmware

2.1.2

### Printer model

Tronxy X5SA-500 Pro

### Electronics

full stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I am not uploading the conf files because they are not relevant.
Feel free to ask for it.

ellensp, 2023-03-22:
Tronxy have a bad habit of using pins for functions that different from the boards labeled functions.

From the information I can see the Z endstop pin is PA14 as listed in pins_CHITU3D_V6.h

PG9 is listed as the X+ endstop  (which doesn't seem to actually exist as a plug) 

See circuit diagram https://github.com/MarlinFirmware/Marlin/files/3401484/x5sa-main_board-2.pdf

I would firstly check, where is your z stop actually plugged into on the board, what port?

lukasradek, 2023-03-22:
> PG9 is listed as the X+ endstop (which doesn't seem to actually exist as a plug, is probably part of the large connector)

I think this might be a part of the problem. X5SA are CoreXY machines so the probe wiring goes through the printhead chain to the secondary board, where they are just grouped into the 40-pin connector.

Meaning you have no option to plug it differently at the main board and you possible can plug it differently at the secondary board... I will check it later today, but I don't think there will be much more other ports.

lukasradek, 2023-03-22:
@ellensp 

A took a look at the secondary board and there were (among others) ports labeled `xstop `and `level` with correct devices plugged into them. There were no unoccupied ports.

So from the user interface perspective, my printer is ok.

Can @AdUser chime in with more info about the PR that changed the pin? 🙂

lukasradek, 2023-03-23:
No, no, no... misunderstanding.

There is the mainboard. Similar to this, but older model. [image]

Because quite a few elements (e-motor, filament sensor, bed level sensor,...) are on the far side of the printer (from the mainboard), there is the 30-pin connector that groups all those into one flat cable. And I am not sure if the 30-pin connector is labeled on the mainboard pin by pin... but even if it was, it would be quite tedious to find out which individual wire goes where.

This 30-pin cable goes to the secondary board... which is technically a board but function-wise it is an adapter from 30-pin to individual ports. There is no remapping (at least by my definition).
[image]
This adapter board is labeled and in my case plugs are in the correct ports.

ellensp, 2023-03-23:
So no z-stop on that board. so it is not possible to be using z-stop for a probe attached to this board.

to me the config your using is simply wrong

You should not be using Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN as you are not using z-min pin
You should be using #define USE_PROBE_FOR_Z_HOMING with #define Z_MIN_PROBE_PIN PG9

Adding #define Z_MIN_PROBE_PIN PG9 to pins_CHITU3D_V6.h is also an option

lukasradek, 2023-03-23:
I based my config on the official Marlin Configuration repository, where the Tronxy X5SA series has its own directory and Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is defined there, which kind of made sense since the printer has no dedicated Z-STOP switch.

In either case, I think it should be more plug and play for the user in terms of pins.

It is more than likely that user will base their config on the defaults from Marlin Config repository.
He will probably leave the _pin_ related stuff as is since that is rather printer model dependent (which is why config repo exists) than user preference.
Even if he knew that the probe is actually not plugged into Z-STOP port (assuming he correctly deduces that it is a PIN problem), he would have to know the pin number of Z_MIN_PROBE_PIN.
That is a lot of hurdles I would say.

In this regard I would suggest some changes either to the config repo or to the pin definition files, but since I have seen only one CHITU board in my life, I am not the person to do it.
As I mentioned before, I think it would help to know the original intention of the PR.

lukasradek, 2023-03-23:
If it helps, my board is CXY-V6 190319.

ellensp, 2023-03-23:
I just fund this https://3dtoday.ru/blogs/artem-sr/tronxy-x5sa-pro-ustanovka-bfp-touch-na-board-chitu3d-v6-cxy-v6-191017

It documents all the pins on that 30 pin cable/board

It says that PG9 is Proxivity pin... ie it should be defined 

So based on this

pins_CHITU3D_V6 shoudl have
`#define Z_MIN_PROBE_PIN PG9
`
and the Configuration.h file should have 

/```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
//#define Z_MIN_PROBE_PIN {what ever number here}

```

Please test these changes work as expected.

ellensp, 2023-03-23:
There is no Marlin provided Configuration example for a TronXY X5SA-500 Pro

So I don;t know what Example config your are referring to

The provided Example Config for a X5SA is a different controller, and as such has different requirements.

lukasradek, 2023-03-23:
I viewed X5SA(-xxx) as a family of printers. Since they are all the same layout and it actually worked great.

thisiskeithb, 2023-03-27:
https://github.com/MarlinFirmware/Marlin/pull/25552 has been merged.

closed 2023-03-27 by thisiskeithb