MarlinFirmware/Marlin #25549: [BUG] (Failed to install 'marlinfirmware.auto-build'.)

Hueristic, 2023-03-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Failed to install 'marlinfirmware.auto-build'.

### Steps to Reproduce

Failed to install 'marlinfirmware.auto-build'.
Download manually
Install from vsix
MarlinFirmware.auto-build-2.1.45.vsix install failes with error
"message	"Request was blocked due to exceeding usage of resource 'Count' in namespace 'AnonymousId'. For more information on why your request was blocked, see the topic \"Rate limits\" on the Microsoft Web site (https://go.microsoft.com/fwlink/?LinkId=823950)."
typeName	"Microsoft.TeamFoundation.Framework.Server.RequestBlockedException, Microsoft.TeamFoundation.Framework.Server"
typeKey	"RequestBlockedException"
errorCode	0
eventId	3000"
https://marketplace.visualstudio.com/_apis/public/gallery/publishers/MarlinFirmware/vsextensions/auto-build/2.1.45/vspackage

### Version of Marlin Firmware

2.1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Love everything depending on M$ to work, really makes life easy doesn't it?

ellensp, 2023-03-22:
Marlin uses platformio under the vscode editor. this is not visual studio

https://marlinfw.org/docs/basics/install_platformio_vscode.html

closed 2023-03-22 by ellensp

Hueristic, 2023-03-22:
Found the issue, apparently M$ likes to hide errors. Updated OP

"MarlinFirmware.auto-build-2.1.45.vsix install failes with error "Unable to install Extension 'marlinfirmware.auto-build-2.1.45' as it is not compatible with Code '1.25.1'."
"

Fix: updated to VSCode 1.70.2 as apparent;y M$ doesn't allow 1.70.3 downloads yet states it is last version supported by OS.

So why no binaries for download?

I'm going to take a guess M$ pulled a fast one, ergo forcing VScode use?