MarlinFirmware/Marlin #25553: [BUG] Crash during Print, LA, IS, Bugfix.2.1x

jesterhead82, 2023-03-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Maybe related to PR 25541?

Printer crashes during print @2.84mm Layer, tried on 2 different sd cards
Prints fine if I disable LA, ~50/100mm Print speeds, IS @ ~43, LA 0.02, ESTEPS ~700

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11053618/Configuration.zip)
[GCode.zip](https://github.com/MarlinFirmware/Marlin/files/11053629/GCode.zip)

### Steps to Reproduce

[](url)

### Version of Marlin Firmware

Bugfix 2.1.x as of this week

### Printer model

Custom Corexy

### Electronics

BTT 1.4T

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ThomasToka, 2023-03-23:
I can confirm that last commits are somehow faulty. I poke around today around my printer stopping at my usual cube at around 40%.

My last working build has this as last commit: https://github.com/MarlinFirmware/Marlin/commit/10983d0cfe3e353ba465cfcadf8aeb51327f1197

Will disable LA now to check if it works without LA with my recent includes of the latest commits.

ThomasToka, 2023-03-23:
Reverting this one fixes it https://github.com/MarlinFirmware/Marlin/commit/ca77850cbb8ed57d4cdcf29a12a278d6bfa5c0d5#

I have not fully understood it but maybe this missing else here is it: 

https://github.com/MarlinFirmware/Marlin/pull/25541/commits/e8b0f38fef0909dd05cecb6655026f9f83765f47

thisiskeithb, 2023-03-23:
@tombrazier Can you take a look at this?

tombrazier, 2023-03-23:
@jesterhead82 @ThomasToka Can you describe exactly what your printer does at the point where the print fails? My guess is movement freezes and then after a few seconds the firmware reboots. But if it is something other than that please describe the behaviour. Also: does the E stepper have any noticeable behaviour when the crash occurs?

tombrazier, 2023-03-23:
> I have not fully understood it but maybe this missing else here is it:

@ThomasToka What mainboard do you have? I think @jesterhead82 has a 32 bit MCU, in which case that code is not compiled.

ThomasToka, 2023-03-23:
print stops and printer reboots. always at the same position. cube at around 40%.

i have creality f4 board on a ender 3 s1 pro. removing the commit fixes it.

ThomasToka, 2023-03-23:
nothing else hearable.

jesterhead82, 2023-03-23:
> > I have not fully understood it but maybe this missing else here is it:
> 
> @ThomasToka What mainboard do you have? I think @jesterhead82 has a 32 bit MCU, in which case that code is not compiled.

Board has a LPC1769, so yes.

As for the stepper, there is some "eechy" noise shortly before the reboot, i guess some motor is doing something stupid. But it could be the board/display maybe, could try again tomorrow is that would be helpfull..

tombrazier, 2023-03-23:
It could just possibly be a division by zero caused by the changes to all the calls to `calc_timer_interval()` in ca77850. Could one or both of you try changing them back?

ThomasToka, 2023-03-23:
yes as i said if i revert this commit it works.

PendulumPit, 2023-03-24:
I setup the latest bugfix the other day to do a speed test on a benchy  and got through that no issue. After I turned on LA as usual and printer would stop and reboot. Tried again and same.  I was going to mention this the other day but I got busy with work.
Was trying to beat Tom's 13 minute benchy lol I failed from a layer shift :)

thisiskeithb, 2023-03-24:
> As for the stepper, there is some "eechy" noise shortly before the reboot

My Prusa Bear with a BTT002 (STM32F4) has been printing all day with `bugfix-2.1.x` (724ba4b) without issue, but the Biqu Hurakan with Manta M4P (STM32G0) freezes & reboots in the same spot consistently with a similar "eechy" symptom after ~4-5 layers. Both are running Linear Advance & Input Shaping is disabled.

After reverting ca77850, the reboots stop.

tombrazier, 2023-03-24:
Could someone test #25557?

ThomasToka, 2023-03-24:
printing. result will follow in 20 min.

ThomasToka, 2023-03-24:
Confirmed. Works again.

ThomasToka, 2023-03-24:
Trying now a print with the follow up https://github.com/MarlinFirmware/Marlin/pull/25557/commits/a3c4f25a7a850c6cca40136a4f8201b4353e8939 .

result in some minutes..

tombrazier, 2023-03-24:
Thanks. So it appears that it's not a division by zero. I would really like to understand what is going on before proceeding.

I don't have time now to try and narrow it down but if anyone wants to play, it would be good to know exactly which of the four calls to `calc_timer_interval()` is the one that is causing the error. And if anyone is able to insert a `SERIAL_ECHOLNPGM`to report the offending value that is passed to `calc_timer_interval()` that would be very helpful.

ThomasToka, 2023-03-24:
Can you give me a matching SERIAL_ECHOLNPGM so i could try?

tombrazier, 2023-03-24:
For the call to `calc_timer_interval()` on around line 2260 it would be
```
SERIAL_ECHOLNPGM("step_rate ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
```

That will print out on the serial terminal, assuming you're using something like Octoprint or Prontrface.

tombrazier, 2023-03-24:
And you can keep adding pairs of strings and values. e.g.
```
SERIAL_ECHOLNPGM("step_rate ", (acc_step_rate + la_step_rate) >> current_block->la_scaling, " asr ", acc_step_rate, " lsr ", la_step_rate, " las ", current_block->la_scaling);
```

ThomasToka, 2023-03-24:
Can you give me a examples for all 4? I will then do the tests and post. I dont know what you exactly need on the four points.
But can invest the time to debug.

tombrazier, 2023-03-24:
Just copy the expression that is passed to `calc_timer_interval()` and, if it makes sense, also the variables that make up that expression. Sorry to be difficult, I just really need to be doing something else today.

ThomasToka, 2023-03-24:
Sorry i am not that good in serial operations.. but will help best i can.

1)

```
            SERIAL_ECHOLNPGM("step_rate ", (acc_step_rate + la_step_rate) >> current_block->la_scaling, " asr ", acc_step_rate, " lsr ", la_step_rate, " las ", current_block->la_scaling);
            la_interval = calc_timer_interval((acc_step_rate + la_step_rate) >> current_block->la_scaling);
```
            
 2) 
 
 `SERIAL_ECHOLNPGM("step_rate2 ", (reverse_e ? la_step_rate - step_rate : step_rate - la_step_rate) >> current_block->la_scaling, " sr ", step_rate, " lsr ", la_step_rate, " las ", current_block->la_scaling);
              la_interval = calc_timer_interval((reverse_e ? la_step_rate - step_rate : step_rate - la_step_rate) >> current_block->la_scaling);`

3)

`SERIAL_ECHOLNPGM("step_rate3 ", (current_block->initial_rate + la_step_rate) >> current_block->la_scaling, " lsr ", la_step_rate, " las ", current_block->la_scaling);
          la_interval = calc_timer_interval((current_block->initial_rate + la_step_rate) >> current_block->la_scaling);`
 
 like this or after the la_interval ?

ThomasToka, 2023-03-24:
```
Recv: step_rate 5396 asr 1396 lsr 4000 las 0
Recv: step_rate 5601 asr 1601 lsr 4000 las 0
Recv: step_rate 5755 asr 1755 lsr 4000 las 0
Recv: step_rate 5866 asr 1866 lsr 4000 las 0
Recv: step_rate 5938 asr 1938 lsr 4000 las 0
Recv: step_rate 5979 asr 1979 lsr 4000 las 0
Recv: step_rate 5997 asr 1997 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
Recv: step_rate3 6000 lsr 4000 las 0
```

it spams so lets see what comes out at 40% ;)

ThomasToka, 2023-03-24:
Oh it emediately stopped at the skirt.

ThomasToka, 2023-03-24:
sorry had to restart my octoprint as the logs were not written automatically.

just started a print again.

It looks like it stopps extruding.

Will post log in some minutes.

I could not see anything relevant while live watching last time.

ThomasToka, 2023-03-24:
[serial(1).log](https://github.com/MarlinFirmware/Marlin/files/11062452/serial.1.log)
[octoprint(3).log](https://github.com/MarlinFirmware/Marlin/files/11062453/octoprint.3.log)

cant see anything. will stop debugging now as i dont have any time left today.

ThomasToka, 2023-03-24:
short video

thisiskeithb, 2023-03-24:
Adding some debug lines causes the print to fail a lot faster now. I can do a basic purge line and then the machine reboots right after it moves to the middle of the bed to start printing the actual part. It fails right after this call:

https://github.com/MarlinFirmware/Marlin/blob/28666f2fb28e8b68f641e507359a72dbf4bc117a/Marlin/src/module/stepper.cpp#L2387

Log output:

```log
Send: N51 G1 X148.024 Y133.735 E.12833*102
Recv: Call 3 Start
Recv: Call 3 End
Recv: 
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
```

tombrazier, 2023-03-24:
What value for la_interval is returned, Keith? And what was passed to calc_timer_interval()?

thisiskeithb, 2023-03-24:
> What value for la_interval is returned, Keith? And what was passed to calc_timer_interval()?

I'll recompile and check. Many of those variables were "not declared in this scope" when adding your debug output & would not compile (that's why I wrapped the calls the way I did), but here's a copy of `stepper.cpp` that compiles for me:

[stepper.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/11064415/stepper.cpp.zip)

<details><summary>And the diff for faster reading with offending debug lines/variables commented out:</summary>
<p>

```diff
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 5c0034a..57ea3e4 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -2254,7 +2254,16 @@ hal_timer_t Stepper::block_phase_isr() {
         #if ENABLED(LIN_ADVANCE)
           if (la_active) {
             const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+            SERIAL_EOL();
+            SERIAL_ECHO("Call 1 Start");
+            SERIAL_EOL();
             la_interval = calc_timer_interval((acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+            SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+            SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+            SERIAL_ECHO("Call 1 End");
+            SERIAL_EOL();
           }
         #endif
 
@@ -2326,7 +2335,16 @@ hal_timer_t Stepper::block_phase_isr() {
             const uint32_t la_step_rate = la_advance_steps > current_block->final_adv_steps ? current_block->la_advance_rate : 0;
             if (la_step_rate != step_rate) {
               bool reverse_e = la_step_rate > step_rate;
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 2 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval((reverse_e ? la_step_rate - step_rate : step_rate - la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call  2 End");
+              SERIAL_EOL();
 
               if (reverse_e != motor_direction(E_AXIS)) {
                 TBI(last_direction_bits, E_AXIS);
@@ -2384,7 +2402,16 @@ hal_timer_t Stepper::block_phase_isr() {
 
           #if ENABLED(LIN_ADVANCE)
             if (la_active)
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 3 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval(current_block->nominal_rate >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              //SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call 3 End");
+              SERIAL_EOL();
           #endif
         }
 
@@ -2706,7 +2733,16 @@ hal_timer_t Stepper::block_phase_isr() {
       #if ENABLED(LIN_ADVANCE)
         if (la_active) {
           const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+          SERIAL_EOL();
+          SERIAL_ECHO("Call 4 Start");
+          SERIAL_EOL();
           la_interval = calc_timer_interval((current_block->initial_rate + la_step_rate) >> current_block->la_scaling);
+          //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+          //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+          SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+          SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+          SERIAL_ECHO("Call 4 End");
+          SERIAL_EOL();
         }
       #endif
     }
```

</p>
</details>

I'm using the [default Biqu Hurakan config](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/BIQU/Hurakan) so you can see which features are enabled.

thisiskeithb, 2023-03-24:
Updated `stepper.cpp`: [stepper.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/11064587/stepper.cpp.zip)

<details><summary>And the updated diff:</summary>
<p>

```diff
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 5c0034a..70f2f64 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -2254,7 +2254,17 @@ hal_timer_t Stepper::block_phase_isr() {
         #if ENABLED(LIN_ADVANCE)
           if (la_active) {
             const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+            SERIAL_EOL();
+            SERIAL_ECHO("Call 1 Start");
+            SERIAL_EOL();
             la_interval = calc_timer_interval((acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+            SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+            SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+            SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+            SERIAL_ECHO("Call 1 End");
+            SERIAL_EOL();
           }
         #endif
 
@@ -2326,7 +2336,17 @@ hal_timer_t Stepper::block_phase_isr() {
             const uint32_t la_step_rate = la_advance_steps > current_block->final_adv_steps ? current_block->la_advance_rate : 0;
             if (la_step_rate != step_rate) {
               bool reverse_e = la_step_rate > step_rate;
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 2 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval((reverse_e ? la_step_rate - step_rate : step_rate - la_step_rate) >> current_block->la_scaling);
+              SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call  2 End");
+              SERIAL_EOL();
 
               if (reverse_e != motor_direction(E_AXIS)) {
                 TBI(last_direction_bits, E_AXIS);
@@ -2384,7 +2404,17 @@ hal_timer_t Stepper::block_phase_isr() {
 
           #if ENABLED(LIN_ADVANCE)
             if (la_active)
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 3 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval(current_block->nominal_rate >> current_block->la_scaling);
+              SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              //SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call 3 End");
+              SERIAL_EOL();
           #endif
         }
 
@@ -2706,7 +2736,17 @@ hal_timer_t Stepper::block_phase_isr() {
       #if ENABLED(LIN_ADVANCE)
         if (la_active) {
           const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+          SERIAL_EOL();
+          SERIAL_ECHO("Call 4 Start");
+          SERIAL_EOL();
           la_interval = calc_timer_interval((current_block->initial_rate + la_step_rate) >> current_block->la_scaling);
+          SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+          //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+          //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+          SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+          SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+          SERIAL_ECHO("Call 4 End");
+          SERIAL_EOL();
         }
       #endif
     }
```

</p>
</details>

`la_interval` was `571` when it failed. Here's some of the log I have before it rebooted (right after the purge line / after the move to the center of the bed):

<details><summary>log output:</summary>
<p>

```prolog
Recv: Call 3 Start
Recv: la_interval: 209
Recv: las: 0
Recv: Call 3 End
Recv: ok
Send: N50 G1 X150.372 Y131.346 E.09582*97
Recv: ok
Send: N51 G1 X148.024 Y133.735 E.12833*102
Recv: Call 3 Start
Recv: la_interval: 571
Recv: las: 0
Recv: Call 3 End
[...]
Recv: 
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
```

</p>
</details>

tombrazier, 2023-03-25:
Keith I should have spotted your "Call 3" code is outside the preceding `if` statement. Needs braces.

thisiskeithb, 2023-03-25:
> Keith I should have spotted your "Call 3" code is outside the preceding `if` statement. Needs braces.

Those are missing from the current code: 

https://github.com/MarlinFirmware/Marlin/blob/8d3142911c017b542f1b55e9d2c09b68beacadd9/Marlin/src/module/stepper.cpp#L2385-L2388

Here's the updated `stepper.cpp` I used for testing: [stepper.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/11069852/stepper.cpp.zip)

<details><summary>...and the updated diff:</summary>
<p>

```diff
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 5c0034a..c7f1171 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -2254,7 +2254,17 @@ hal_timer_t Stepper::block_phase_isr() {
         #if ENABLED(LIN_ADVANCE)
           if (la_active) {
             const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+            SERIAL_EOL();
+            SERIAL_ECHO("Call 1 Start");
+            SERIAL_EOL();
             la_interval = calc_timer_interval((acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+            SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+            SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+            SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+            SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+            SERIAL_ECHO("Call 1 End");
+            SERIAL_EOL();
           }
         #endif
 
@@ -2326,7 +2336,17 @@ hal_timer_t Stepper::block_phase_isr() {
             const uint32_t la_step_rate = la_advance_steps > current_block->final_adv_steps ? current_block->la_advance_rate : 0;
             if (la_step_rate != step_rate) {
               bool reverse_e = la_step_rate > step_rate;
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 2 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval((reverse_e ? la_step_rate - step_rate : step_rate - la_step_rate) >> current_block->la_scaling);
+              SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call  2 End");
+              SERIAL_EOL();
 
               if (reverse_e != motor_direction(E_AXIS)) {
                 TBI(last_direction_bits, E_AXIS);
@@ -2383,8 +2403,19 @@ hal_timer_t Stepper::block_phase_isr() {
           ticks_nominal = calc_multistep_timer_interval(current_block->nominal_rate << oversampling_factor);
 
           #if ENABLED(LIN_ADVANCE)
-            if (la_active)
+            if (la_active) {
+              SERIAL_EOL();
+              SERIAL_ECHO("Call 3 Start");
+              SERIAL_EOL();
               la_interval = calc_timer_interval(current_block->nominal_rate >> current_block->la_scaling);
+              SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+              //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+              //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+              //SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+              SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+              SERIAL_ECHO("Call 3 End");
+              SERIAL_EOL();
+            }
           #endif
         }
 
@@ -2706,7 +2737,17 @@ hal_timer_t Stepper::block_phase_isr() {
       #if ENABLED(LIN_ADVANCE)
         if (la_active) {
           const uint32_t la_step_rate = la_advance_steps < current_block->max_adv_steps ? current_block->la_advance_rate : 0;
+          SERIAL_EOL();
+          SERIAL_ECHO("Call 4 Start");
+          SERIAL_EOL();
           la_interval = calc_timer_interval((current_block->initial_rate + la_step_rate) >> current_block->la_scaling);
+          SERIAL_ECHOLNPGM("la_interval: ", la_interval);
+          //SERIAL_ECHOLNPGM("step_rate: ", (acc_step_rate + la_step_rate) >> current_block->la_scaling);
+          //SERIAL_ECHOLNPGM("asr: ", acc_step_rate);
+          SERIAL_ECHOLNPGM("lsr: ", la_step_rate);
+          SERIAL_ECHOLNPGM("las: ", current_block->la_scaling);
+          SERIAL_ECHO("Call 4 End");
+          SERIAL_EOL();
         }
       #endif
     }

```

</p>
</details>

It still fails right after the move to the center of the bed, after the purge line (where the `.gcode` sets a LA/`M900`) value). I also no longer get any "Case #" / debug lines on serial after adding those braces to the "Call 3" section:

<details><summary>Error details:</summary>
<p>

```prolog
Send: N47 G1 X143.803 Y84.524 E.21104*85
[...]
Recv: ok
Send: N48 G1 X145.326 Y96.785 E.47335*90
Recv: ok
Send: N49 G1 X150.372 Y128.845 E1.2434*111
Recv: ok
Send: N50 G1 X150.372 Y131.346 E.09582*97
Recv: ok
Send: N51 G1 X148.024 Y133.735 E.12833*102
Recv: 
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
```

</p>
</details>

tombrazier, 2023-03-25:
> Those are missing from the current code:

It's a C++ thing.

I have replicated a hang on a BTT SKR Mini E3 V3.0. Even with commit ca77850cbb8ed57d4cdcf29a12a278d6bfa5c0d5 reverted, I still get a hang with the debug code in place. I suspect something more subtle is going on.

thinkyhead, 2023-03-25:
> I have replicated a hang on a BTT SKR Mini E3 V3.0. Even with commit ca77850cbb reverted, I still get a hang with the debug code in place. I suspect something more subtle is going on.

Printing to serial from within an ISR (e.g., `block_phase_isr`) will cause all kinds of mayhem. You can instead set some values in a global array and then print to serial from the main loop.

thinkyhead, 2023-03-25:
I'm peeking at the code changes since 194f58740a and will just write down my suggestions of things to check, starting with…

Make sure that `calc_timer_interval` and `calc_multistep_timer_interval` are never returning 0. Logic was updated so that a `ticks_nominal` value of 0 means it needs recalculation. So it should never get set to 0 by anything other than the "please recalculate" code.

tombrazier, 2023-03-26:
Okay so...
- The bug in the OP will always be characterised by the "eechy" sound from the E stepper. (This is the E stepper running as fast as the stepper ISR can generate steps.)
- I have replicated the bug with Keith's gcode. It is, indeed, caused by a division by zero and #25557 solves it. Thinky has merged that PR so I think this issue is resolved.
- Quite a few of the hangs observed above are actually caused by the serial output code I asked people to add. So that just confused things.
- @ThomasToka I am not convinced that you have the same bug because a) lack of "eechy" sound and b) commit ca77850 from #25557 does not resolve it. I also don't think that when you tested either commit of #25557 you had any serial debug statements. So this may be another bug entirely. Can you post your config and gcode?

thisiskeithb, 2023-03-26:
I'm running the same test print I've been using without all that serial debug stuff on the Biqu Hurakan (BTT Manta M4P / STM32G0) on the latest `bugfix-2.1` (241c062), so I'll report back in a few.

@tombrazier: Did you see this comment from @RV-from-be: https://github.com/MarlinFirmware/Marlin/pull/25557#issuecomment-1484131460? That PR didn't fully fix the issue apparently.

tombrazier, 2023-03-26:
Further observation: the small step rate being passed to `calc_timer_interval()` which becomes zero due to the bit shift occurs during deceleration when the E stepper switches from forwards to backwards movement. It is momentary. Even if the number is not zero, there is still a bug because it causes a long delay between updates of `nextAdvanceISR` in `Stepper:isr()` during which time the step rate will have increased significantly but the stepper ISR will have generated only 1 E step. I am considering changing
```
      else if (nextAdvanceISR == LA_ADV_NEVER)          // Start LA steps if necessary
        nextAdvanceISR = la_interval;
```
to
```
      else if (nextAdvanceISR > la_interval)          // Start or accelerate LA steps if necessary
        nextAdvanceISR = la_interval;
```

@thinkyhead

tombrazier, 2023-03-26:
> Did you see this comment from @RV-from-be: https://github.com/MarlinFirmware/Marlin/pull/25557#issuecomment-1484131460? That PR didn't fully fix the issue apparently.

Thanks for pointing me at that Keith.

tombrazier, 2023-03-26:
@RV-from-be @ThomasToka Can you test an idea for me please. Change `Stepper::calc_timer_interval()` in `Marlin/src/module/stepper.cpp` from
```
return step_rate ? uint32_t(STEPPER_TIMER_RATE) / step_rate : HAL_TIMER_TYPE_MAX;
```
to
```
return (step_rate > uint32_t(STEPPER_TIMER_RATE) / 65535) ? uint32_t(STEPPER_TIMER_RATE) / step_rate : HAL_TIMER_TYPE_MAX;
```

ThomasToka, 2023-03-26:
@tombrazier just saw this.. printing.. result in some minutes.

thinkyhead, 2023-04-01:
I believe this one may be solved, but we should test a variety of conditions just to be sure.

thisiskeithb, 2023-04-01:
For anyone following this issue / experienced crashes, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Be sure to attach a ZIP file containing your `Configuration.h`, `Configuration_adv.h`, and G-code in your reply.

vovodroid, 2023-04-01:
> please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Any specific print options? I gave BTT SKR 2 board.

thisiskeithb, 2023-04-01:
> Any specific print options and print duration? I gave BTT SKR 2 board.

We’re you experiencing board halts with `LIN_ADVANCE` & a K-factor (`M900`) other than 0?

I couldn’t get STM32F4-based boards to halt (including the BTT002 & SKR V2) in my testing of this issue.

vovodroid, 2023-04-01:
> Were you experiencing board halts with `LIN_ADVANCE` & a K-factor (`M900`) other than 0?

I didn't build Marlin with suspected commits. But I'll try latest build with fixes anyway.

vovodroid, 2023-04-01:
More than 1 hour print on BTT SKR 2 seems to be OK.

dwzg, 2023-04-01:
Haven't had a crash again with the most recent bugfix branch. Even a print that would always trigger the crash at a specific point is now running fine on my SKR E3 Turbo.

PendulumPit, 2023-04-02:
> For anyone following this issue / experienced crashes, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us [...]

I will give it a go. I had the issue when testing. Thanks to the Team for your hard work. :+1:

SpannMagoo, 2023-04-02:
Crashed right when its about to do purge line
[Configuration_4.2.7.zip](https://github.com/MarlinFirmware/Marlin/files/11131050/Configuration_4.2.7.zip)

thisiskeithb, 2023-04-02:
> Crashed right when its about to do purge line. Creality 4.2.7 Ender 3 [Configuration_4.2.7.zip](https://github.com/MarlinFirmware/Marlin/files/11131050/Configuration_4.2.7.zip)

Can you attach the g-code you were printing with?

SpannMagoo, 2023-04-02:
Posted in same post. Used vscode, Auto Build Marlin.

tombrazier, 2023-04-02:
@SpannMagoo Can you check whether it still happens in my new test branch?

https://github.com/tombrazier/Marlin/tree/lafreeze2

SpannMagoo, 2023-04-02:
OK, ran the newest branch as request. I ran 3 tests and failed on the 2nd test at the purge line. See below for configs and the .gcode
[Configuration_LatestPR4.2.7.zip](https://github.com/MarlinFirmware/Marlin/files/11133322/Configuration_LatestPR4.2.7.zip)
[ringing_tower_56m_0.20mm_200C_PLA-3.0.zip](https://github.com/MarlinFirmware/Marlin/files/11133323/ringing_tower_56m_0.20mm_200C_PLA-3.0.zip)

SpannMagoo, 2023-04-04:
Also just noticed it only does the first 3 points when doing the g29 and stops.

tombrazier, 2023-04-04:
Additional info: @SpannMagoo tells me that when the hang occurs, it is an M112 halt, complete with a screen saying that. This is not the same bug.

SpannMagoo, 2023-04-04:
> Additional info: @SpannMagoo tells me that when the hang occurs, it is an M112 halt, complete with a screen saying "M112 Shutdown". This is not the same bug.
> 
> 
> 
> [Edit: looking at the code, it appears that the only way you can get there is if M112 is sent to the printer.]

This is interesting because I did not have the pi plugged in the last time.

tombrazier, 2023-04-05:
@SpannMagoo reports (offline conversation) that the M112 bug began after the 18 March edition of bugfix. There were a lot of merges on 18 March and depending on quite where the snapshot occurred which SpannMagoo downloaded, this may or may not mean that there are a lot of other potential changes that could be related to this bug.

I think this probably warrants a new bug report.

As for this one, i.e. #25553, no one else is reporting anything here.

[11 more replies]

closed 2023-04-05 by thisiskeithb