MarlinFirmware/Marlin #25555: [BUG] Artillery Hornet motherboard fan doesn't work

DarkPako84, 2023-03-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I compiled the stock firmware for the Ruby board, and when printing, the mainboard fan stays still and emits small ticks.

### Version of Marlin Firmware

2.1.x

### Printer model

Artillery Hornet

### Electronics

stock electronics

### Add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11055798/Configuration.zip)

DarkPako84, 2023-03-25:
In the file "Configuration_adv.h" I tried to change "#define CONTROLLER_FAN_PIN FAN2_PIN" to "#define CONTROLLER_FAN_PIN PC6" and now it turns on. Is this the problem?

ellensp, 2023-03-25:
FAN2_PIN is defined in pins_ARTILLERY_RUBY.h as the following.

```
#ifndef FAN2_PIN
  #define FAN2_PIN                          PC6   // Fan2
#endif
```

So that doesn't make any sense as it should be identical behavior

DarkPako84, 2023-03-25:
I don't understand why but that's the way it works, it doesn't make sense

ellensp, 2023-03-25:
If you chang it back, does it break again?

DarkPako84, 2023-04-05:
I tried installing the Klipper firmware and the fan works 100%, it is definitely a setting problem!

closed 2023-08-09 by github-actions[bot]