MarlinFirmware/Marlin #25559: [BUG] (All Axis remaining trusted when steppers are off)

Martian-35, 2023-03-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The printer trusts the last position it was even though the steppers were inactive. This can cause crashes if the Axis is moved manually. My ender 3V2 has a Magnetic probe that is stored on the gentry all the way on the right side. To dock it just needs to move to X243 and it attaches. 

### Bug Timeline

Seems to have been an issue in the past: https://github.com/MarlinFirmware/Marlin/issues/23095

### Expected behavior

(VIDEO 1 IN THE LINKED FOLDER)
If its the first G28 it trusts no axis so it homes X and Y, picks up the probe (X243), goes to the center of the bed (X114) and the lowers Z till it hits the be with the probe. Prefect. if it would do this every time the Steppers get disabled everything would be good.

### Actual behavior

But what actually happens is on your second print, after the motors have been disabled, it grabs the probe, but instead of going to the center of the bed, it goes to down to Z20 first and the over to middle of the bed.(VIDEO 2) This works most of the time out of luck, BUT if the the Gentry has been lower manually since the last print, it crashes into the probe removing mechanism/edge of bed on the way down to Z20. (VIDEO 3) So it trusts the previous Z position instead of doing what it does on the first homing after start up. I even ran a "G28 O" which only probe untrusted axis and it did nothing. So all axis are trusted but "Z" is the biggest problem, in my case.

### Steps to Reproduce

1.  Get an Ender 3 with a Klack Magnetic probe.
2. Home all axis.
3. Raise the gentry up about Z100 through the menu.
4. Disable steppers.
5. Lower gentry manually by 20mm or more.
6. Home again. 
***IT WILL CRASH! SO HAVE YOUR FINGER ON THE POWER TO STOP IT AFTER IT GOES PAST WHERE Z20 SHOULD BE***

### Version of Marlin Firmware

2.1.2 bugfix 

### Printer model

Creality Ender 3v2

### Electronics

Creality 4.2.7 board

### Add-ons

Klackender magnetic probe, removed Z homing switch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

(LINK TO VIDEOS AND CONFIG FILES)
https://drive.google.com/file/d/1dfbowAQxYEUUU37z1stysA27-hXN2aYc/view?usp=sharing

RELATED FIRMWARE CHANGES:
-Enabled 
...#define MAG_MOUNTED_PROBE...

-Enabled 
...#define USE_PROBE_FOR_Z_HOMING...

-Changed 
...#define X_MAX_POS X_BED_SIZE +15...
so the X axis can dock

-Enabled:
...#define AUTO_BED_LEVELING_UBL...

-Enabled
...#define RESTORE_LEVELING_AFTER_G28...

-Enabled
...#define Z_SAFE_HOMING...

-Changed the V422 in
...#define MOTHERBOARD BOARD_CREALITY_V422...
to V427

all the changes I made are marked "//SS"

thisiskeithb, 2023-03-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/25559

thisiskeithb, 2023-03-24:
https://github.com/MarlinFirmware/Marlin/issues/23095

thisiskeithb, 2023-03-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23095

closed 2023-03-24 by thisiskeithb