MarlinFirmware/Marlin #25562: [BUG] SKR E3 Mini V3 Flash Blue Screem

DIYdie, 2023-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have been trying to compile the latest version of the bugfix and I am using the default config files from the example config and it compiles. After I put it into the printer to flash the screen stays blue and after I take the SD card out it does turn into a .cur file so it is taking it every time. I have an older version I compile a while ago and it still works.
[Default Configs for SKR mini.zip](https://github.com/MarlinFirmware/Marlin/files/11067974/Default.Configs.for.SKR.mini.zip)

### Bug Timeline

As early as I know is on 3/20/23

### Expected behavior

To flash with the most recent version of marlin.

### Actual behavior

Printer turns file into .cur but stays at a blue screen and is unresponsive.

### Steps to Reproduce

Compile most recent bugfix with SKR E3 MIN V3.0 using the default example configs for the board.

### Version of Marlin Firmware

Marlin Bugfix 2.1.x

### Printer model

Creality Ender 3 Pro

### Electronics

SKR E3 MINI V3.0

### Add-ons

BLtouch, NeoPixels

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-03-26:
are you able to compare the files of the source code of the one that works to the newer marlin?

I bet that the file youre trying to flash is too large. I had the same kind of issue, until I realized that the mainboard chip memory wasn't as large enough to accept the .bin file.

ellensp, 2023-03-26:
provided configs build to:

Checking size .pio/build/STM32G0B1RE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.0% (used 19096 bytes from 147456 bytes)
Flash: [====      ]  41.8% (used 218892 bytes from 524288 bytes)
Building .pio/build/STM32G0B1RE_btt/firmware.bin

ellensp, 2023-03-26:
The same Provided Config on current bugfix builds and works without issues on SKR E3 Mini V3 with CR10_STOCKDISPLAY boots up and dispalys without issues

Issue is probably your sdcard, need reformatted or is just not compatible with the bootloader.

closed 2023-03-26 by ellensp

DIYdie, 2023-03-26:
Sorry to open this again, but I am still having this issue. I have reformatted my sd card and I also tried 3 different sd cards. They all work when flashing an earlier version of marlin. I have spent my time today trying different commits to see what works and have finally narrowed it down. Commit 444259da079f07490de016566bc66950aab473f3 is the last working commit that has worked for me. Where it breaks is commit 2bc711ce58cfc5594b19593935d1c62f8261fa24 anything after that my screen remains blue as if I were install the wrong firmware or configs. My config files are slightly different now in order to get them to work with earlier versions so I'll upload those. I used the same configs between these two commits as well.
[SKR Mini Config.zip](https://github.com/MarlinFirmware/Marlin/files/11073078/SKR.Mini.Config.zip)

thisiskeithb, 2023-03-26:
I just used the provided configs using the latest `bugfix-2.1.x` (241c062) from 11 hours ago and after updating the `*_ENDSTOP_HIT_STATE` settings to allow a successful compile, it flashed/works fine on an SKR Mini E3 V3.0 with a `CR10_STOCK_DISPLAY`.

Here's a copy of the firmware.bin generated: [firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/11073119/firmware.bin.zip).

Are you sure you have a V3.0 board (STM32G0B1RE) and not a V3.1 (STM32F401RC)?

DIYdie, 2023-03-26:
That file did work for. I'm certain it's v3.0 I got this board right when it came out and the STM32G0B1RE build environment in marlin auto build always worked for me in the past. Maybe it's something to do with the way it is setup then? Still does make sense that past commits still work.

thisiskeithb, 2023-03-26:
> That file did work

So it's something with your computer/PIO install.

Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

---

At any rate: 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

classicrocker883, 2023-03-27:
try to make a clean up. PlatformIO: Clean

its the little trashcan at the bottom.

you might have to go in the terminal, its the tab in the same box where Output is or press ( crtl + ` or ~ ). you may have to press Q in order to get to the command line. then there are some **pio** commands you can enter 
>pio system prune

@DIYdie did you manage to find an actual working firmware file that you know works as it should? because that would leave the hardware being the issue off the table.

try renaming the file, put it in a different folder, or in the root of the drive.

DIYdie, 2023-03-27:
Thanks for the all the help, I'll definitely go to one of those links first next time. Thought it was a code error when it turn out to be a user error. What worked was deleting .platformio\ file and it finally flashed correctly.