MarlinFirmware/Marlin #25565: [BUG] Additional bed move while G29 probing

unitrix0, 2023-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I run the latest version of trhe bugfix-2.1.x branch and recently the printer started to additionally lower the bed when running a G29. Is that a new configuration option or abug? 

I configured `MULTIPLE_PROBING 2` and `BLTOUCH_HS_MODE true`

A video showing the problem: https://youtu.be/SN9cbFaW7ks

### Bug Timeline

new

### Expected behavior

In previous versions it did the two probes and then lowered to `Z_CLEARANCE_BETWEEN_PROBES` (I think) and then moved to the next point.

### Actual behavior

After the two probes it lowers the bed to `Z_CLEARANCE_BETWEEN_PROBES` and an additional ammount which I cannot tell where it comes from.

### Steps to Reproduce

Simply run a G29

### Version of Marlin Firmware

Bugfix-2.1.x 2023-03-24

### Printer model

Ender 5 plus

### Electronics

Bigtreetech SKR MINI E3 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11069036/Configurations.zip)

GMagician, 2023-03-25:
On code, not checked properly, there is something like:
_MAX(Z_CLEARANCE_BETWEEN_PROBES, Z_CLEARANCE_DEPLOY_PROBE)
when it try to deply probe. May be your case?

unitrix0, 2023-03-25:
I didn't change those values and they worked before. So, I don't know what your suggestion is

GMagician, 2023-03-25:
your extra move may be due to Z_CLEARANCE_DEPLOY_PROBE (5mm) vs Z_CLEARANCE_BETWEEN_PROBES (2mm). Not checked when/if something has been changed on probing recently

unitrix0, 2023-03-25:
Ok, so I've played around with those values now and, while they of course have an effect, it still does an additional move. It seems to me that there is something hard coded?

GMagician, 2023-03-25:
Maybe this. RElated to bltouch hispeed mode:
static float z_extra_clearance() { return high_speed_mode ? 7 : 0; }

unitrix0, 2023-03-25:
that would match up...
I turned the printer of right after the additional move and then checked the distance with a drill between the nozzle and the bed. 9mm was just fitting through. So those 7 and the 2 from my Z_CLEARANCE_BETWEEN_PROBES value would add up. Allthough why now, when it's there for 2 years...

unitrix0, 2023-03-25:
So I gave it a try and changed the line:
` // const float safe_z = _MAX(current_position.z, SUM_TERN(BLTOUCH, Z_CLEARANCE_BETWEEN_PROBES, bltouch.z_extra_clearance()));`
To:
`const float safe_z = _MAX(current_position.z, SUM_TERN(BLTOUCH, Z_CLEARANCE_BETWEEN_PROBES, 0));`

and voila. No additional move anymore.

unitrix0, 2023-03-25:
I've found another commit from March 18. 88da531e5d873c342a4a6625d9cb57adb2644cce. There this line was added.

thisiskeithb, 2023-03-25:
> I've found another commit from March 18. [88da531](https://github.com/MarlinFirmware/Marlin/commit/88da531e5d873c342a4a6625d9cb57adb2644cce). There this line was added.

I noticed an new, extra Z move with the latest bugfix on a Ender-3 with a BLTouch and suspected this was the offending commit but I currently have my machine torn apart for a hotend replacement and couldn't test it 🙂

V1EngineeringInc, 2023-03-30:
My untrained brain was just thinking, these issues popped up in the same timeframe. One has an extra Z move, and the other has a missing Z move. I was hoping this fix might also fix the other. I will try to test it in the next few days if no one gets there before I do.

thinkyhead, 2023-03-30:
I'm in the midst of trying to obtain consistent probe behavior across different probing procedures. I've added more debugging to get a better idea where and how to adjust things. Please test the latest code…

- Download Marlin `bugfix-2.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs I can get a better idea of where the raises are occurring and apply additional refinements.

V1EngineeringInc, 2023-03-31:
Hope this helps, My G29 works, it does seem to move extra far before probes. Next up trying the bltouch hs mode for the g34 error.
[g29log.txt](https://github.com/MarlinFirmware/Marlin/files/11127379/g29log.txt)

jamespearson04, 2023-03-31:
would you say about 7mm extra?

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/feature/bltouch.h#L79

That function is now used for some reason, implemented in the suspected PR #25498, commit 88da531

V1EngineeringInc, 2023-03-31:
Maybe. My other printers will be printing most of the night so I can't easily compare.

HS mode, for me, just keeps the bed in contact with the BLTouch, and doesn't ever clear all the way. I do not like the way it crashed so I prefer not to use it. Everything else seemed the same. Still missing the clearing move after the bed physically (g34) levels.

jamespearson04, 2023-04-01:
@unitrix0 For clarity, when you say extra move, do you mean it moves further than it should, but in one movement, or that it makes two distinct moves? because extra distance will definitely be from that commit, an additional move however, definitely not

V1EngineeringInc, 2023-04-01:
I'm not the right one to ask about this. G29 works fine for me. I was just hoping my logs might help clarify the actual moves happening.

I only have an issue with g34.

I do not see two distinct moves.

jamespearson04, 2023-04-01:
That's why I @'d unitrix ;) 

I've tried taking a look into G34, but I can't do any testing since my printer isn't dual Z

jamespearson04, 2023-04-03:
I had a bit of a mess around with it, I could tell that's what you were going for, so tried to tidy up the loose ends :) I'll open up a PR for it, so hopefully it'll just need merging in.

unitrix0, 2023-04-03:
> @unitrix0 For clarity, when you say extra move, do you mean it moves further than it should, but in one movement, or that it makes two distinct moves? because extra distance will definitely be from that commit, an additional move however, definitely not

it actually makes an addittional move. I've changed my configuration to make another video with slower probing speeds. Now it's also clear that the second move comes from somewhere else since it uses not the probing speed value.

https://youtu.be/USxZ_kpf-90

thisiskeithb, 2023-04-04:
Likely related, but possibly a different bug:

I updated to the current `bugfix-2.1.x` (c25a673) on a Prusa Bear with a PINDA V2 (`FIX_MOUNTED_PROBE`) and `MULTIPLE_PROBING 3`, it will only probe once on each point now.

unitrix0, 2023-04-04:
> I updated to the current `bugfix-2.1.x` ([c25a673](https://github.com/MarlinFirmware/Marlin/commit/c25a6737c6845c75474891203447101c06bace81)) on a Prusa Bear with a PINDA V2 (`FIX_MOUNTED_PROBE`) and `MULTIPLE_PROBING 3`, it will only probe once on each point now.
> 
> Changing `MULTIPLE_PROBING 3` to `MULTIPLE_PROBING 2` and enabling `EXTRA_PROBING 1` doesn't affect the probing behavior. It still only probes once per point.

Yes, can confirm that.

thisiskeithb, 2023-04-04:
Reverting https://github.com/MarlinFirmware/Marlin/commit/49f1cc8efe0fcb8df79581c0b71eedef250e83f7 fixes the single probe per point issue for me.

jamespearson04, 2023-04-04:
This should be fixed with #25631 - testing on my printer shows no extra lifts now.

unitrix0, 2023-04-04:
> This should be fixed with #25631 - testing on my printer shows no extra lifts now.

For G34 I assume?

thisiskeithb, 2023-04-04:
> For G34 I assume?

Both `G34` & `G29` issues.

Please test https://github.com/MarlinFirmware/Marlin/pull/25631 and report your findings in that PR.

closed 2023-04-05 by thisiskeithb