MarlinFirmware/Marlin #25569: [BUG] Creality V4.2.7 board with CR10 display - no sound

TzOk83, 2023-03-25:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

With the current v2.1.2 firmware, my Ender 3 with Creality V4.2.7 board makes no sound at all. I've tried compiling firmware with and without #define SPEAKER.

### Expected behavior

Buzzer is expected to beep when navigation the LCD menu.

### Actual behavior

Buzzer doesn't make any sounds.

### Version of Marlin Firmware

V2.1.2

### Printer model

Creality Ender 3

### Electronics

Creality 4.2.7

### Add-ons

BL-Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11070449/Configuration.zip)

classicrocker883, 2023-03-26:
in configuration.h look for
>//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
//#define LCD_FEEDBACK_FREQUENCY_HZ 5000

uncomment/enable this

ellensp, 2023-03-26:
Tested provided config as is, on stock Creality motherboard

Display Beeps on enter and exit of menus as expected.

I can see no issue here.

There are a number of build environments for this Controller.

env:STM32F103RE_creality 
env:STM32F103RE_creality_xfer 
env:STM32F103RE_creality_maple
env:STM32F103RC_creality 
env:STM32F103RC_creality_xfer 

What did you select and what chip is on your Controller?

Also verify you dont have a pin 27 board installed

[image]

These plug in the LCD port and steal the buzzer pin for a blouch

closed 2023-03-26 by ellensp

TzOk83, 2023-03-26:
Thank you for your answers.

I have tried all combinations of commenting and uncommenting of

> #define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
> #define LCD_FEEDBACK_FREQUENCY_HZ 5000

... as well as

> #define SPEAKER

... and I don't have a pin 27 board installed.

I am using "STM32F103RE_creality" build. BTW what is the difference between this and "STM32F103RE_creality_xfer" build?

classicrocker883, 2023-03-27:
`XFER_BUILD` gets enabled 

then it says this for comment in the **stm32f1.ini** file
> Creality (STM32F103Rx)
With custom upload to SD via Marlin with binary protocol.
Requires Marlin with BINARY_FILE_TRANSFER already installed on the target board.
If CUSTOM_FIRMWARE_UPLOAD is also installed, Marlin will reboot the board to install the firmware.
Currently CUSTOM_FIRMWARE_UPLOAD must also be enabled to use 'xfer' build envs.

use the creality default_envs for this board. you may also be able to use STM32F103RE_creality_maple if you so choose. that i wish I knew the difference between Maple and non-Maple. all i know is that Maple uses more memory, and some boards require it inorder for the flash to work. like Voxelab Aquila needs Maple, it wont flash without.

does anyone know the difference and if its possible to use without Maple for the Aquila? or some how disable some unnecessary features it preloads
?