MarlinFirmware/Marlin #25576: If Then style operations in start/end code

3DPRescue, 2023-03-26:
### Describe the feature you want

The ability to choose startup functionally like a purge line.

thisiskeithb, 2023-03-26:
> The ability to choose startup functionally like a purge line.

This would be handled via your slicer.

closed 2023-03-26 by thisiskeithb

3DPRescue, 2023-03-26:
Thanks for closing my ticket without actually grasping the concept. --Sent from my Android phone with mail.com Mail. Please excuse my brevity.

Nuck-TH, 2023-03-28:
You don't seem to grasp it either. Gcode is not programming language, it is set of instructions that can't directly depend on each other, only via state of machine(like waiting for heatup).
So the answer was correct - you need to edit start code in your slicer to add or remove desired actions.