MarlinFirmware/Marlin #25581: [BUG] (bug summary)Non responsive system with latest bugfix - Downloaded and compiled 27/3/2022

hexitex, 2023-03-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Non responsive system, no LCD information, system power supply kicks in and stay on.
Ramps 1.4, had a working system from bugfix 2 weeks ago, downloaded latest version with Fan0 changes approx 21:00 today - applied config changes where required. 

### Bug Timeline

today 

### Expected behavior

lcd screen display

### Actual behavior

blank screen - looks bricked 

### Steps to Reproduce

load onto ramps 1.4

### Version of Marlin Firmware

latest bugfix as of 21:00 gmt 26/3/2022

### Printer model

mendel90

### Electronics

ramps 1.4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

00:24:50.821 : Printer reset detected - initializing
00:24:50.855 : start
00:24:50.855 : External Reset
00:24:50.855 : Watchdog Reset
00:24:50.856 : Software Reset
00:24:50.856 : Marlin bugfix-2.1.x
00:24:50.856 : echo: Last Updated: 2023-03-26 | Author: (hexitex, default config)
00:24:50.856 : echo: Compiled: Mar 26 2023
00:24:50.856 : echo: Free Memory: 20478  PlannerBufferBytes: 1456
00:24:51.067 : N1 M110*34
00:24:51.068 : N2 M115*36
00:24:51.068 : N4 M114*35
00:24:51.068 : N5 M111 S6*98
00:24:51.070 : N6 T0*60
00:24:51.070 : N7 M20*22
00:24:51.070 : N8 M80*19
00:25:02.618 : N9 M111 S247*105
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11073334/Marlin.zip)

hexitex, 2023-03-27:
Something very odd with this release - downloaded latest src and made changes to config where in error - see attached.
If I compile this without EEprom enabled it works until I insert an SD card, then it crashes. If I compile with with EEProm enabled then just a blank screen. I've slimed down
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11081542/Marlin.zip)
 all the memory I can

hexitex, 2023-03-27:
Okay this turns out to be a RAM issue, something added recently has used a lot of RAM/data. A couple of weeks ago, I had everything I wanted enabled, now I have had to disable half of the features.

Bergerac56, 2023-03-28:
I got the same type of issue when I activated the Input Shaping option . It uses a lot of RAM

ellensp, 2023-03-28:
You have input shaping enabled, on a controller with 8k of memory
Disabling input shaping reduced the memory usage to an acceptable values

With input shaping disabled

RAM:   [======    ]  63.8% (used 5223 bytes from 8192 bytes)
Flash: [======    ]  62.0% (used 157432 bytes from 253952 bytes)

If you wish to enable input shaping on this very limited controller, you need to tune the amount of memory it uses by lowering DEFAULT_MAX_FEEDRATE  X and Y values.

hexitex, 2023-03-28:
Thank you Olivier, I changed max feed rate without considering the ram impact. That would explain why it was working with everything enabled. My bad