MarlinFirmware/Marlin #25582: [FR] Add option to change machine travel limits with GCODE

Tupson444, 2023-03-27:
### Describe the feature you want

In some cases, it would be convenient to be able to change max axis movement without changing firmware, using GCODE commands, then save to EEPROM. For example, if there are no other needed changes in the firmware but you change hardware to allow more clearance or change hotend and thus reduce maximum print height.

Sorry if it was already suggested (I couldn't find any) or if it's actually possible to do.

### Additional context

Disabling software endstops is not a good solution for this, as it adds risk of crashing.

ellensp, 2023-03-30:
Except this is counter to the Marlin philosophy. That hardware changes necessitate a recompile. This to keep the memory requirements and code size as small as possible. .

V1EngineeringInc, 2023-03-30:
I am fairly certain machine size has nothing to do with memory requirements. The only issues I see with dynamically changing bed size would be things like autoleveling, I am not sure if those points are adjusted dynamically or not. But for those of use not using bed leveling changing bed size would be a wonderful feature.

If you take it a step further and let it work with bed leveling, it could be used for things like the MK4 just announced, bed leveling only in the region you are using to speed up probe times.

ellensp, 2023-03-30:
your forgetting we still support older board with very limited resources.

V1EngineeringInc, 2023-03-30:
It could just as easily be optional, like almost everything is in Marlin. Like I said for our machines we use Extruders=0 at that point very ram or processor is used.

V1EngineeringInc, 2023-03-30:
If you are dead set against this I will unsubcribe from this feature request and forget about it. I think I have made it clear that I belive it would be very usefull for a lot of Marlin users. If you feel it is a waste of resources, so be it.

V1EngineeringInc, 2023-03-30:
Is that where the new .ini config will come into play that I am seeing in the 2.1.x branch? That would also be another solution.

ellensp, 2023-03-30:
config.ini files is just applied to config files, easily allowing for setting different static bed sizes and other settings 
It is only used at build time. not run time

Tupson444, 2023-03-30:
Of course, this feature should be optional (I forgot to say that). There could be something like "TRAVEL_LIMITS_GCODE" in the config, disabled by default. This way users can enable it when compiling, and then later if needed, just use GCODE instead of recompiling.

(although it should be advised to have it enabled whenever code size is not a problem, because you never know if you'll need it, and needing to recompile firmware just to enable it almost defeats the purpose of the feature)

EvilGremlin, 2023-03-30:
At this point, this will be unnecessary maintenance and support overhead. Well, as with any software, you're free to propose features or submit the code, and project maintainers are free to accept or reject anything at their discrertion. Talks about runtime geomerty changes been popping up for years, yet it's not in vanilla yet.

parallyze, 2023-03-31:
> Like the MK4 only probing the area under the current model being printed. This would save all probing printers a lot of time.

Slightly OT, but this is something that can already be done (with some limitations).
Changing the travel limits would only affect the area where probing will take place, not the amount of probe points. And
this can be done already depending on your slicer.

PrusaSlicer for example has variables for this:
`{first_layer_print_min[0]}, {first_layer_print_min[1]}` - x, y of the front left corner
`{first_layer_print_max[0]}, {first_layer_print_max[1]}` - x, y of the back right corner
`{first_layer_print_size[0]}, {first_layer_print_size[1]}` - x, y size

Using those in conjunction with G29 would allow to set the probing area. Using UBL one could also define the
"resolution" by using the x/y size to set the amount of probing points using Jn. 

The downsides I see: All fixed moves in the start gcode (purge/intro lines) have to be looked at very carefully as
the area where plastic will be extruded might be outside of the probed area. But handling this is very specific to the
machine in use. Also one would have to check if all required things are included in the variables, prime/purge
towers for example. I guess that's the reason why the purge/intro line is drawn just below the benchy in the 
sub 20 minute video Prusa just released for the MK4. If Y_MIN_POS is < 0 you could simply draw the line 1mm below the
probed area and wouldn't run into trouble even when the area starts at y=0. But the length along X is something
to be considered in case a print is placed near the right border...

Another thing not ideal: As far as I can see there's no way to define GRID_MAX_POINTS_X or GRID_MAX_POINTS_Y
during runtime and UBL/G29 Jn will always give you a n*n matrix. So in case of a print area like 180mmx40mm
and using J3 you'd end up with 3 points along 180mm on X and 3 points along 40mm on Y. Bilinear will
always probe the amount defined in GRID_MAX_POINTS_X/Y as far as I can see, please correct me if I'm wrong.

Thinking about possible implications on some machines (nozzle off the bed while heating, not homing to
minx/miny, origin at the bed center, homing offsets) when changing travel limits I'd much rather prefer a feature
request to add one or two options to G29 to achieve something like custom probe areas using a custom x/y resolution. 

Something like "GRID_RESOLUTION_X" and "GRID_RESOLUTION_Y" so the probing could happen in a user defined
area using size_x/grid_resolution_x and size_y/grid_resolution_y to set the amount of probing points.

hsyngndz85, 2024-04-05:
Bu ayar çok gerekli bi gelişme varmı

hsyngndz85, 2024-04-05:
Aslında menü lazım değil sadece treveel hızı gibi m coduyla max pos degerini usb üzeründen ayarlasakta olur. Sadece m codu tanımı gerekıyor ve hafızaya kaydedilmesi gerekiyor.

thisiskeithb, 2024-04-05:
> Actually, it does not require a menu, it just adjusts the maximum pos value via USB with the m code, such as the treveel speed.

For initial support that may be fine, but we almost always see a follow up request to add a menu option so values can be adjusted without needing an external host.

For this request, a menu option is pretty straight forward and not the main holdup.

hsyngndz85, 2024-04-05:
There is a custom menu, so you don't need to work on the menu for now. But max setting with m code is absolutely necessary urgently.

thisiskeithb, 2024-04-05:
>  But max setting with m code is absolutely necessary urgently.

There's nothing about this feature that warrants an urgent implementation, especially with our [backlog of bug reports](https://github.com/MarlinFirmware/Marlin/issues?q=is%3Aopen+is%3Aissue+label%3A%22Bug%3A+Potential+%3F%22) and [over 100 pull requests](https://github.com/MarlinFirmware/Marlin/pulls).

This open source project has a single full-time maintainer and a handful of regular contributors like myself attending to bug reports & upkeep. If you'd like to see this feature implemented "urgently", please [join our Discord](https://discord.gg/n5NJ59y) and post in the `#marlin-job-listings` channel along with your paying rate and someone from the open source community may implement it.

hsyngndz85, 2024-04-05:
I am happy to pay a fee. Thank you very much