MarlinFirmware/Marlin #25584: Marlin not compiling in BQ Hephestos 2 with heated bed

jorgemmg14, 2023-03-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin\src\module\temperature.h:511:55: error: 'TEMP_BED_HYSTERESIS' was not declared in this scope

Can't compile

### Bug Timeline

2.1   In 2.0.x it works

### Steps to Reproduce

Use BQ Hephestos 2 config.

Enable HEPHESTOS2_HEATED_BED_KIT in Configuration_adv.h

Compile

### Version of Marlin Firmware

2.1.x

### Printer model

BQ Hephestos 2

### Electronics

Heated Bed

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

From the example BQ Hephestos 2 after enabling heated bed in config_adv

GMagician, 2023-03-27:
You don't have a bed sensor (#define TEMP_SENSOR_BED 0), don't know which value to put because I don't know your printer

jorgemmg14, 2023-03-27:
Hi,
in Marlin 2.0.x I can compile it, but the Y-stepper does not work properly.
In marlin 2.1.x (and 2.1.x bugfix) I can't compile it.
The parameters in that firmware are #define TEMP_SENSOR_BED 0 too. The heated bed is the official one.
In configuration_adv theparameter TEMP_SENSOR_BED is set to 70

GMagician, 2023-03-27:
That's a bug, it should be defined in configuration.h

ellensp, 2023-03-28:
This PR https://github.com/MarlinFirmware/Marlin/pull/24528 added a #if block around TEMP_BED_HYSTERESIS and others.
This breaks #define HEPHESTOS2_HEATED_BED_KIT in Configuration_adv.h 

```
#if TEMP_SENSOR_BED
  #define TEMP_BED_RESIDENCY_TIME     10  // (seconds) Time to wait for bed to "settle" in M190
  #define TEMP_BED_WINDOW              1  // (°C) Temperature proximity for the "temperature reached" timer
  #define TEMP_BED_HYSTERESIS          3  // (°C) Temperature proximity considered "close enough" to the target
#endif
```

Bergerac56, 2023-03-28:
I confirm that this bug exists for a long time now. I solved it by adding  #define TEMP_BED_HYSTERESIS 3  in configuration_adv.h
And... sorry, I forgot to report this issue

#define HEPHESTOS2_HEATED_BED_KIT
#if ENABLED(HEPHESTOS2_HEATED_BED_KIT)
  #undef TEMP_SENSOR_BED
  #define TEMP_SENSOR_BED 70
  #define TEMP_BED_HYSTERESIS 3 // solve the compilation issue
  #define HEATER_BED_INVERTING true
#endif

ellensp, 2023-03-28:
Issue was introduced on Aug 4, 2022

ellensp, 2023-03-29:
The defining of TEMP_SENSOR_BED in Configuration_adv.h has been removed, instead replaced with a Error
"#error "HEPHESTOS2_HEATED_BED_KIT requires TEMP_SENSOR_BED 70."

Example config will be updated with this shortly.

Closing as done.