MarlinFirmware/Marlin #25592: [BUG] (bug summary)V87 stored settings retrieved (613 bytes; crc 53739)

TolTolich, 2023-03-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

During printing, the printer may stop at any time, the error "V87 stored settings retrieved (613 bytes; crc 53739)" appears on the screen, the numbers in brackets may be different depending on one or another firmware. I tried to install different firmware created by other participants, and compiled it myself. In addition, after flashing, the M502 command returns "Hardcoded Default Settings Loaded", M500 returns "Settings Stored (613 bytes; crc 53739)". The problem appeared after the transition to the Marlin 2.0 version. The motherboard was replaced with a similar one to rule out a hardware problem. Drivers for ТМC 2209 have been replaced on the board. Corresponding changes have been made to the firmware.

### Bug Timeline

The problem appeared about six months ago, when I decided to transfer the printer to custom firmware using Linear Advance

### Expected behavior

The error will disappear

### Actual behavior

The error is not gone

### Steps to Reproduce

1. Download the compiled firmware.
2. Through the terminal, enter M502 M500 M501
3. M501 shows message V87
4. You can reset the error, start printing, but after any time the printer may stop with error "V87 stored settings retrieved (613 bytes; crc 53739)" 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Mar 29 2023 13:24:07) 

### Printer model

Artillery Saidwinder X1

### Electronics

MKS GEN L V0.1 

### Add-ons

drivers TMC2209

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11099134/Configuration.zip)
[M501.txt](https://github.com/MarlinFirmware/Marlin/files/11099177/M501.txt)

ellensp, 2023-03-29:
this error message is not an error
This is the expected output om a M501 
this also happens on reboot/power on

TolTolich, 2023-03-29:
> это сообщение об ошибке не является ошибкой. Это ожидаемый вывод M501, это также происходит при перезагрузке/включении питания.
> 
> Так что это явно перезагрузка.

Are you saying that the message is not an error?

> this error message is not an error This is the expected output om a M501 this also happens on reboot/power on
> 
> So its clearly rebooting.

Are you saying that the message
...V87 stored settings retrieved (613 bytes; crc 53739)...
 is not an error?

TolTolich, 2023-03-29:
My printer stops printing when this message appears on the screen

TolTolich, 2023-03-29:
I don't know how I can fix this.

TolTolich, 2023-03-31:
Again the printer stopped in the process of printing with a message on the screen.  I don't know what not to do to solve the problem.  Please help!!![image]

[image]

thinkyhead, 2023-04-03:
Try disabling all optional features and doing more testing to try to narrow down the cause. Rebooting is often an indication that there is not enough free memory.

TolTolich, 2023-04-04:
Thanks for the answer. I'm not well versed in the details of the firmware, so I ask you to tell me which functions can be painlessly removed. Preferably by line numbers in configuration files. I am very tired of dealing with this problem.

Johnny846, 2023-04-22:
I don't know much about marlin either, but I guess simple things you can try disabling are bed leveling (which from your config seems already disabled), by commenting ( `//`) the following lines below in configuration.h 
`//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING`

You can also for example disable `//#define S_CURVE_ACCELERATION`  in configuration.h and disable `//#define LIN_ADVANCE` in configuration_adv.h

Perhaps you could also disable the display (somewhere in the configuration) and control the printer only by USB cable to check if it still keeps on rebooting. 

I must say that I myself, after uploading Marlin 2.0.9.5 LTS (I tried first 2.1.x, but it had too many other bugs), noticed that my printer rebooted a couple of times unexpectedly. However, I still didn't print anything after this firmware upload and haven't been using my printer much, so I don't know if this issue is also affecting me or if it was just a coincidence that it rebooted unexpectedly.

thisiskeithb, 2023-04-22:
The ribbon cables & connectors on these machines are also known to short, which would also cause reboots (among other things).

Disconnect each ribbon cable and carefully inspect each cable end and connector, checking for discolored pins/burn marks/melted plastic.

closed 2024-06-21 by github-actions[bot]