MarlinFirmware/Marlin #25599: [BUG] confusing error message when compiling Marlin for SKR 1.3 and TMC2130 and sensorless homing

Elchschmied, 2023-03-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[image]

Sensorless homing requires X_MIN_ENDSTOP_HIT_STATE HIGH for X_MIN homing with TMC2209

No TMC2209 activated but getting this message. 

### Expected behavior

should compile without error. TMC2209 should not be stated in message, because not activated

### Actual behavior

does not create firmware.bin

### Steps to Reproduce

1. setup TMC2130 for XYZ
2. define sensorless homing for X and Y
3. set X_MIN_ENDSTOP_HIT_STATE HIGH, Y_MIN_ENDSTOP_HIT_STATE HIGH, , Z_MIN_ENDSTOP_HIT_STATE HIGH

### Version of Marlin Firmware

bugfix 2.1.x STRING_DISTRIBUTION_DATE "2023-03-29"

### Printer model

Ender 3

### Electronics

SKR 1.3

### Add-ons

sensorless homing with TMC2130

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Repetier server is used
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11105117/Configuration.zip)

ellensp, 2023-03-30:
Error messages seem to be switched from what is needed

Elchschmied, 2023-03-30:
I found out that TMC2130 and TMC5160 need to be set to MIN_ENDSTOP_HIT_STATE LOW. Compiling than works.
Error messages I get, when setting is HIGH, referencing TMC2209 does not help even the error may be correct. 
I would await a message that the used driver names would be used. I searched to find a solution for my wrong setting, but this
message confuses more than it is helpfull. I now know what to do and let it go as it is. @ellensp  Thank you for looking into my issue.

ellensp, 2023-03-30:
Are you planning a PR to fix this? as it does need to be fixed 

Yes its even in the comments

    // Stall detection DIAG = HIGH : TMC2209
    // Stall detection DIAG = LOW  : TMC2130/TMC2160/TMC2660/TMC5130/TMC5160

Elchschmied, 2023-03-30:
What is a PR?

ellensp, 2023-03-30:
A pull request to add fixed code onto main code 

Since your asking this,the answer would seem to be no, so this should remain open

Elchschmied, 2023-03-30:
> Are you planning a PR to fix this? as it does need to be fixed
> 
> Yes its even in the comments
> 
> ```
> // Stall detection DIAG = HIGH : TMC2209
> // Stall detection DIAG = LOW  : TMC2130/TMC2160/TMC2660/TMC5130/TMC5160
> ```

where did you find this?

[image]

A PR would make sense to make more clear what has to be done. :)

ellensp, 2023-03-30:
In https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/SanityCheck.h#L3642-L3643

Elchschmied, 2023-03-30:
I haven't seen this, sorry. I'm not a programmer so I did'nt look into this file. How is the procedure to get this information even into the section of Endstop "Hit" State?

ellensp, 2023-03-30:
Don't worry about it, have created a PR to fix it.  Correct will be shown after this PR is accepted and added to the code

closed 2023-03-30 by ellensp

Elchschmied, 2023-03-30:
Thank you @ellensp