MarlinFirmware/Marlin #25602: [BUG] SD image can't be loaded after pressing "Generate Empty Image" in the simulator.

mriscoc, 2023-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The simulator allows you to test some Marlin functions without having a physical printer. It is possible to simulate a SD memory card by loading an image file, the simulator itself can create an empty image by pressing Generate Empty Image button:
[image]
But, after the image is created and loaded an error is shown in the console: "Error: volume, init failed"
[image]

Simulator compiled and running under Windows 10 with the example configuration files from the configuration repository.

The image is generated correctly, it was tested with third-party software.
[image]

### Bug Timeline

There are no previous notices about it

### Expected behavior

SD image must be loaded by the simulator without errors.

### Actual behavior

SD images can't be loaded by the simulator.

The image was then formatted to FAT32 with the same error.

### Steps to Reproduce

Download latest bugfix
Install simulator dependencies
Download simulator configuration files
Compile simulator
Run simulator
Pressing "Generate Empty Image" button:
[image]
The image is created but an error is shown in the console: "Error: volume, init failed"
[image]

### Version of Marlin Firmware

Latest Marlin Bugfix

### Printer model

Simulator

### Electronics

Simulator

### Add-ons

None

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-03-30:
Probably better to file up in actual sim repo https://github.com/p3p/MarlinSimUI
Tried it in linux yet?

p3p, 2023-04-07:
https://github.com/p3p/MarlinSimUI/issues/22

Turns out the issue was in Marlin and not the sim after all, although can't really blame it for the compiler on Windows messing up the bit packing.

mriscoc, 2023-04-07:
@p3p thanks!

thisiskeithb, 2023-04-07:
Closing since there’s now a PR.

closed 2023-04-07 by thisiskeithb