MarlinFirmware/Marlin #25605: [BUG] TinBee Board does not home Z if Input Shaping is enabled

ahmetcemturan, 2023-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If Input shapıing is enabled, the printer will home x and then y axis.  Then stops responding (does not home z).
Homing works fine if I send "M593 F0" before homing

### Bug Timeline

It is existent in the 2.1.2 and the latest bugfix

### Expected behavior

I expect it to home

### Actual behavior

It stops responding during the homing routine.

### Steps to Reproduce

enable input shaping and set it to a value >=1 
Home the printer

### Version of Marlin Firmware

2.1.2 or 2.1.bugfix

### Printer model

Prusa clone (Bedslinger)

### Electronics

Tinybee v3

### Add-ons

Z-probe as Z-endstop

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
echo:M105
echo:M105
echo:M105
echo:G28
echo:busy: processing
echo:busy: processing
echo:G92 E0
echo:M114
echo:M105
echo:M105
echo:M105
echo:M105
>>> M593 F22
SENDING:M593 F22
echo:M593 F22
echo:M105
echo:M105
echo:G28
echo:busy: processing
>>>>>Becomes Unresponsive....

will only respond after a reset.

Only other modification is That I have changed the ADC voltage to 2.565 as that gives correct values

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11113294/Marlin.zip)

tombrazier, 2023-04-04:
IS is not supported with ESP32. My naive guess is the FT is not either. We should add a sanity check.

ESP32 works around some of the core stepper logic. We really need a refactor of stepper.cpp that puts all the step generation in one place and prevents more than one step being generated per pass through the loop. Then ESP32 could be integrated with IS properly. (Not to mention baby stepping and really getting it working with LA.)

> when trying to compile Marlin with SHAPING_FREQ_ =0, I get an error. (not a big problem as it makes basically no sense)

You could want to compile with `SHAPING_FREQ_[XY]=0` so that IS defaults to disabled. That is possible if you define `SHAPING_MIN_FREQ`. There should be a sanity check error which tells you this.

thisiskeithb, 2023-04-04:
> IS is not supported with ESP32. My naive guess is the FT is not either.

I'll mark this as a feature request instead.

cbagwell, 2023-04-05:
>> when trying to compile Marlin with SHAPING_FREQ_ =0, I get an error. (not a big problem as it makes basically no sense)

>You could want to compile with SHAPING_FREQ_[XY]=0 so that IS defaults to disabled. That is possible if you define SHAPING_MIN_FREQ. There should be a sanity check error which tells you this.

I created #25358 to apply the related AVR sanity check to everything.

vfbank, 2023-05-27:
All hardware currently using the ESP32 chipset is experiencing the same issue. This is because the ESP32 chipset communicates through the I2S Stream. Regrettably, most of the Cods firmware does not properly support the I2S Stream.

Therefore, the following features are currently not implemented:

Incompatible with INPUT_SHAPING
Incompatible with Babystep
Incompatible with FT motion
Incompatible with Linear Advance

And there are currently issues with inaccurate temperature sensors and layer shift issues.

The ESP32 chipset has powerful processing performance, but unfortunately, only ARM or STM32 series are mainly developed in Marlin, so it's better not to use the ESP32 chipset until a capable person comes out. The reason why this board is inexpensive is because of the above reasons, and the manufacturer will no longer update after version 2.0.9.

EvilGremlin, 2023-05-27:
@vfbank you didn't pay attention, LA and babystep are supported now

ellensp, 2023-05-27:
@vfbank Your thinking is backwards, Need more controllers using esp32 to attract esp32 developers to the project

vfbank, 2023-05-28:
Unfortunately, when I activate the extension, it compiles, but Babysteps still doesn't work.

EvilGremlin, 2023-05-28:
INTEGRATED_BABYSTEP still don't work on i2s? Someone send Tom TinyBee already! :laughing:

vfbank, 2023-05-28:
> INTEGRATED_BABYSTEP 여전히 i2에서 작동하지 않습니까? 누군가 이미 Tom TinyBee를 보냅니다! 😆

If you enable define INTEGRATED_BABYSTEPPING, it compiles but babystep does not work. No matter how much you turn the dial, only the value changes, but the Z-axis does not move. Same for other functions
It compiles but doesn't work

vfbank, 2023-05-28:
> 

For reference, the ESP32 hardware should control all communication with 32 pins. About 2-4 times less compared to ARM or STM32 processors. This causes problems with layout organization. The EPS32 chipset operates at an amazing 480Mhz speed and has Bluetooth/WIFI functions, but due to the size of the chipset, there is a problem in configuring the communication interface.

vfbank, 2023-05-28:
> Unfortunately, when I activate the extension, it compiles, but Babysteps still doesn't work.

For reference, the ESP32 hardware should control all communication with 32 pins. About 2-4 times less compared to ARM or STM32 processors. This causes layout configuration issues. The EPS32 chipset operates at an amazing 480Mhz speed and has Bluetooth/WIFI functions, but due to the size of the chipset, there is a problem in configuring the communication interface.

The reason I think of it negatively is that I actually tried to improve it, but eventually gave up due to a lot of limitations. Improving the features won't be easy unless you're an engineer or developer with great skills.

vfbank, 2023-05-28:
> @vfbank Your thinking is backwards, Need more controllers using esp32 to attract esp32 developers to the project

For reference, the ESP32 hardware should control all communication with 32 pins. About 2-4 times less compared to ARM or STM32 processors. This causes layout configuration issues. The EPS32 chipset operates at an amazing 480Mhz speed and has Bluetooth/WIFI functions, but due to the size of the chipset, there is a problem in configuring the communication interface.

The reason I think of it negatively is that I actually tried to improve it, but eventually gave up due to a lot of limitations. Improving the features won't be easy unless you're an engineer or developer with great skills.

EvilGremlin, 2023-05-28:
> ESP32 chipset not properly supporting I2S streams

Maybe you meant framework?
Though as general stepping works, problem lie in forming pulse train which is 100% marlin code. Or could it be FreeRTOS messing with interrupts somehow?

ellensp, 2023-05-28:
except stepper drivers step pins are not directly connected to the esp32...
They are connected to another chip chip via i2s, the stepper step pins are off of that chip.

ellensp, 2023-05-28:
never-mind, just going senile..

vfbank, 2023-05-28:
> ESP32 chipset not properly supporting I2S streams

It seems to me that the ESP32 chipset's G-code processing part is the problem. I guess it's probably a communication problem (not sure)

I intercept the part that has some function in the pure G code (original source through SD card or USB) (sorry, English is not my native language, so I don't know how the translation will go) and replace that part with a new G code. , but it seems that this part is not working properly due to some problem.

So, while the existing data is being processed, it seems that all related functions do not work because it cannot be changed in the part where it is to be changed by modulating it with other data.

This is speculative, so it may not be correct. So I can't give you any good advice on this matter. sorry.

tombrazier, 2023-05-28:
> INTEGRATED_BABYSTEP still don't work on i2s? Someone send Tom TinyBee already!

First I have some X axis improvements to make on my printer so I can print a 15ish minute benchy on 8 bit and 12V just to showcase how awesome Marlin is.

Then I need to finish fixing my coffee machine and make a very niche youtube video for anyone else wanting to to do the same fix.

And I want to get on with my variation on rq3's TAP-XXX bed sensor.

Then it would be really great to look into why bugfix is still causing stepper issues for some users.

And there's a weird E stepper thing that might be related to a backlash correction bug.

And some memory improvements I've had in mind for ages.

And smooth multistepping.

And I've started contributing to another github project for head tracking for simulators.

And I really should look over the FT stuff.

And having a life.

How did I get in so deep? :open_mouth:

EvilGremlin, 2023-05-28:
At least you deserve some new shiny toys :stuck_out_tongue:

thinkyhead, 2023-05-30:
> First I have some X axis improvements to make on my printer so I can print a 15ish minute benchy on 8 bit and 12V just to showcase how awesome Marlin is.

I'll be very happy to post a video of that to Twitter to steal a little of the limelight from Prusa who are arriving late to this party but getting all the jellybeans.

tombrazier, 2023-05-30:
Great, thanks Scott. I'll let you know.

ellensp, 2026-10-19:
Need testers for the #28710 PR. I've done what I can with a bare board