MarlinFirmware/Marlin #25609: [BUG] M73 doesn't show remaing time.

vovodroid, 2023-03-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

M73 doesn't show remaing time (R) despite command exists on gcode file.

PrusaSlicer issues commands like 

```
	Line 87: M73 P1 R41
	Line 607: M73 P2 R41
	Line 655: M73 P2 R40
	Line 1852: M73 P3 R40
	Line 2926: M73 P4 R40
	Line 3351: M73 P4 R39
	Line 3692: M73 P5 R39
	Line 3967: M73 P6 R39
	Line 4325: M73 P6 R38

```
and Marlin is build with
```
// Add 'M73' to set print job progress, overrides Marlin's built-in estimate
#define SET_PROGRESS_MANUALLY
#if ENABLED(SET_PROGRESS_MANUALLY)
  #define SET_PROGRESS_PERCENT            // Add 'P' parameter to set percentage done, otherwise use Marlin's estimate
  #define SET_REMAINING_TIME            // Add 'R' parameter to set remaining time, otherwise use Marlin's estimate
  //#define SET_INTERACTION_TIME          // Add 'C' parameter to set time until next filament change or other user interaction
  //#define M73_REPORT                    // Report M73 values to host
  #if BOTH(M73_REPORT, SDSUPPORT)
    #define M73_REPORT_SD_ONLY            // Report only when printing from SD
  #endif
#endif

// LCD Print Progress options. Multiple times may be displayed in turn.
#if HAS_DISPLAY && EITHER(SDSUPPORT, SET_PROGRESS_MANUALLY)
  #define SHOW_PROGRESS_PERCENT           // Show print progress percentage (doesn't affect progress bar)
  #define SHOW_ELAPSED_TIME               // Display elapsed printing time (prefix 'E')
  #define SHOW_REMAINING_TIME             // Display estimated time to completion (prefix 'R')
  #if ENABLED(SET_INTERACTION_TIME)
    #define SHOW_INTERACTION_TIME         // Display time until next user interaction ('C' = filament change)
  #endif
  //#define PRINT_PROGRESS_SHOW_DECIMALS  // Show/report progress with decimal digits, not all UIs support this

  #if EITHER(HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL)
    //#define LCD_PROGRESS_BAR            // Show a progress bar on HD44780 LCDs for SD printing
    #if ENABLED(LCD_PROGRESS_BAR)
      #define PROGRESS_BAR_BAR_TIME 2000  // (ms) Amount of time to show the bar
      #define PROGRESS_BAR_MSG_TIME 3000  // (ms) Amount of time to show the status message
      #define PROGRESS_MSG_EXPIRE      0  // (ms) Amount of time to retain the status message (0=forever)
      //#define PROGRESS_MSG_ONCE         // Show the message for MSG_TIME then clear it
      //#define LCD_PROGRESS_BAR_TEST     // Add a menu item to test the progress bar
    #endif
  #endif
#endif
```

Nevertheless only progress bar is displayed:
[image]

### Expected behavior

Remaining time is shown and updated.

### Actual behavior

No remaining time is shown.

### Steps to Reproduce

Slice with Prusa with remaining time enabled.

[B1 SE.zip](https://github.com/MarlinFirmware/Marlin/files/11124992/B1.SE.zip)

### Version of Marlin Firmware

2.1.x bugfix 4b4d98d2, 21/12/22

### Printer model

Biqu B1 SE

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

lukasradek, 2023-03-31:
I concur!
I have tried all of the SHOW_ family options and it is not working on TFT_COLOR_UI.

thisiskeithb, 2023-03-31:
Not every feature works with `TFT_COLOR_UI`, so this is a feature request, not a bug.

lukasradek, 2023-03-31:
I just wanted to say that after looking at the code. It seems like it is not implemented in Color UI. I remember seeing PR implementing this though. If not, I might look into this in the future. I would like that feature as well.

thisiskeithb, 2023-03-31:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19780

thisiskeithb, 2023-03-31:
We have a PR open for this: https://github.com/MarlinFirmware/Marlin/pull/24843

closed 2023-03-31 by thisiskeithb