MarlinFirmware/Marlin #25610: [FR] Moment menu: Stop movement option

lukasradek, 2023-03-31:
### Is your feature request related to a problem? Please describe.

The printer movement is impossible to stop during the move itself. Which is potentially dangerous and impractical.

### Describe the feature you want

I suggest adding "stop" functionality to the movement screen (eg. graphical UI in TFT_COLOR_UI).  
I can implement it, but I want to ask (preferably some Marlin designer/developer) clarifying questions.

In the TFT_COLOR_UI there already is `off` button in the Movement menu. Intuitively I was pushing the button to stop the movement even though (as I later found) it is just to disable steppers.

I could add separate `stop` button, but I don't want to add duplicates.

Is there any reason why `Off` just disables steppers or can it stop current movement as well?  
If there is... what do you feel is more useful
1. `Stop` button to stop movement and  `Off` button to stop movement and disable steppers
2. `Stop` button to stop movement and `Off` button to disable steppers

thisiskeithb, 2023-04-01:
This would be tricky to implement on non-TFT’s since you’d need it to be a menu item and getting to that in time would be difficult.

From a safety perspective, if you want to halt motion, most LCDs/TFTs have a physical reset button that can reboot everything in an emergency.

There’s also `FREEZE_FEATURE` for a “Potentially useful for emergency stop that allows being resumed.” This would use a physical button to temporarily halt motion:

https://github.com/MarlinFirmware/Marlin/blob/1be2a7e4c17b8e19fade64ed32e101a65f9e8c32/Marlin/Configuration_adv.h#L4083-L4092

lukasradek, 2023-04-01:
I agree that it would be most usable on TFT_COLOR_UI but is that a reason not to implement it?

1. Differences in usability are inherent to the variety of UIs and input methods but I think it should not mean that features should be limited by the lowest common denominator. Just as there is a graphical motion menu (accessible from status screen) for TFT_COLOR_UI that is not present on character LCDs.
2. I have 3 printers in my experimentation sample. Průša Mk3s+ which has a dedicated reset button, Elegoon Saturn 2 which has a stop button in motion menu and TronXY with touch TFT, where I run Marlin with no ability to stop because there is no HW button and Marlin does not support stopping. And I think that most printers using TFTs with touch don't have HW buttons (because they don't need them).
3. If it is really necessary to implement it into the menu system (for character LCDs), there can simply be an item "stop" together with "move 10mm", "move 1mm" etc. in the terminal branch of the menu.

What say you? 🙂

thinkyhead, 2023-04-03:
Once a move is planned and posted to the planner buffer, it cannot be stopped without causing step-loss. All complete moves begin with acceleration and end with deceleration, and this is all pre-calculated. So, we have to wait for all moves in the planner to reach their completion. Sudden stops are simply not machanically practical, and we have no facility at the level of the planner or stepper classes to do gradual stops.

lukasradek, 2023-04-03:
To be honest, when my nozzle is digging into the bed because of probe failed to trigger (have seen this a lot in forums), I would kill for a stop button and I couldn't care less if steps were skipped or if the decelaration was abrupt (because usually the motion is already stopped by the crash, but the steppers are still trying to break the printer).

Maybe I did not present the use case clearly... completely real scenario, which happened (not only) to me many times.  
I was doing some testing with newly released 2.1.2 firmware, the printer was homing but because of misconfigured PINs, the probe did not trigger. There is about a second to react (assuming you expect the problem) before the nozzle crashes into the bed. Since my printer does not have a hardware stop button (only touch screen) I had to quickly reach for the power supply switch, which is not always at convenient spot. And since there are capacitors, that are able to drive the steppers for quite a while I ended up watching for about a second or two how my nozzle is digging into the bed absolutely helplessly (because the power supply was already off). There were already many steps lost and there was no motion to stop... it was already at standstill... just digging.

And similar things can happen with other axis too, which is especially painful on CoreXY, where is starts to pull the X axis out of square.

And for those situations I would love to have a stop button.  
I know that this can be blamed on user error, but those happen and that is why we have (emergency) stop buttons.
I know that I can add hardware stop as @thisiskeithb suggested, but software solution doesn't require any hardware purchase, any disassembly and it only costs time to implement, which I will happily do.

I just cannot see the disadvantage.