MarlinFirmware/Marlin #25615: [BUG] (bug summary)

saskot, 2023-04-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While installing Marilyn 2.1.2 with anet a8 config.
this bug accures https://prnt.sc/vBBKeQVtGnsb
it says: In file included from .pio\libdeps\sanguino1284p\LiquidCrystal\LiquidCrystal_SR1W.cpp:48:0:
.pio\libdeps\sanguino1284p\LiquidCrystal\LiquidCrystal_SR1W.h:168:1: warning: multi-line comment [-Wcomment]
 //             |    |      0.1uF            | 
[image]
I have tried different versions same outcome.
Thanks in advance.

### Version of Marlin Firmware

2.1.2

### Printer model

Anet A8

### Electronics

stock motherboard

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-04-01:
That's a warning & not a bug. You can ignore it.

closed 2023-04-01 by thisiskeithb

saskot, 2023-04-01:
Thanks but how can I continue?

thisiskeithb, 2023-04-01:
> Thanks but how can I continue?

Simply ignore it.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).