MarlinFirmware/Marlin #25618: [BUG] CoreXY machine with no Z axis would not compile

dansity, 2023-04-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a drawing machine with 2 axes only. My config always compiled but now that I have rebuilt it to coreXY I'm unable to compile.

### Bug Timeline

new issue

### Expected behavior

I expect it to compile without Z

### Actual behavior

VS code compilation errors: https://pastebin.com/km601R6S

### Steps to Reproduce

1. Download fresh copy of Marlin 2.1.2
2. overwrite config files with attached
3. Try to compile
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/11132980/config_files.zip)

### Version of Marlin Firmware

2.1.2

### Printer model

custom

### Electronics

SKR mini v2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-03:
This is easy enough to add, only 4 lines to change

```DIFF
diff --git a/Marlin/src/module/planner.cpp b/Marlin/src/module/planner.cpp
index dadb22fbde..8ef052f5db 100644
--- a/Marlin/src/module/planner.cpp
+++ b/Marlin/src/module/planner.cpp
@@ -1972,7 +1972,9 @@ bool Planner::_populate_block(
   #if ANY(CORE_IS_XY, MARKFORGED_XY, MARKFORGED_YX)
     if (da < 0) SBI(dm, X_HEAD);                  // Save the toolhead's true direction in X
     if (db < 0) SBI(dm, Y_HEAD);                  // ...and Y
-    if (dc < 0) SBI(dm, Z_AXIS);
+    #if HAS_Z_AXIS
+      if (dc < 0) SBI(dm, Z_AXIS);
+    #endif
   #endif
   #if IS_CORE
     #if CORE_IS_XY
@@ -2096,7 +2098,9 @@ bool Planner::_populate_block(
   #if ANY(CORE_IS_XY, MARKFORGED_XY, MARKFORGED_YX)
     steps_dist_mm.head.x = da * mm_per_step[A_AXIS];
     steps_dist_mm.head.y = db * mm_per_step[B_AXIS];
-    steps_dist_mm.z      = dc * mm_per_step[Z_AXIS];
+    #if HAS_Z_AXIS
+      steps_dist_mm.z      = dc * mm_per_step[Z_AXIS];
+    #endif
   #endif
   #if IS_CORE
     #if CORE_IS_XY
@@ -2244,7 +2248,7 @@ bool Planner::_populate_block(
       stepper.enable_axis(X_AXIS);
       stepper.enable_axis(Y_AXIS);
     }
-    #if DISABLED(Z_LATE_ENABLE)
+    #if HAS_Z_AXIS && DISABLED(Z_LATE_ENABLE)
       if (block->steps.z) stepper.enable_axis(Z_AXIS);
     #endif
   #elif CORE_IS_XZ
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 1062a778d8..157d5fc349 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -3276,7 +3276,7 @@ void Stepper::_set_position(const abce_long_t &spos) {
     #if CORE_IS_XY
       // corexy positioning
       // these equations follow the form of the dA and dB equations on https://www.corexy.com/theory.html
-      count_position.set(spos.a + spos.b, CORESIGN(spos.a - spos.b), spos.c);
+      count_position.set(spos.a + spos.b, CORESIGN(spos.a - spos.b) OPTARG(HAS_Z_AXIS, spos.c));
     #elif CORE_IS_XZ
       // corexz planning
       count_position.set(spos.a + spos.c, spos.b, CORESIGN(spos.a - spos.c));

```

thisiskeithb, 2023-04-03:
> thank you it builds and working

Since this works, it’d be good to keep this feature request open until code is merged.

closed 2023-04-04 by ellensp

dansity, 2023-04-05:
@thisiskeithb @ellensp 
I have further tested the changes and upon building the fw with no Z and corexy the following compilation error pops up (same config files as posted before). For me it seems to be related and only happens when there is no Z and corexy enabled
´´´
Marlin\src\gcode\motion\G2_G3.cpp: In function 'void plan_arc(const xyze_pos_t&, const ab_float_t&, bool, uint8_t)':
Marlin\src\gcode\motion\G2_G3.cpp:142:15: warning: unused variable 'part_per_circle' [-Wunused-variable]
  142 |               part_per_circle = RADIANS(360) / total_angular;                   // Each circle's part of the total
      |               ^~~~~~~~~~~~~~~
´´´´