MarlinFirmware/Marlin #25626: Latest Bugfix will not allow PROBING_MARGIN value over 20 [BUG] (bug summary)

ahmetcemturan, 2023-04-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have NOZZLE_TO_PROBE_OFFSET { 13, 26, -2 } and hence PROBING_MARGIN 26

When I tried to compile the latest bugfix, I got an error:
 TRAMMING_POINT_XY point 0 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.

Only If I set both values to { 13, 20, -2 } and 20 does it compile.

### Bug Timeline

It is existent in the latest bugfix

### Expected behavior

I expect to give a higher offset value than 20

### Actual behavior

It wont allow to set the value higher than 20

### Steps to Reproduce

Use above values to compile

### Version of Marlin Firmware

2.1.bugfix as of April 3rd

### Printer model

Prusa Clone

### Electronics

Tinybee

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11137388/Marlin.zip)

ellensp, 2023-04-03:
you have #define PROBING_MARGIN 26  this is a value you set, takes 26mm off all edges of the bed.
It is not related to any other setting. this is something you have chosen to define.
 
#define X_BED_SIZE 235
#define Y_BED_SIZE 215
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE

with #define NOZZLE_TO_PROBE_OFFSET { 13, 26, -2 }

Ie probe is right of the hotend 13mm in x and back of hotend 26mm in Y

so you have told marlin the probe can reach 
X min probable position = X_MIN_POS + 13 + PROBING_MARGIN = 39
Y min probable position =  X_MIN_POS + 26 + PROBING_MARGIN = 52

you have 
  #define TRAMMING_POINT_XY { {  20, 20 }, { 180,  20 }, { 180, 180 }, { 20, 180 } }

Again. 
X min probable position is 39, 
Y min probable position is 52.

first point { <span style="color: red;">20</span> , 20 }   is not reachable
Second point  { 180,  20 } is not reachable
Forth point { 20, 180 }  is not reachable

The closest you can get is 

#define TRAMMING_POINT_XY { {  39, 59 }, { 180,  59 }, { 180, 180 }, { 39, 180 } }

Your PROBING_MARGIN are very large. 

A probing margin is normally only if you probe requires to be over the bed a bit more, such as an inductive probe. in wich case it should be set to about half the diameter of the probe 

There is no bug here.

closed 2023-04-03 by ellensp

ellensp, 2023-04-03:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ahmetcemturan, 2023-04-03:
Thank You,  
I was actually looking for (#define TRAMMING_POINT_XY ) şn configuration file as the (PROBING_MARGIN) is there but couldn't find it.   Also it made me think of a bug as I basically just transferred my settings to the new Bugfix...  So it just couldn't be my fault.. It had to be a BUG... :)  Sorry for that..  (Also that TinyBee board seems to have a lot of problems with Marlin...)

I set  #define TRAMMING_POINT_XY { {  26, 13 }, { 209,  13 }, { 209, 202 }, { 26, 202 } }  and compiled now.