MarlinFirmware/Marlin #25628: Ender 3v2 DWIN Clone Display with Marlin UI - Refresh issue

weiaoli2, 2023-04-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using Marlin UI with the DWIN Ender 3v2 Display (on the Bigtreetech SKR E3 V3.0.1) does not perform refresh in all number areas. Only when switching back and forward to the menu the refresh will be performed. 

Display Firmware has been updated. 
Tested with Firmware 2.1 and 2.1.2. 
Both Marlin UIs (Portrait and Landscape (yes I know Landscape is not supported by the DWIN clone) have the same issues. 

DWIN_CREALITY_LCD_JYERSUI is working. So I assume the issue is not in the Display Firmware but in the UI of Marlin. 

### Expected behavior

If a number is changing anywhere on the screen, the screen must perform a refresh

### Actual behavior

Refresh is not performed at all. This wrong behavior only applies to numbers. It doe snot apply to graphical items, such as Menu selections.

### Steps to Reproduce

Use Marline UI on Ender 3v2 DWIN clone display (which was the original display in my ender 3v2). 

### Version of Marlin Firmware

2.1.2

### Printer model

Ender 3v2

### Electronics

Bigtreetech SKR MINI E3 V3.0.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Screenshot 2023-04-03 at 23.40.23.png.zip](https://github.com/MarlinFirmware/Marlin/files/11140067/Screenshot.2023-04-03.at.23.40.23.png.zip)
[Screenshot 2023-04-03 at 23.40.41.png.zip](https://github.com/MarlinFirmware/Marlin/files/11140068/Screenshot.2023-04-03.at.23.40.41.png.zip)
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/11140081/Config.zip)

The-EG, 2023-04-03:
This clone may not actually support the `bShow` parameter of `DWIN_Draw_String`: <https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/common/dwin_api.cpp#L236>

Which clone is particular is it? There are at least 3 now: <https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2#display-update-guide-for-ender-3-v2-family-screens>

Note: don't just re-enable the `DWIN_Draw_Rectangle` hack there in `DWIN_Draw_String`, this breaks other things that rely upon being able to show text without a solid background.

weiaoli2, 2023-04-05:
> This clone may not actually support the `bShow` parameter of `DWIN_Draw_String`: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/common/dwin_api.cpp#L236
> 
> Which clone in particular is it? There are at least 3 now: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2#display-update-guide-for-ender-3-v2-family-screens
> 
> Note: don't just re-enable the `DWIN_Draw_Rectangle` hack there in `DWIN_Draw_String`, this breaks other things that rely upon being able to show text without a solid background.

Hi EG, sorry to leave of the detail: it is: DACAI display - and I use the private folder for the Display Firmware. 
The display came with my stock ender 3v2. 
The issue only happens with the Marlin UI - all other UI work perfect in Marlin 2.1.2 (and 2.1).

JonasV328, 2023-04-27:
I have the same problem with Marlin 2.1.2 on my Ender 3 V2, with stock DACAI display and stock Creality 4.2.2 mainboard. Unfortunately, as I would like to use custom menu items, switching to a different UI is not a good solution for me.

If anyone who doesn't have the printer but knowledge about the code could give any advice, that would be great! Thank you!

The-EG, 2023-04-27:
First see if this is still happening in current bugfix-2.1.x. If so, remove the `IS_DWIN_MARLINUI` from [this line](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/common/dwin_api.cpp#L237) and change the following line to only draw the rectangle if `bShow` is true. ie.

```c++
 #if NONE(DWIN_LCD_PROUI, DWIN_CREALITY_LCD_JYERSUI)
    if (bShow) DWIN_Draw_Rectangle(1, bColor, x, y, x + (fontWidth(size) * strlen_P(string)), y + fontHeight(size));
  #endif
```

If that works then it would confirm the DACAI displays do not fully support the DWIN protocol.

JonasV328, 2023-04-27:
> First see if this is still happening in current bugfix-2.1.x.

Yes, it still happens in today's bugfix-2.1.x.

After changing the two lines as you suggested the problem is fixed and all values including the status bar at the info screen's bottom are displayed correctly.

I tried the same "fix" in the release 2.1.2 version, but although it compiled, the printer does not work (blank screen) properly with that.

Thank you!

The-EG, 2023-04-27:
It's more of a hack than a fix. It'd probably be best to have that optionally included if the user has the DACAI display. At least you can have a functioning display for the moment, though.

Lord-Pieter, 2023-08-25:
I also had the refresh issue with the Creality Ender 3 V2 using the DWIN_CREALITY_LCD_JYERSUI option. Control board is V4.2.2. Found out that I have the DACAI display. Tried the fix with a small modification leaving out DWIN_CREALITY_LCD_JYERSUI  from the #if condition:

 #if NONE(DWIN_LCD_PROUI)
    if (bShow) DWIN_Draw_Rectangle(1, bColor, x, y, x + (fontWidth(size) * strlen_P(string)), y + fontHeight(size));
  #endif

Now the display works fine. Even the status bar showing the xyz coordinates with the flashing -?- now work well. So this fix is very needed for DACAI displays. Hopefully this fix can be added to future releases permanently.

closed 2023-11-05 by github-actions[bot]