MarlinFirmware/Marlin #25640: M112 Shutdown: Printer Halted right before purge line start

SpannMagoo, 2023-04-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am not sure when this started as I have been using the BF from 3-18 which no issue and when I tried Tom's PR fixes (#25553), the Bug Fix Build on 4/1 and 4/2. I get an M112 error on occasion right before the purge line is about to start. I do not get it when I go back to my 3-18 bug fix build or before.

[image]

### Bug Timeline

See above

### Expected behavior

Expected it to complete purge line and print the loaded gcode.

### Actual behavior

M112 Shutdown error

### Steps to Reproduce

See #25553 for sample gcode and configs. This does not happen ever print. When I tested Toms PR it was 1 out of 3. When I tested the two bug fixes it was about the same rate.

### Version of Marlin Firmware

2.1.2 Bug Fixes

### Printer model

Ender 3

### Electronics

Creality 4.2.7

### Add-ons

BLTouch, Direct Drive, Dragonfly BMS(no change to thermistor or heater cartridge), Dual Z

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

When I had the errors I was using SD card. I did have Octoprint running on at least 2 of the 3 times I got the m112 Error. I am pretty sure on the last one I did not have Octoprint running. I did not include files as both sets of configs and gocodes are on #25553

SpannMagoo, 2023-04-09:
Newest configs from 04-05 bugfix and gcode
[configs and gcode.zip](https://github.com/MarlinFirmware/Marlin/files/11186465/configs.and.gcode.zip)

thinkyhead, 2023-04-09:
"OctoPrint will be sending an `M112` command to shut the printer down in response to an error reported.
Find the previous lines in the terminal to see what error & why OctoPrint has shut it down."

You can also configure OctoPrint to not do a hard shutdown in response to seeing the word "error," which might allow Marlin to report the issue and continue.

SpannMagoo, 2023-04-10:
I'll have to find that in Octoprint and see what happens.

SpannMagoo, 2023-04-11:
What processor does your 4.2.7 board have? (Mine has the STM32F103RET6.)
- Same, its for sure the 512 version
What version of OctoPrint do you have? (I have 1.7.3).
- Whatever the latest version is
Elsewhere you have posted the OctoPrint log and there was very little in it. OctoPrint can suppress all kinds of log messages. Did you have any messages suppressed?
- Not that I am aware of, I did just turn off the error handling last night, but haven't tested.
Can you confirm for definite that this can happen without OctoPrint connected? We need more than being pretty sure, we need a definite observation to rule out OctoPrint sending M112.
- I can only confirm it happening with Octoprint, I thought I had one without, but cant be for sure as its only happening with Octoprint.
Does the bug happen if you do not run G28 and G29 at the start of the print? (Since I only have a bare board I have had to comment these out for the code.)
- Havent tried this, but these seem to work fine and the error is tossed once its at home on its final warm up tor the purge.
How do you start your prints? I am guessing it's from the LCD. Is that correct? Does the bug occur if you start the prints from the OctoPrint UI?
- It was from SD when the last few happened. I will have to try it from Octoprint later.

Should I turn on error handling again?

tombrazier, 2023-04-11:
> Should I turn on error handling again?

Try with both and see what happens. The more info we get on this the better.

SpannMagoo, 2023-04-11:
Sounds good. Next update in roughly 9 hours. I am at work.

SpannMagoo, 2023-04-12:
3 prints now and I all of a sudden can't replicate. The BLTouch bug still happened.

tombrazier, 2023-04-13:
> 3 prints now and I all of a sudden can't replicate.
Is that after switching the M112 behaviour off in OctoPrint?

SpannMagoo, 2023-04-13:
I had turned the error handling back on. Unfortunately, I ran out of time for testing and haven't been able to turn on printer last 2 days since to run a few more.

SpannMagoo, 2023-04-16:
Quick FYI. I have not ce back to this and appear to be only with it. You can close. I do not have time to chase this one sadly. I've been good since going back to 2.1.2 release version.

closed 2023-04-17 by thinkyhead