MarlinFirmware/Marlin #25641: [BUG] SKR 1.4 TMC2209 TURBO not compiling with CR-10s configuration files

joshlacr43156, 2023-04-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello everyone, 

This is my first time posting attempting to get help so if there is any information I have left out I am very sorry and will happily provide it. 

I am currently swapping out my old creality v1.2 motherboard for the SKR 1.4 TMC 2209 TURBO in my Creality CR-10s and have hit a major road block trying to compile the firmware. 
I have changed the "default_envs =  LPC1769"
When compiled VS code gives this error (full error log at the end): 

(user)\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory

I tried deleting "toolchain-gccarmnoneeabi" and recompiling, which resulted in it successfully downloading and reinstalling toolchain-gccarmnoneeabi, and then giving the same no such file or directory error. 

I have tried bugfix version 2.1 of marlin, 2.0, even tried an earlier version and they all give the same error. 

I tried compiling with configuration files for a different board and compiled perfectly, leading me to believe there is a problem with the configuration files for this board for CR-10s.

I am at a complete loss and would be incredibly grateful for any help or guidance. Thank you in advance!

Here is the complete terminal output:

PS C:\Users\joshw\Downloads\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x> platformio run --silent -e LPC1769
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\array:38,
                 from C:\Users\joshw\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:26,
                 from Marlin\src\HAL\LPC1768\MarlinSerial.h:24,
                 from Marlin\src\HAL\LPC1768\MarlinSerial.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o] Error 1
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,  
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
In file included from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\stl_pair.h:59,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\utility:70,
                 from c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\algorithm:60,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:33,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\MinSerial.cpp:24:
c:\users\joshw\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\11.3.1\bits\move.h:33:10: fatal error: bits/c++config.h: No such file or directory        
   33 | #include <bits/c++config.h>
[... 99 lines cut ...]
Try to compile the configuration files.

### Version of Marlin Firmware

bugfix 2.1

### Printer model

Creality CR-10s

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-06:
From discord chat.
"it appears platformio, or whoever controls it, have released a broken toolchain-gccarmnoneeabi package, if errors come up with bits/os_defines.h you will need to add  toolchain-gccarmnoneeabi@1.100301.220327 to the platform_packages"

thisiskeithb, 2023-04-06:
See also: https://github.com/platformio/platformio-core/issues/4590#issuecomment-1497847597

closed 2023-04-06 by thisiskeithb