MarlinFirmware/Marlin #25645: [BUG] do_blocking_move_to_xy in home_z_safely prevents probing full distance to bed with G28 (all axis)

flukejones, 2023-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

From what I can tell everything is working correctly hardware side. And doing a single Z home works repeatedly well.

The problem here is nested in `Z_SAFE_HOMING` being enabled - the first try of G28 (no args) will home fine, but further calls to it result in the nozzle moving down what looks like `Z_HOMING_HEIGHT`, then pulling back up.

In the `home_z_safely` function if I comment out the `do_blocking_move_to_xy` call I get the nozzle moving the full distance down for bed contact.

### Bug Timeline

Some time after 38e633023603 ?? Unsure

### Expected behavior

G28 without args and `Z_SAFE_HOMING` should home XY then move to center and home Z. Repeated G28 calls should result in the nozzle centering after XY home and travelling the full distance down to the bed for contact.

### Actual behavior

The first call to G28 with `Z_SAFE_HOMING` enabled will home Z in the center after XY successfully but further calls to G28 result in the nozzle travelling to center after XY and not going the full distance down to the bed for contact (only travelling a few mm then pulling up).

### Steps to Reproduce

1. enable `Z_SAFE_HOMING` with a piezo setup
2. home with G28 (no args)
3. home with G28 (no args)

### Version of Marlin Firmware

bugfix-2.1.x @ 7e700c139485

### Printer model

Custom with biqu SKR mini E3 v3, hemera XS, piezo probe

### Add-ons

Trianglelab piezo sensor board

### Bed Leveling

No Bed Leveling

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11170114/Configuration.zip)

## G28 all axis, second try that fails to travel full Z distance to bed

READ: >>> G28  X112.0000 Y112.0000 Z3.0000
READ: Machine Type: Cartesian
READ: Probe: FIX_MOUNTED_PROBE
READ: Probe Offset X0.0000 Y0.0000 Z0.0000 (Aligned With & Same Z as Nozzle)
READ: Auto Bed Leveling: BILINEAR (disabled)
READ: >>> set_bed_leveling_enabled  X112.0000 Y112.0000 Z3.0000
[... 165 lines cut ...]
READ: <<< do_blocking_move_to  X112.0000 Y112.0000 Z0.1500
READ: <<< homeaxis(Z)
READ: <<< home_z_safely  X112.0000 Y112.0000 Z0.1500
READ: do_z_clearance(3.0000, 1)
READ: do_blocking_move_to_z(3.0000, 8.0000)
READ: >>> do_blocking_move_to  X112.0000 Y112.0000 Z0.1500
READ: >  X112.0000 Y112.0000 Z3.0000
READ: <<< do_blocking_move_to  X112.0000 Y112.0000 Z3.0000
READ: current_position= X112.0000 Y112.0000 Z3.0000 : sync_plan_position
READ: X:112.0000 Y:112.0000 Z:3.0000 E:0.0000 Count X:8736 Y:8736 Z:1194
READ: <<< G28  X112.0000 Y112.0000 Z3.0000

---

My probe pins changed after the log above, so the configs will reflect that. Same results however.

If I turn the trigger on the piezo way up so it doesn't trigger at all then the nozzle travels the full distance down to the bed. When I check the piezo board pins with multimeter I see open-line until triggered as expected.

---

**Just tried this**

I replaced the line here https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/gcode/calibrate/G28.cpp#L147  with `do_blocking_move_to_xy(destination.x, destination.y, 100.0);` and this solved my issues with homing. If I change `100.0` to anything higher like 200, then I get the issues described above. So I don't know... maybe the...

Yeah changing to `#define XY_PROBE_FEEDRATE (100*60)` also solved the issue. Seems likely that the piezo probe board is getting noisy  from the fast move - it sure isn't showing up on the LED on the board but those seem slow to activate so maybe something is triggering before it should..

thisiskeithb, 2023-04-06:
Give https://github.com/MarlinFirmware/Marlin/pull/25631 a try. There have been a rash of probing/leveling features that have been broken for the past month and that PR aims to resolve all of them.

flukejones, 2023-04-18:
I will close this as the issue is fully related to the piezo sensor - it occasionally requires some tweaking to recalibrate.

closed 2023-04-18 by flukejones