MarlinFirmware/Marlin #25646: [BUG] (Unable to compile for CR-10S BigTreeTech SKR Mini E3 2.0 using 2.1.2 + current Bugfix and example configuration)

svde, 2023-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin doesn't compile.

### Expected behavior

I expect that the example configuration files for releases (and bugfixes) compile.

### Actual behavior

It doesn't compile.

### Steps to Reproduce

1. Clone Configurations & Marlin:
```
rm -rf Configurations_bugfix-2.1.x ; git clone -b bugfix-2.1.x https://github.com/MarlinFirmware/Configurations.git Configurations_bugfix-2.1.x --single-branch
rm -rf Marlin_bugfix-2.1.x ; git clone -b bugfix-2.1.x https://github.com/MarlinFirmware/Marlin.git Marlin_bugfix-2.1.x --single-branch
```
2. Symlinked Marlin_bugfix-2.1.x to Marlin
```
ln -s Marlin_bugfix-2.1.x Marlin
```
3. symlinked configuration files as follows:
```
cd Marlin/config
ln -s ../../Configurations_bugfix-2.1.x/config/examples/Creality/CR-10S/BigTreeTech\ SKR\ Mini\ E3\ 2.0/Configuration.h
ln -s ../../Configurations_bugfix-2.1.x/config/examples/Creality/CR-10S/BigTreeTech\ SKR\ Mini\ E3\ 2.0/Configuration_adv.h
cd ../..
```
4. clean & compile
```
platformio run --target clean -e genericSTM32F103RC; platformio run -e genericSTM32F103RC
```

```
❯ platformio run -e genericSTM32F103RC 2>&1 | grep -i error
warning: Calling missing SConscript without error is deprecated.
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../pins/ramps/env_validate.h:31:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
   31 |   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
... many many of these lines, dpulicates removed ...
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/DUE/fastio/G2_PWM.cpp.o] Error 1
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../pins/ramps/env_validate.h:31:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
   31 |   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../pins/ramps/env_validate.h:31:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
   31 |   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560 or 1280' in 'Tools > Board.'"
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
... many many of these lines, dpulicates removed ...
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/LPC1768/tft/xpt2046.cpp.o] Error 1
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/LPC1768/tft/tft_spi.cpp.o] Error 1
```

### Version of Marlin Firmware

trying to compile 2.1.2 or bugfix-2.1.x

### Printer model

Creality CR-10S

### Electronics

BigTreeTech SKR Mini E3 2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration.h and Configuration_adv.h are stock (2.1.2 and bugfix-2.1.x) both don't compile, same result.

[env:genericSTM32F103RC]
platform    = ststm32
framework   = arduino
board       = genericSTM32F103RC
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__

EvilGremlin, 2023-04-06:
user error: `cd Marlin/config`, you've put configs in wrong place

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

svde, 2023-04-06:
> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

I think you're mistaken with marking this as user error.

svde, 2023-04-06:
OK, thank you, so Marlin/Marlin in my case.

It still fails to build.

❯ platformio run -e genericSTM32F103RC 2>&1 | grep -i error
warning: Calling missing SConscript without error is deprecated.
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../HAL/./STM32/HAL.h:73:6: error: #error "SERIAL_PORT must be from 1 to 6."
   73 |     #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:114:56: error: template argument 2 is invalid
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:137:47: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:139:55: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:149:37: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:153:56: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:158:39: error: request for member 'println' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:162:43: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:165:53: error: request for member 'println' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:168:45: error: request for member 'flush' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:169:45: error: request for member 'flushTX' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:272:19: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/DUE/fastio/../../../inc/../core/serial.h:279:70: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/DUE/fastio/G2_PWM.cpp.o] Error 1
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../HAL/./STM32/HAL.h:73:6: error: #error "SERIAL_PORT must be from 1 to 6."
   73 |     #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../HAL/./STM32/HAL.h:73:6: error: #error "SERIAL_PORT must be from 1 to 6."
   73 |     #error "SERIAL_PORT must be from 1 to 6."
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:114:56: error: template argument 2 is invalid
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:137:47: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:139:55: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:149:37: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:153:56: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:158:39: error: request for member 'println' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:162:43: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:165:53: error: request for member 'println' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:168:45: error: request for member 'flush' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:169:45: error: request for member 'flushTX' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:272:19: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
[... cut ...]
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:272:19: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
Marlin/Marlin/src/HAL/LPC1768/tft/../../../inc/../core/serial.h:279:70: error: request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/LPC1768/tft/xpt2046.cpp.o] Error 1
*** [.pio/build/genericSTM32F103RC/src/Marlin/src/HAL/LPC1768/tft/tft_spi.cpp.o] Error 1

EvilGremlin, 2023-04-06:
Looks like you still doing something wrong becasue it's still default conf with SERIAL_PORT 0. Or pio is slow on uptake with caching.

closed 2023-04-06 by thisiskeithb

thisiskeithb, 2023-04-06:
> platformio run --target clean -e genericSTM32F103RC; platformio run -e genericSTM32F103RC

`genericSTM32F103RC` is not correct for this board. You should be using one of these environments:

```prolog
Use one of these: STM32F103RC_btt, STM32F103RC_btt_USB, STM32F103RE_btt, STM32F103RE_btt_USB, STM32F103RC_btt_maple, STM32F103RC_btt_USB_maple, STM32F103RE_btt_maple, STM32F103RE_btt_USB_maple
```

Compiling with `STM32F103RC_btt` without any other changes to configs works as expected:

```prolog
Linking .pio/build/STM32F103RC_btt/firmware.elf
Checking size .pio/build/STM32F103RC_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  26.8% (used 13152 bytes from 49152 bytes)
Flash: [=======   ]  67.1% (used 175912 bytes from 262144 bytes)
Building .pio/build/STM32F103RC_btt/firmware.bin
===================================== [SUCCESS] Took 52.86 seconds =====================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  SUCCESS   00:00:52.858
====================================== 1 succeeded in 00:00:52.858 ======================================
 *  Terminal will be reused by tasks, press any key to close it.
```