MarlinFirmware/Marlin #25647: [BUG] Ender 3 Pro + SKR Mini E3 v1.2 screen doesn't work

landgraf, 2023-04-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Somewhere between released version (02010200) and 4aec74f246 screen stopped worked. 
My configuration: Ender 3 with SKR mini E3 v1.2 board and factory screen.

Reproducible with Marlin revision 4aec74f246 and config (Configurations/config/examples/Creality/Ender-3\ Pro/BigTreeTech\ SKR\ Mini\ E3\ 1.2/) revision d208d4d7cbd0e6b765e9e3208d6257da8496416e without any modifications

Blank screen, no beep.

### Expected behavior

Working screen :)

### Actual behavior

Empty screen

### Steps to Reproduce

build marlin with example configuration and flash

### Version of Marlin Firmware

4aec74f246

### Printer model

ender 3 pro

### Electronics

skr mini e3 v1.2

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-07:
Since you have a known good and bad commit, please do a git bisect to identify the commit that is responsible

landgraf, 2023-04-07:
> Since you have a known good and bad commit, please do a git bisect to identify the commit that is responsible
Of course I'll do but a bit later, my 3d printer is busy next few days...

thisiskeithb, 2023-04-07:
Which environment are you building with?

All of them work fine for me using today's bugfix (f0f5cd4) on an SKR Mini E3 V1.2 with the stock screen (`CR10_STOCKDISPLAY`) with the default configs from [`/config/examples/Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 1.2/`](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%201.2) (including `*_maple` environments):

[image]

Here's a `firmware.bin` built with `STM32F103RC_btt_USB`: [STM32F103RC_btt_USB-firmware_bin.zip](https://github.com/MarlinFirmware/Marlin/files/11181744/STM32F103RC_btt_USB-firmware_bin.zip) for reference.

ellensp, 2023-04-08:
There have been a bunch of non working 128x64 mono lcd reports of late.  Non of which where reproducible.
Deleting  directory .platformio to force a fulll platformio reinstall and rebuilding marlin seems to fix it,

thisiskeithb, 2023-04-11:
Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

closed 2023-04-11 by thisiskeithb

landgraf, 2023-04-11:
Right, something went wrong with vscode/platformio cache and any compiled firmware stopped working.
I've deleted .pio and .platformio  and compiled marlin without vscode involved and it boots and works as expected.
Sorry for the false alarm.