MarlinFirmware/Marlin #25653: Undefined XY axis movement during printing

DAVEOOH, 2023-04-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

During printing, Unpredictably x or y axis make a slow movement in a wrong direction. The movement takes place slowly. the nozzle then moves back to the original position and printing continues. Sporadically, the unit also stops during printing and the software restarts. Tried version from 2.1.1 to 2.1.3 with the example config files.
Same problem with cura and prusa sliced object.

The error does not occur when printing via cura over usb !!!

The following settings seem to make the problem occur less frequently:
//#define BINARY_FILE_TRANSFER
 #define SDCARD_READONLY   
 #define SD_SPI_SPEED SPI_HALF_SPEED

Thanks Michael

### Bug Timeline

2.0.93 fromTrey Wentz worked. Versions since 2.1 have the this problem

### Actual behavior

See bug details

### Steps to Reproduce

Occurs in a print of 30-40 min so far always.
Sometimes in first minutes. Sometimes to the end.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11182751/Configuration.zip)
[MArlin 2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/11182760/MArlin.2.1.x.zip)

### Version of Marlin Firmware

2.1.x

### Printer model

Elegoo Neptune 2S

### Electronics

Stock

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

DAVEOOH, 2023-04-15:
Get a working 2.1.1 version.  Versions after that doesn´t work.

DAVEOOH, 2023-04-19:
I am sure there is something wrong with sd "support".
In some versions of marlin (after 2.0.9.5) the printer restarts by starting a print from sd card or even opening the menue to show the files.
The problem is not present when printing via USB with Cura or beagle cam. 

Any ideas if i can tweak something in config ?

classicrocker883, 2023-04-30:
have you tried changing `BLOCK_BUFFER_SIZE`? try 32, as well as 32 for `BUFSIZE` and see about increasing the others.

u can also try making `DEFAULT_MINSEGMENTTIME` 50000
and `SLOWDOWN_DIVISOR` 4

if you have octoprint you can run a debug diag which will show what is going on.

DAVEOOH, 2023-05-12:
Tried the settings from @classicrocker883. But still the same problem.

classicrocker883, 2023-05-13:
here are my typical settings [Marlinconfigs.zip](https://github.com/MarlinFirmware/Marlin/files/11469414/Marlinconfigs.zip)
I would say youre on the right track for it being SD card related if there is no problem printing over USB. try using a different SD card? anyway, what size are you using? apparently these machines dont like an SD card that is too big. and it should be formatted to something like fat32 and 4096 cluster size.

try enabling `MEATPACK_ON_SERIAL_PORT_1` `FASTER_GCODE_PARSER`, and disable `BINARY_FILE_TRANSFER`

 I cant comment on your specific machine and lcd, but this seems more like a **YOU** problem - no offense or disrespect.

classicrocker883, 2023-07-19:
@DAVEOOH have you solved this? maybe you can help out @phobos79

that is strange this happened both on Neptune 2.

if I had to guess it probably has something to do with the SD card/card port. for instance, have you tried a smaller SD card like 8gb? what about formatted to 4096k cluster size or whatever the manufacturer recommends?

how about printing via USB such as from raspberry pi / octoprint, Pronterface, or sent from Cura or another Slicer?

phobos79, 2023-07-23:
Hello, i'm using a new sd card, with a fat partition of 512mb and 4k clusters,

no way, 'im trying to compile with the TFT_LVGL_UI  too but still working on compiler errors

 sometime also add another behaviour, printer do the g28 twice

classicrocker883, 2023-07-23:
im not sure what you mean by 
>printer do the g28 twice  
but if you need help with compiler errors shoot, ill try to help if i can

phobos79, 2023-07-24:
Hello, it's the g29 that runs twice..a really strange and random behaviours, it does g29, after prime the extruder..and start again with the g29 (sometime do 3 times but rarely), this is a superrandom behaviour, i cannot find a pattern on which it happens.

yes, i'm trying to use the  TFT_LVGL_UI  interface like daveooh to check if it works, i've this error on compiler "Marlin\src\module\settings.cpp:3330:74: error: 'DISPLAY_SLEEP_MINUTES' was not declared in this scope"

but at row 1822 of config advanced i've "#define DISPLAY_SLEEP_MINUTES 10  // (minutes) Timeout before turning off the screen. Set with M255 S.
"

i'f a replace at settings.cpp the variable name with an int ( 5 for example) compiler still failing but only with some warning

so i don't know

classicrocker883, 2023-07-25:
the G29 running more than once can maybe be something with the code. I'm not familiar with the tft menus. but it can also be like sending the code several times by accident. say it sends it and doesn't hear back, it will send again. so maybe it needs a delay?

as for display sleep minutes in settings, I can look into that. that may be easier for me to fix.

if you don't kind posting your config files? if like to recreate this error so I know if it is fixed.

phobos79, 2023-07-25:
Hello, my config files are in the second post.

i solved, i think, version 2.1.2 and 2.1.2.1 had a lot of problems on my printer, regardless of sd card, the are: random movement in x and y, G29 looping, random freeze of the printer and weird behaviour on M600

i took the 2.1.1.1 and tested it, it worked fine (i printed the same gcode on the same card that give problems) but i cannot not having the input shaping to i took (a lot of work) all code reference of input shaping (M593) from 2.1.2.1 and "transplanted" in 2.1.1.1 ..it took half day but now i've the working 2.1.1.1 plus the input shaping (i assume it work considering that the printer answer and interact with M593 command)

if someone has the same problem i can give the folder, ask me.

classicrocker883, 2023-07-25:
as for settings.cpp, if you have a touch_screen, simply go to **src/inc/Conditionals_LCD.h** line ~1816 and comment out 
` #undef TOUCH_SCREEN`

so it is like 
```y
  #if !HAS_GRAPHICAL_TFT
   // #undef TOUCH_SCREEN
```

Sortexx, 2023-07-26:
here is another one with a neptune 2 with exactly the same problem

ellensp, 2023-07-26:
@Sortexx please attach Configs, so we can see any similarities and or differences

classicrocker883, 2023-07-26:
@thinkyhead @tombrazier  hey Scott, Tom, would this issue be related to the <ins>[stepper noise issue over at my repo issues](https://github.com/classicrocker883/MriscocProUI/issues/27) </ins>?

it seems like the same or similar thing because v2.0.9+ from Alex's FW (alexqzd/Marlin) and Earlier this year's *Marlin bugfix* worked fine until the more recent **bugfix-2.1.x**, basically I have a hunch these issues seem to be related in some way, since they are both stepper related, and the versions kind of point to each other.

---
I want to also point out that when I tried compiling using the *Neptune 2* configurations, I did get this odd error in **settings.cpp**
```c
ui.sleep_timeout_minutes = TERN(TOUCH_SCREEN, TOUCH_IDLE_SLEEP_MINS, DISPLAY_SLEEP_MINUTES);
```
even though TOUCH_SCREEN was enabled...  
in **src/Conditionals_LCD.h** line ~1816
```c
  #if !HAS_GRAPHICAL_TFT
    #undef TOUCH_SCREEN
```
yet `MKS_ROBIN_TFT35` is defined for the display, does that count as `HAS_GRAPHICAL_TFT`?

it `#undef TOUCH_SCREEN` which causes that error in **settings.cpp**

in conclusion, **src/Conditionals_LCD.h** should be edited to include `MKS_ROBIN_TFT35` as `HAS_GRAPHICAL_TFT`

phobos79, 2023-07-27:
Hello, thanks for support about compiling the other UI, but as i write over, now all work well so it's no more needed.

Sortexx, 2023-07-27:
I still have problems, I have tried with the latest bugfix, with 2.1.1.1, and with the latest versions and I still have the same problem.

DAVEOOH, 2023-07-27:
Switching to TFT_CLASSIC_UI solved the problem for me.

Sortexx, 2023-07-27:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/12187520/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/12187522/Configuration_adv.txt)

phobos79, 2023-08-03:
> I still have problems, I have tried with the latest bugfix, with 2.1.1.1, and with the latest versions and I still have the same problem.

Try this one: https://github.com/MarlinFirmware/Marlin/tree/lts-2.1.1.
it work flawlessly

it lack the input shaping, but if you want to have it i can help you

tombrazier, 2023-08-13:
> @thinkyhead @tombrazier hey Scott, Tom, would this issue be related to the [stepper noise issue over at my repo issues](https://github.com/classicrocker883/MriscocProUI/issues/27) ?

Reading the thread this looks like it is related to motors stuttering and something to do with SD card. So no, it does not seem like the same thing.

closed 2023-10-24 by github-actions[bot]