MarlinFirmware/Marlin #25657: [FR] Automatic Z offset calibration

aero4, 2023-04-08:
### Is your feature request related to a problem? Please describe.

I have swappable hotend blocks based on a Chimera/Cyclops hotend.
The problem is, that the individual hotend block heights can vary with up to 1mm and they can't be plugged in exactly enough.
Therefore **the nozzle to probe offset is different** after every hotend swap.
This means I have to use babystepping to compensate this offset variations after every swap and have to restart the print multiple times to get a good first layer.

### Describe the feature you want

For the reasons described above, an _automatic Z offset calibration_ (nozzle to probe offset) would be a very usable feature for me.
A probe could be mounted on the heatbed and the nozzle be lowered at the same position as the probe is located until the probe triggers. If this is done after homing, the **difference between the homing z height and the probed nozzle height can be used to calculate the nozzle to probe offset before every print**.
This would eliminate the manual compensation of the nozzle to probe offset via babystepping.

As a probe a strain gauge cantilever or even a switch with a metal pin can be used.

### Additional context

Klipper, which I don't want to use, seems to have such an automatic Z offset calibration.
More informations can be found in this article at all3dp: [link](https://all3dp.com/2/klipper-z-offset-simply-explained/)

DerAndere1, 2023-04-09:
For your setup, the manual tool change feature (https://github.com/MarlinFirmware/Marlin/pull/23575) would be an appropriate solution. You would configure your firmware for multiple hotends and assign a tool index for each hotend. Then once set the `NOZZLE_TO_PROBE_OFFSET` for hotend 0. Then you need to calibrate the `HOTEND_OFFSET_Z` for each hotend, e.g. by using command `M218`. So what we actually need is is automatic tool offset calibration. The company Haas uses G-code G37 for this,

DerAndere1, 2023-04-09:
There is an open source solution for automatic tool offset calibration: TAMV (https://github.com/HaythamB/TAMV). It just requires proper implementation of G-code G10 as an alias for M218 (https://github.com/MarlinFirmware/Marlin/pull/24608)

thisiskeithb, 2023-04-09:
Closing since we have an open PR:

- https://github.com/MarlinFirmware/Marlin/pull/23575

closed 2023-04-09 by thisiskeithb

aero4, 2023-04-09:
> For your setup, the manual tool change feature (#23575) would be an appropriate solution. You would configure your firmware for multiple hotends and assign a tool index for each hotend. Then once set the `NOZZLE_TO_PROBE_OFFSET` for hotend 0. Then you need to calibrate the `HOTEND_OFFSET_Z` for each hotend, e.g. by using command `M218`. So what we actually need is automatic hotend offset / tool offset calibration. The company Haas uses G-code G37 for this,

Looks like a good idea, but this is unfortunately not suitable for my problem, because even after inserting the same hotend the offset can vary due to lack of accuracy when inserting the hotend. So, a autmatical calibration would be the only solution for my problem like the one the Voron 3D printer already have.

> There is an open source solution for automatic tool offset calibration: TAMV (https://github.com/HaythamB/TAMV). It just requires proper implementation of G-code G10 (#24608)

This solution is very advanced and needs significant processing power which I am not planning to implement. This would be over engineering and too expensive for a way simpler solution like a probe switch just because the software part is missing...

As there seems to be too little demand for others for this feature I will implement that as my own G-code on my printer...

DerAndere1, 2023-04-18:
It is under-engineering if your tools cannot be attached in a repeatable position

aero4, 2023-04-18:
So, the Voron 3D printer is also under engineered if it uses this kind of automatic compensation? And UBL would also not be necessary if you view it in that way. Also when buying V6 throats legnth tolerances are within the range of millimeters which I can't influence... Only a few people use swappable hotends which makes this FR not very popular.

DerAndere1, 2023-04-18:
what I meant: it is under-engineered if your nozzle-to-probe offset for tool 0 varies. It is totaly normal to have tools with different lengths. But only the tool length offset or hotend offset is a variable. Here is how tool holders with low tolerance look like: https://portlandcnc.com/blog/2021/5/how-to-setup-an-er32-cnc-tool-holder

aero4, 2023-04-19:
I don't get it. In the video you linked to the CNC machine compensates the tool offset: https://youtu.be/SmNj51O6uuI?t=137
This is exactly what the FR is about? It even uses the same technique I suggested (mechanical switch at the corner of the machining area)...