MarlinFirmware/Marlin #25658: tool head crashes into the bed trying to deploy mag mounted Z probe

yassinelessawy, 2023-04-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

tool head crashes into the bed trying to deploy/stow mag mounted Z probe during G34 and bed leveling. deploying and sowing the probe works fine from the LCD and During Z Calibration via Z Probe offset wizard, but once I try to Z auto align or bed level, the tool head to the X & Y Coordinates  correctly but in z it Goes further down and crashes into the bed and the probe holder while showing z=1 at the LCD, for more info I have a separate z end stop at max. the machine is an IDEX with dual Y and Z, running on a BBT octopus v1.1 f407 chip board.

### Bug Timeline

since I installed the 2.1.2 firmware it is a new custom build machine

### Expected behavior

deploying and stowing the probe normally during G34 Z AUTO ALIGNING AND BED LEVELING  

### Actual behavior

It crashes into the bed

### Steps to Reproduce

Call G34

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

I3 style Idex

### Electronics

BBT Octopus v1.1 F407 Chip

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11185291/Configuration.zip)

tiff-lee, 2023-04-09:
I have a similar issue when conducting Z-stepper auto align.. Previously worked fine on bugfix 2.0.x on 2.1.x it will do a single iteration (left to right) then when it moves left to start the second cycle it does not stop and just crashes straight into the bed even when the probe has been activated (this is using BLTouch).

thisiskeithb, 2023-04-09:
https://github.com/MarlinFirmware/Marlin/pull/25631 should fix several probe-related issues. Please give it a try and report your findings in that PR.

closed 2023-04-28 by thinkyhead