MarlinFirmware/Marlin #25660: [FR] Variable point amount of probing grid

Voxel3DPrinting, 2023-04-10:
### Is your feature request related to a problem? Please describe.

after seeing that Prusa only probes the area that the first layer will use, I decided to program that myself in a macro.
I want to only probe the area of the first layer convex hull, which I achieved with some placeholders and the G29 Ln Fn Rn Bn commands to set boundaries for the probing grid.

However, I also want to change the density of the grid;
if the print is small (< 200x200) , I want to probe a 3x3 grid
if it is bigger (> 200x200 and < 350x350), a 5x5 grid
and if it is really big (> 350x350 and < 500x500), a 7x7 grid

At this moment it is not possible to change it with Gcode commands

### Describe the feature you want

I would like it if there was a Gcode command available that changes the number of points of the probing grid.
For example; M557 Xn Ym
"n" changes the GRID_MAX_POINTS_X value in "configuration.h" 
"m" changes the GRID_MAX_POINTS_Y value in "configuration.h" 

I hope to hear if it is possible to implement this

ellensp, 2023-04-10:
Was looking into how prusa does this
Seem to use non standard gcode  M555: Set print area
Eg M555 X83.5722 Y68.0045 W82.8644 H82.2061
To define the print area, thus it knows where to probe and where to skip.

ellensp, 2023-04-10:
Similar to https://github.com/MarlinFirmware/Marlin/issues/20974 and https://github.com/MarlinFirmware/Marlin/issues/10547

lukasradek, 2023-04-14:
I agree that the variable number of grid points would be nice. It could be set as GRID_DENSITY in config for example.

But let me actually give huge kudos to the forsight of whomever (probably @thinkyhead ) implemented the bounds in G29 command. Meaning that now, when PrusaSlicer supports print bounds, it can be immediatelly put in use on Marlin machines 🙂.

thisiskeithb, 2023-04-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/10547

closed 2023-04-16 by thisiskeithb

Voxel3DPrinting, 2023-04-21:
Hi @thinkyhead and @lukasradek 

I have given it some thought and programmed a start Gcode to set bounds for probing. Although this does not change the density, it does work at changing the area...
If the compiling capabilities of the boards are an issue, maybe you could implement a much simpler option such as being able to probe a pre-defined point??

For example;
lets say you pre-define a 7x7 grid in marlin configuration. If you would be able to send M557 Xaa:bb Yaa:bb to select which pre-defined points to probe, you can achieve the same thing. For example M557 X3:5 Y3:5 will probe from the third to fifth point on X and the third to fifth point on Y, effectively a 3x3 grid in the middle of the bed. This way you pre-compile a grid spacing in visual studio, and in prusaslicer scripts you can verify which area to probe?

Prusaslicer code-example;
`; Probe area of the print in variable point density
{if (first_layer_print_max[0] - first_layer_print_min[0]) < 200}
    M557 X3:5 Y3:5 ; change probe point density to 3x3
	G29 L{(max(55, first_layer_print_min[0]) - 50)} F{(max(55, first_layer_print_min[1]) - 50)} R{(min(print_bed_max[0], first_layer_print_min[0] + 55) - 5)} B{(min(print_bed_max[1], first_layer_print_min[1] + 55) - 5)} ; probe area of first print
{elsif (first_layer_print_max[0] - first_layer_print_min[0]) > 200 and (first_layer_print_max[0] - first_layer_print_min[0]) < 300}
    M557 X2:6 Y2:6 ; change probe point density to 5x5
	G29 L{(max(55, first_layer_print_min[0]) - 50)} F{(max(55, first_layer_print_min[1]) - 50)} R{(min(print_bed_max[0], first_layer_print_min[0] + 55) - 5)} B{(min(print_bed_max[1], first_layer_print_min[1] + 55) - 5)} ; probe area of first print
{elsif (first_layer_print_max[0] - first_layer_print_min[0]) > 300 and (first_layer_print_max[0] - first_layer_print_min[0]) < 400}
    M557 X7:7 Y7:7 ; change probe point density to 7x7
	G29 L{(max(55, first_layer_print_min[0]) - 50)} F{(max(55, first_layer_print_min[1]) - 50)} R{(min(print_bed_max[0], first_layer_print_min[0] + 55) - 5)} B{(min(print_bed_max[1], first_layer_print_min[1] + 55) - 5)} ; probe area of first print
{endif}`