MarlinFirmware/Marlin #25661: [BUG] Fixed-Time Motion observations and problems.

ahmetcemturan, 2023-04-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

First of all I am convinced that FTM does a much better job than IS.  It is a bit buggy though.
I have also already posted a few comments to the original direction bug.
Robin Nano 1.2:
- Direction is inverted on Z axis
-Full homing works only if X axis reches endstop first otherwise will omit X homing
-Not tested further as I shorted my board
BTT SKR3:
- Tuning difficult as printer sops responding when an m493 command is sent during print.  Will work if command is sent during pause or after a dwell command. (had to add 1s dwell before commands in gcode)
- No possibility to see currnet settings (?) I tried M493 and got nothing in response.
- Settings are not stored in EEPROM it seems as they are lost when restarting.

### Bug Timeline

It is existent in the latest bugfix

### Expected behavior

Described above

### Actual behavior

Described above

### Version of Marlin Firmware

2.1.bugfix as of April 9th

### Printer model

Prusa clone for SKR3 and  Corexy for Nano

### Electronics

SKR3 and Nano 1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]
That's a 15min PLA bench with e3d v6 hotend.
So the ringing is gone for sure..

thisiskeithb, 2023-04-11:
> * Direction is inverted on Z axis

This was already reported in https://github.com/MarlinFirmware/Marlin/issues/25635

> Don't forget to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You forgot to do this.

ahmetcemturan, 2023-04-11:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11202257/Configuration_adv.zip)
Sorry.

danbrianwhite, 2023-05-13:
Thanks for posting about your success with using Dwell/G4 to fix the problem of the print stalling.

I found that I could add as little as 1ms of dwell to get the printer to continue. `G4 P1` When I don't add 1ms of dwell, the print will actually fail to run when it gets this gcode in it's memory. When I say fail to run, the print will home and then run no movements, and continue to show "printing" and cannot be canceled.

I was able to reduce this down to small snippets of gcode for testing

Non working gcode (printer homes and does not run any of the G0 movements)

```
G28 ; home all axes
G21 ; set units to millimeters
G90 ; use absolute coordinates
G92 E0

M493 S11 D0 ; Enable ZVD Input Shaping
M493 A0 ; (Hz) X Input Shaping Test
M493 B0 ; (Hz) Y Input Shaping Test

G0 Z10 F12000
G0 X10 Y10
G0 X100 Y100

M493 A15 ; (Hz) X Input Shaping Test
M493 B15 ; (Hz) Y Input Shaping Test

G0 X10 Y10
G0 X100 Y100
```

working gcode with dwell
```
G28 ; home all axes
G21 ; set units to millimeters
G90 ; use absolute coordinates
G92 E0

M493 S11 D0 ; Enable ZVD Input Shaping
G4 P1 ; not needed as A0 and B0 won't crash the print
M493 A0 ; (Hz) X Input Shaping Test
M493 B0 ; (Hz) Y Input Shaping Test

G0 Z10 F12000
G0 X10 Y10
G0 X100 Y100

G4 P1 ; dwell added here will enable the print to continue
M493 A15 ; (Hz) X Input Shaping Test
M493 B15 ; (Hz) Y Input Shaping Test

G0 X10 Y10
G0 X100 Y100
```

closed 2023-07-24 by github-actions[bot]