MarlinFirmware/Marlin #25682: Build error when enabling usb.

Drommotech, 2023-04-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I enable the second serial port in Configuration.h, and set it to -1 to enable USB for Octoprint. It will build before, but as soon as I enable another serial port it fails. I'm probably missing something obvious, but I can't figure it out =) As a side note, octoprint can tell that it's connected to something but, can't fully connect/send commands to the printer.

Using Ender 5 pro
Creality Mainboard V427
Fresh install of the latest Octoprint release
MACOS VS Code

Terminal Output from VS Code:

In file included from Marlin/src/HAL/STM32/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/MarlinSPI.h:24,
                 from Marlin/src/HAL/STM32/MarlinSPI.cpp:27:
Marlin/src/HAL/STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:31,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/./STM32/HAL.h:62:4: error: #error "SERIAL_PORT must be from 1 to 6."
   62 |   #error "SERIAL_PORT must be from 1 to 6."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:3183:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
 3183 |     #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:3183:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
 3183 |     #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:3183:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
 3183 |     #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."
      |      ^~~~~
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/MarlinSPI.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MYSERIAL2'?
   61 |   #define _SERIAL_LEAF_1 MYSERIAL1
........
Marlin/src/HAL/STM32/../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MYSERIAL2'?
........
Marlin/src/HAL/STM32/../../inc/../core/serial.h:114:56: error: template argument 1 is invalid
  114 |   typedef MultiSerial< REPEAT_1(NUM_SERIAL, _S_MULTI) 0> SerialOutputT;
      |                                                        ^
compilation terminated due to -fmax-errors=5.
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/HAL.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MYSERIAL2'?
........
*** [.pio/build/STM32F103RE_creality/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1
======================================================== [FAILED] Took 26.00 seconds ========================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:26.003
=================================================== 1 failed, 0 succeeded in 00:00:26.003 ===================================================

Octoprint Screen:

[image]

### Bug Timeline

Just got in the marlin loop, current vertion

### Expected behavior

USB over serial to be enabled

### Actual behavior

Build "crashed"

### Steps to Reproduce

Use Creality V427 Ender 5 pro example file

enable Serial 2 with -1 for USB

Attempt to build. 

### Version of Marlin Firmware

2.1.2

### Printer model

Ender 5 pro 

### Electronics

4.2.7 mainboard

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11237913/Marlin.zip)

EvilGremlin, 2023-04-15:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

EvilGremlin, 2023-04-15:
this board don't have native USB so -1 is invalid

closed 2023-04-15 by thisiskeithb

ellensp, 2023-04-15:
Serial port 1 is the usb cable via the ch340 chip
Serial port 3  is the serial port on the lcd connector
There are the only to ports you can use on this controller (without making board modifications)