MarlinFirmware/Marlin #25689: [FR] Adaptive Mesh Levelling

BlackHawk2212, 2023-04-15:
### Is your feature request related to a problem? Please describe.

Hi!
I'm not too sure if this has already been implemented or I'm totally missing something, but I thought that I should ask here (and make a feature request if it hasn't which I assume).
Essentially, I've got a large 300x300mm print bed and I don't want to mesh level the entire thing every time I start a print that is 10x10mm. Can Marlin automagically find the bounds of the gcode commands (or get it in the .gcode file) to actively mesh level the part of the build surface that I'm printing on in that print.
There is an equivalent for Klipper (https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging) and I recall that the Prusa XL has it as well,  but I can't seem to find it. 

This will reduce my print time where my printer already takes 5 minutes for me to level the entire surface and then print a tiny benchy on it. 

### Describe the feature you want

To proactively clarify, Marlin will have to find the maximum/minimum areas of the printed model, and within that area, mesh bed level a predefined grid e.g. 5x5, 3x3. I'm not too sure if Marlin can calculate (or already knows) the bounds of the gcode commands its being passed to generate the area that it needs to level. 

### Additional context

I'd be totally interested to help, test and seek further feedback on developing this feature. I have a intermediate understanding of C++ programming (in the context of microcontrollers). Thank you for your time.

ellensp, 2023-04-16:
same as FR https://github.com/MarlinFirmware/Marlin/issues/20974 and https://github.com/MarlinFirmware/Marlin/issues/10547
https://github.com/MarlinFirmware/Marlin/issues/25660

rondlh, 2023-04-16:
That would be useful of course. I believe the Prusa MK4 does this, it seems the slicer communicates the part location and size to Marlin before issuing the bed leveling command.

BlackHawk2212, 2023-04-16:
So seeing as all these feature requests are still open, I assume a solution has been found??However, I’ve found a solution where (only I’m Prusaslicer as far as I can tell) you can throw in :‘’’ xyz ‘’’To do it. It doesn’t have the smarts to automatically expand the grid beyond a certain size but it works for me so far.

thisiskeithb, 2023-04-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/25660

closed 2023-04-16 by thisiskeithb