MarlinFirmware/Marlin #25692: [BUG] Status Message cleanup commit compilation error

ThomasToka, 2023-04-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After applying this commit https://github.com/MarlinFirmware/Marlin/commit/9a1c02591ba4c3d5b41f4c64edd819ea1860b75b compilation of my fork stopped.

Probaly due to the usage of the Creality lcd_rts.cpp in the config. Before this commit it compiled fine.

Error Message in actual behavior.

### Bug Timeline

New bug after applying upstream commits

### Expected behavior

Compilation works

### Actual behavior

```
Marlin\src\gcode\bedlevel\G26.cpp: In member function 'bool::turn_on_heaters()':
Marlin\src\gcode\bedlevel\G26.cpp:325:12: error: 'class MarlinUI' has no member named 'quick_feedback'
325 | ui.quick_feedback();
| ^~~~~~~~~~~~~~
Marlin\src\gcode\bedlevel\G26.cpp:342:8: error: 'class MarlinUI' has no member named 'quick_feedback'
342 | ui.quick_feedback();
| ^~~~~~~~~~~~~~
Marlin\src\gcode\bedlevel\G26.cpp:350:8: error: 'class MarlinUI' has no member named 'completion_feedback'
350 | ui.completion_feedback();
| ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\bedlevel\G26.cpp: In member function 'bool::prime_nozzle()':
Marlin\src\gcode\bedlevel\G26.cpp:403:10: error: 'class MarlinUI' has no member named 'quick_feedback'
403 | ui.quick_feedback();
```

### Steps to Reproduce

- apply commit with lcd_rts.cpp enabled

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 S1 Pro

### Electronics

Creality F4 Mainboard

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11241599/configurations.zip)

thisiskeithb, 2023-04-16:
> apply commit with `lcd_rts.cpp` enabled

Is this custom code/from some other fork? We don’t have this file.

ThomasToka, 2023-04-16:
yeah sure. i am porting crealitys ender 3 s1 pro code. pr for this will be made on the 9th of may.

https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/src/lcd/dwin/lcd_rts.cpp

thisiskeithb, 2023-04-16:
I wouldn't consider this a Marlin bug since that code doesn't exist here. You'll need to fix it in your fork, but here are some places to look for help:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-04-16 by thisiskeithb

ThomasToka, 2023-04-16:
Ok i understand. no problem. thought it would affect other implementations also.

Will close this then and build a workaround for my fork.

ellensp, 2023-04-21:
Because this issue mentioned Creality lcd_rts.cpp and pointed to a non official marlin fork, this was miss diagnosed as an issue with that fork only.