MarlinFirmware/Marlin #25693: [BUG] M206 not usable anymore

ThomasToka, 2023-04-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After a rebuild of my Fork with bugfix-2.1.x in the beginning of the mont i have seen that M206 is not usable anymore to adjust x_min_pos and y_min_pos.

What is the way now to set those min positions like it was possible before the change with a gcode by a user to adjust his/her min positions?

### Bug Timeline

New bug, but probably caused by changes that i missed

### Expected behavior

Setting x_min and y_min postitions to adjust by gcode like

M206 x-0.8 y-2.0
M500

-> Printer uses this as x_min_pos and y_min_pos as home endstop position.

### Actual behavior

```
Recv: area:{full:{min:{x:-0.80,y:-2.00,z:0.00},max:{x:241.00,y:235.00,z:270.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:270.00}}}

Recv: echo:  M206 X0.00 Y0.00 Z0.00
```

so m206 is empty and x_min / y_min moved to area.

### Steps to Reproduce

look in M503. M206 is empty.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 S1 Pro

### Electronics

Creality F4 Mainboard

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11241616/configurations.zip)

ellensp, 2023-04-18:
I just tested this on a ramps. 

Fresh bugfix-2.1.x  firmware mismatched versions as expected so eeprom  was reset to defaults

M503 results, As expected

echo:  M206 X0.00 Y0.00 Z0.00

enter 
M206 X-1Y-2
M500

check M503, resulta as expected
echo:  M206 X-1.00 Y-2.00 Z0.00

reboot and re check M503, still has M206 values.

ThomasToka, 2023-04-18:
yes. that is right. it saves it in the fields. but put your printer on g28 x0 y0 and watch what your display says.

also area section is not updated.

firmware value for x_min_pos=-0.8 y_min_pos=-2

Recv: area:{full:{min:{x:-0.80,y:-2.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}}}
Recv: echo:  M206 X0.00 Y0.00 Z0.00

x and y endstop positions shown are x=-0.8 y=-2

setting 

M206 X-2 Y-2

results in 

Recv: area:{full:{min:{x:-0.80,y:-2.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}}}
Recv: echo:  M206 X-2.00 Y-2.00 Z0.00

x and y endstop positions shown are x=-0.8 y=-2

so it is saved but not used.

ThomasToka, 2023-04-18:
Just to verify again:

- installation of firmware with x_min_pos=-2 and y_min_pos=-2

Recv: area:{full:{min:{x:-2.00,y:-2.00,z:0.00},max:{x:241.00,y:235.00,z:270.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:270.00}}}

Recv: echo:  M206 X0.00 Y0.00 Z0.00

setting 

M206 x-3 y-3
M500
m503
shows:
Recv: echo:  M206 X-3.00 Y-3.00 Z0.00

Recv: echo:enqueueing "G28 X Y"
[...]
Send: M114
[...]
Recv: X:-5.00 Y:-5.00 Z:6.75 E:0.00 Count X:-159 Y:-159 Z:2700

so the offsets are added.

rebooting printer.

area shows:

Recv: area:{full:{min:{x:-5.00,y:-5.00,z:0.00},max:{x:238.00,y:232.00,z:270.00}},work:{min:{x:-3.00,y:-3.00,z:0.00},max:{x:232.00,y:232.00,z:270.00}}}

so in full it was added. in first max x was added with+ in x it was subtracted, in min the values are show as set with m206
and finally in last max the are all subtracted.

well.. 

can you explain?

ThomasToka, 2023-04-18:
Just to verify again:

    installation of firmware with x_min_pos=-2 and y_min_pos=-2

x_bed_size 235 + 6
y_bed_size 235

Recv: area:{full:{min:{x:-2.00,y:-2.00,z:0.00},max:{x:241.00,y:235.00,z:270.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:270.00}}}

Recv: echo: M206 X0.00 Y0.00 Z0.00

setting

M206 x-3 y-3
M500
m503
shows:
Recv: echo: M206 X-3.00 Y-3.00 Z0.00

Recv: echo:enqueueing "G28 X Y"
[...]
Send: M114
[...]
Recv: X:-5.00 Y:-5.00 Z:6.75 E:0.00 Count X:-159 Y:-159 Z:2700

so the offsets are added.

rebooting printer.

area shows:

Recv: area:{full:{min:{x:-5.00,y:-5.00,z:0.00},max:{x:238.00,y:232.00,z:270.00}},work:{min:{x:-3.00,y:-3.00,z:0.00},max:{x:232.00,y:232.00,z:270.00}}}

in full it was added. i think it should overwrite it.
in max it reduced max positions. here it should not affect it so not be visible.
in min it is ok as it should be
second max it reduced by subtracting. what is this used for?

well..

can you explain?

ThomasToka, 2023-04-18:
@ellensp please do G28 X Y and M114 after setting it. there you should see it.

ThomasToka, 2023-04-18:
i maybe did a false example initially i realize. but the problem comes more out the further i dig in it.

users are not able anymore to set their offsets correctly with M206. I know it worked as it should but cant find the change and also cant find my last working version with this atm..

closed 2023-06-28 by github-actions[bot]