MarlinFirmware/Marlin #25694: [BUG] SKR Mini E3 V3.0 does not boot

mistic100, 2023-04-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On the latest bugfix 2.1 I just get a blank screen on my SKR Mini E3 v3.0

I tracked down the issue to this change https://github.com/EvilGremlin/Marlin/commit/4744d403285dd173b263ea683430c93e5bc69e17#diff-6d2f0ad22cd671f2b1ac1899a2841f2cbe46bb187d8b6c25623236decf4bd51aL133
Reverting to the previous code allows to boot again.

### Version of Marlin Firmware

2.1.x

### Printer model

Ender 3

### Electronics

SKR Mini E3 v3.0

### Add-ons

ESP8266 module (with ESP3D) + BIGTREETECH Relay V1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

- [Configuration.h](https://github.com/mistic100/Marlin/blob/e2efa6d3d511f5d2ebe1b0e15eb5370312852c41/Marlin/Configuration.h)
- [Configuration_adv.h](https://github.com/mistic100/Marlin/blob/e2efa6d3d511f5d2ebe1b0e15eb5370312852c41/Marlin/Configuration_adv.h)

ellensp, 2023-04-17:
Tested these configs on latest bugfix, the a bare board with cr10 stock display booted without issue

Checked your fork https://github.com/mistic100/Marlin/tree/e2efa6d3d511f5d2ebe1b0e15eb5370312852c41 with the configs for other  'weirdness' only diff was the patch you made,

Which leaves hardware...   

You sure its not a 3.0.1?
Is the processor still a stm32g0b1 ret6?   (ie they have done a dirty on us and swapped processors?)

thisiskeithb, 2023-04-17:
> Tested these configs on latest bugfix, the a bare board with cr10 stock display booted without issue

No issues here either.

---

Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

closed 2023-04-17 by thisiskeithb