MarlinFirmware/Marlin #25698: [BUG] Watchdog reset loop with  Marlin  bugfix-2.1.x build for Alfawise U20

eskalator, 2023-04-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Greetings!
I've built Marlin-bugfix-2.1.x-d77e921 for Alfawise U20 with configuration options for KlackEnder probe and Input Shaping enabled. 
Unfortunately the firmware hangs and boards resets due to watchdog.
I've tried the Marlin-2.1.x-2399a24 with the same result. 
I've got  postmortem debugging enabled so acquired some data, but to use them properly is somtehing over my head, at least right now. 
Here is information droped on serial(via pronterface):

start
 Watchdog Reset
echo: Last Updated: 2023-04-15 | Author: Hobi, tpruvot
echo: Compiled: Apr 17 2023
echo: Free Memory: 48679  PlannerBufferBytes: 3328
echo:V87 stored settings retrieved (644 bytes; crc 38851)
Unified Bed Leveling System v1.01 inactive
UBL reset
## Software Fault detected ##
Cause: Hard
R0   : 0x0804488C
R1   : 0x00000001
R2   : 0x00000000
R3   : 0x00008000
R12  : 0xC35E9800
LR   : 0x0803B32B
PC   : 0x0803B32C
PSR  : 0x21000000
CFSR : 0x00000000
HFSR : 0x40000000
DFSR : 0x00000000
AFSR : 0x00000000
MMAR : 0xE000ED34
BFAR : 0xE000ED38
ExcLR: 0xFFFFFFF9
ExcSP: 0x2000FF68
Backtrace:#1 : unknown@0x0803B308+36 PC:0x0803B32C
#2 : unknown@0x0803CC58+46 PC:0x0803CC86
#3 : unknown@0x0803CFDC+44 PC:0x0803D008
#4 : unknown@0x0801914C+98 PC:0x080191AE
#5 : unknown@0x080195C8+8 PC:0x080195D0
#6 : unknown@0x0801252E+14 PC:0x0801253C
#7 : unknown@0x08012604+50 PC:0x08012636

Here is backtrace from debugger:

#0  0x08018bd2 in CommonHandler_C (frame=0x2000ff68, lr=4294967289, cause=2051)
    at Marlin\src\HAL\shared\cpu_exception\exception_arm.cpp:250
#1  <signal handler called>
#2  0x0803b32c in DiskIODriver_SPI_SD::init (this=0x200038e4 <CardReader::media_driver_sdcard>,
    sckRateID=<optimized out>, chipSelectPin=<optimized out>) at Marlin\src\sd\Sd2Card.cpp:262
#3  0x0803cc86 in CardReader::mount () at Marlin\src\sd\cardreader.cpp:481
#4  0x0803d008 in CardReader::manage_media () at Marlin\src\sd\cardreader.cpp:535
#5  0x080191ae in idle (no_stepper_sleep=<optimized out>) at Marlin\src\MarlinCore.cpp:837
#6  idle (no_stepper_sleep=<optimized out>) at Marlin\src\MarlinCore.cpp:785
#7  0x080195d0 in loop () at Marlin\src\MarlinCore.cpp:1688
#8  0x0801253c in main ()
    at C:\Users\zoohoo\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\main.cpp:58

I've used configuration files from examples folder for Alfawise U20(without bltouch) and enabled KlackEnder support, Input Shaping and postmortem debugging.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/11249318/config.zip)

Tell my what other information You need and i'll try to get them.

### Bug Timeline

10.04.2023 when first trying to run bugfix-2.1.x

### Expected behavior

Firmware to run

### Actual behavior

Watchdog reset loop

### Steps to Reproduce

1. Build firmware based on config files.
2. Erase the stm32f103 using ST-Link.
3. Flash firmware on address 0x8010000.
4. Flash the Longer/Alfawise bootloader on address 0x8000000(with bootloader running it is not possible to flash the firmware most of the times).
5. The firmware starts emediately(running from 3.3V supplied by ST-Link) and gets into watchdog reset loop(serial is running).

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Alfawise U20

### Electronics

Stock board(stm32f103)

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-17:
If you disable #define USE_WATCHDOG (just for testing) does it hang? or actually do what is expected (all be it a bit slowly)

eskalator, 2023-04-17:
Without watchdog board resets anyway. Instead of watchdog reset i get software reset.

start
 Software Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2023-04-15 | Author: Hobi, tpruvot
echo: Compiled: Apr 17 2023
echo: Free Memory: 48679  PlannerBufferBytes: 3328
echo:V87 stored settings retrieved (644 bytes; crc 38851)
Unified Bed Leveling System v1.01 inactive
UBL reset
## Software Fault detected ##
Cause: Hard
R0   : 0x0804472C
R1   : 0x00000001
R2   : 0x00000000
R3   : 0x00008000
R12  : 0xC35FE000
LR   : 0x0803B2CF
PC   : 0x0803B2D0
PSR  : 0x21000000
CFSR : 0x00000000
HFSR : 0x40000000
DFSR : 0x00000000
AFSR : 0x00000000
MMAR : 0xE000ED34
BFAR : 0xE000ED38
ExcLR: 0xFFFFFFF9
ExcSP: 0x2000FF68
Backtrace:#1 : unknown@0x0803B2B0+32 PC:0x0803B2D0
#2 : unknown@0x0803CBEC+46 PC:0x0803CC1A
#3 : unknown@0x0803CF70+44 PC:0x0803CF9C
#4 : unknown@0x08019108+98 PC:0x0801916A
#5 : unknown@0x0801957C+8 PC:0x08019584
#6 : unknown@0x0801252E+14 PC:0x0801253C
#7 : unknown@0x08012604+50 PC:0x08012636

The registers are slightly different, but the debugger backtrace is completely the same.
The other different thing is that the led on the board is lit steadily instead of blinking. With watchdog it was blinking and then after few seconds there was reset.

eskalator, 2023-04-17:
I though about it earlier from the same reason as You- the code seems to hang around cardreader handling function- but did not think to just disable it. 
Of course it works now! Or at least it goes further without hanging- i have main screen and touchscreen works. I'll be testing the rest. 
It's a blessing for me- i don't use sd card at all and bed level probe is a  must with my table. But it would be nice to have the sd option.
I think it  would be best if somebody could reproduce the bug on the same board/printer, because i now suspect if my card reader is not broken. On the other hand i got 2.0.x running with no problem on it, but i never was able to check the reader. I mean i got only 64GB card and thought that maybe the reader is unable to read it because of the size. Now i'm wondering if maybe it just doesn't work properly. 
That's why it would be best to reproduce the bug before making any more effort to try to solve it- it may be just my hardware.

closed 2023-06-27 by github-actions[bot]