MarlinFirmware/Marlin #25699: [BUG] (bug summary)

T-SquaredProductions, 2023-04-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

System does not start up. Screen lights up, but still non-responsive, no screen, no sound. Power supply fan changes speed as if booted up correctly.

### Bug Timeline

Old issue, first reported on issue #25578: https://github.com/MarlinFirmware/Marlin/issues/25578

### Expected behavior

Bootup without any problems, with sound, graphics, and responsiveness.

### Actual behavior

Blank lit screen, no graphics, no response, no sound.
Boots normally and bug does not happen with EEPROM_SETTINGS disabled.

### Steps to Reproduce

Compile version 2.1 or 2.1.x bugfix with EEPROM_SETTINGS enabled.
Upload using FIRMWARE.bin file through microSD slot on the BIGTREETECH SKR2.

### Version of Marlin Firmware

2.1

### Printer model

Creality 

### Electronics

BigTreeTech SKR 2

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin 2.1.x Bugfix Config Archive.zip](https://github.com/MarlinFirmware/Marlin/files/11249607/Marlin.2.1.x.Bugfix.Config.Archive.zip)

T-SquaredProductions, 2023-04-17:
I used the BIGTREE_SKR_2 build environment, because that has worked for me in the past. (I always take the relevant settings from a previous source of Marlin that I used in the past, even for platformIO-based systems such as the SKR 2.) 

When I first bought the board, there were never any versions with extra hardware. Maybe it's changed or been upgraded now, but the version I got around 2020 or so was the base version with just the USB-A input, micro SD card, and USB-B ports: https://spool3d.ca/bigtreetech-skr-2-32bit-control-board/ (Just as an example. I never bought from this store.)

DForck42, 2023-04-18:
i had a similar issue a few months ago, i had to go back to 2.0.x to get a version that worked for my skr board

T-SquaredProductions, 2023-04-18:
All right, I tried BIGTREE_SKR_2_USB, and it didn't work.

thisiskeithb, 2023-04-18:
Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

T-SquaredProductions, 2023-04-19:
Still not getting a screen. I deleted both the .pio and .platformio folders.

thisiskeithb, 2023-04-19:
I have no issues here with your config and current `bugfix-2.1.x` (19b4ae8) using an SKR 2 Rev. B (STM32F407VGT6) & RRDFG display.

Marlin builds fine, the firmware flashes, and board boots as expected. Other than resetting EEPROM on boot due to a version mismatch (which is expected), there are no issues with the LCD. Saving values to EEPROM and checking with `M503` also return the expected values.

Just to be thorough, I also deleted my `~/.platformio/` folder, reinstalled PIO, rebuilt the firmware, and reflashed. It also works as expected.

Here's the binary that was generated: [Issue-25699-firmware_bin.zip](https://github.com/MarlinFirmware/Marlin/files/11268524/Issue-25699-firmware_bin.zip)

You can try also reinstalling VSCode, but it's likely a hardware issues.

closed 2023-04-19 by thisiskeithb