MarlinFirmware/Marlin #25702: [BUG] BLTOUCH_HS_MODE in SanityCheck.h error if true or false

classicrocker883, 2023-04-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

if `BLTOUCH_HS_MODE` is defined as true or false, this error pops up, otherwise 1 or 0 it is fine

>../../../inc/SanityCheck.h:1904:9: error: static assertion failed: BLTOUCH_HS_MODE must now be defined as true or false, indicating the default state.
         static_assert(!(strcmp(hs, "1") && strcmp(hs, "true") && strcmp(hs, "0") && strcmp(hs, "false")),  

```
expression must have a constant valueC/C++(28)
SanityCheck.h(1904, 23): cannot call non-constexpr function "strcmp" (declared at line 36 of "C:\Users\.platformio\packages\toolchain-gccarmnoneeabi@1.70201.0\arm-none-eabi\include\string.h")
```

### Version of Marlin Firmware

current

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-04-17:
You didn't attach your configs, but I cannot reproduce this error with the current `bugfix-2.1.x` (4d8ab1c).

closed 2023-04-17 by thisiskeithb

classicrocker883, 2023-04-18:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11257206/Configurations.zip)

I can surely compile fine, however when I go into **Marlin/src/inc/SanityCheck.h*, line _1902_, if I hover over the ' ! '
 this is the error I get.

[image]

im wondering what in the code can be changed so this maybe will go away?
should something like  `#include` "somefile.h" work?

classicrocker883, 2023-04-18:
update: for some reason now it compiles fine, I think a Cleanup helped and it solved itself. however it still shows the red squiggle. it just compiles normally now, but before it would only accept 1 or 0.... putting "true or false" would give error. so false alarm i suppose.

MathieuMorrissette, 2023-04-22:
I had this issue when trying to compile with a _maple environment.

Concretely, I'm was building the firmware for an skr mini e3 v2.0 and I had to switch the default env to `STM32F103RC_btt` from `STM32F103RC_btt_maple`. Probably some kind of unsupported c++ feature in the maple env.

classicrocker883, 2023-04-22:
yes it has something to do with the HAL library, one of the STMF1 related folders. I suspect there is a line of code which changes the identity or value of the `BLTOUCH_HS_MODE`.