MarlinFirmware/Marlin #25703: [BUG] HMI_data.z_after_homing not declared in src/module/motion.h

classicrocker883, 2023-04-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Just Enable 
```
#if HAS_Z_AXIS
  #if ALL(DWIN_LCD_PROUI, INDIVIDUAL_AXIS_HOMING_SUBMENU, MESH_BED_LEVELING)
```

>In file included from Marlin\src\MarlinCore.cpp:47:
Marlin\src\module/motion.h: In function 'void do_z_post_clearance()':
Marlin\src\module/motion.h:406:30: error: 'HMI_data' was not declared in this scope
  406 |     #define Z_POST_CLEARANCE HMI_data.z_after_homing
      |                              ^~~~~~~~
Marlin\src\module/motion.h:415:54: note: in expansion of macro 'Z_POST_CLEARANCE'
  415 |   inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
      |                                                      ^~~~~~~~~~~~~~~~
Compiling .pio\build\STM32F103RE_creality\src\src\feature\cancel_object.cpp.o
In file included from Marlin\src\feature\babystep.cpp:29:
Marlin\src\feature\../module/motion.h: In function 'void do_z_post_clearance()':
Marlin\src\feature\../module/motion.h:406:30: error: 'HMI_data' was not declared in this scope
  406 |     #define Z_POST_CLEARANCE HMI_data.z_after_homing
      |                              ^~~~~~~~
Marlin\src\feature\../module/motion.h:415:54: note: in expansion of macro 'Z_POST_CLEARANCE'
  415 |   inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
      |                                                      ^~~~~~~~~~~~~~~~
Compiling .pio\build\STM32F103RE_creality\src\src\feature\e_parser.cpp.o
*** [.pio\build\STM32F103RE_creality\src\src\MarlinCore.cpp.o] Error 1
*** [.pio\build\STM32F103RE_creality\src\src\feature\babystep.cpp.o] Error 1
In file included from Marlin\src\feature\bedlevel\../../module/planner.h:46,
                 from Marlin\src\feature\bedlevel\bedlevel.cpp:28:
Marlin\src\feature\bedlevel\../../module/motion.h: In function 'void do_z_post_clearance()':
Marlin\src\feature\bedlevel\../../module/motion.h:406:30: error: 'HMI_data' was not declared in this scope
  406 |     #define Z_POST_CLEARANCE HMI_data.z_after_homing
      |                              ^~~~~~~~
Marlin\src\feature\bedlevel\../../module/motion.h:415:54: note: in expansion of macro 'Z_POST_CLEARANCE'
  415 |   inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
      |                                                      ^~~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\feature\bedlevel\bedlevel.cpp.o] Error 1
In file included from Marlin\src\feature\../lcd/marlinui.h:26,
                 from Marlin\src\feature\cancel_object.cpp:29:
Marlin\src\feature\../lcd/../module/motion.h: In function 'void do_z_post_clearance()':
Marlin\src\feature\../lcd/../module/motion.h:406:30: error: 'HMI_data' was not declared in this scope
  406 |     #define Z_POST_CLEARANCE HMI_data.z_after_homing
      |                              ^~~~~~~~
Marlin\src\feature\../lcd/../module/motion.h:415:54: note: in expansion of macro 'Z_POST_CLEARANCE'
  415 |   inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
      |                                                      ^~~~~~~~~~~~~~~~
In file included from Marlin\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp:29:
Marlin\src\feature\bedlevel\mbl\../../../module/motion.h: In function 'void do_z_post_clearance()':
Marlin\src\feature\bedlevel\mbl\../../../module/motion.h:406:30: error: 'HMI_data' was not declared in this scope
  406 |     #define Z_POST_CLEARANCE HMI_data.z_after_homing
      |                              ^~~~~~~~
Marlin\src\feature\bedlevel\mbl\../../../module/motion.h:415:54: note: in expansion of macro 'Z_POST_CLEARANCE'
  415 |   inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
      |                                                      ^~~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o] Error 1
*** [.pio\build\STM32F103RE_creality\src\src\feature\cancel_object.cpp.o] Error 1

<br>
<br>

around line _400_ in **src/module/motion.h** this is the following code

<br>
</br>

```
#if HAS_Z_AXIS
  #if ALL(DWIN_LCD_PROUI, INDIVIDUAL_AXIS_HOMING_SUBMENU, MESH_BED_LEVELING)
    #define Z_POST_CLEARANCE HMI_data.z_after_homing
  #elif defined(Z_AFTER_HOMING)
    #define Z_POST_CLEARANCE Z_AFTER_HOMING
  #else
    #define Z_POST_CLEARANCE Z_CLEARANCE_FOR_HOMING
  #endif
  void do_z_clearance(const_float_t zclear, const bool with_probe=true, const bool lower_allowed=false);
  void do_z_clearance_by(const_float_t zclear);
  void do_move_after_z_homing();
  inline void do_z_post_clearance() { do_z_clearance(Z_POST_CLEARANCE); }
#else
  inline void do_z_clearance(float, bool=true, bool=false) {}
  inline void do_z_clearance_by(float) {}
#endif
```

### Bug Timeline

new

### Steps to Reproduce

1. define `DWIN_LCD_PROUI, INDIVIDUAL_AXIS_HOMING_SUBMENU, MESH_BED_LEVELING`
2. must `HAS_Z_AXIS`
3. build

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

perhaps put
```
#if ENABLED(DWIN_LCD_PROUI)
#include "../lcd/e3v2/proui/dwin.h"
#endif
```
at the beginning of **motion.h**

classicrocker883, 2023-04-18:
update, I just realized that
```
#if ALL(DWIN_LCD_PROUI, INDIVIDUAL_AXIS_HOMING_SUBMENU, MESH_BED_LEVELING)
  #include "../lcd/e3v2/proui/dwin.h"
#endif
```
is in **motion.cpp**

and it should be placed in **motion.h**

thisiskeithb, 2023-05-29:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25858

closed 2023-05-29 by thisiskeithb