MarlinFirmware/Marlin #25718: [BUG] (not able to compile)

discip, 2023-04-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I did not change anything.
Just did update to the latest Marlin bugfix-2.1.x

```c#
Marlin/src/gcode/bedlevel/G26.cpp: In member function 'bool<unnamed struct>::turn_on_heaters()':
Marlin/src/gcode/bedlevel/G26.cpp:325:12: error: 'class MarlinUI' has no member named 'quick_feedback'
  325 |         ui.quick_feedback();
      |            ^~~~~~~~~~~~~~
Marlin/src/gcode/bedlevel/G26.cpp:342:8: error: 'class MarlinUI' has no member named 'quick_feedback'
  342 |     ui.quick_feedback();
      |        ^~~~~~~~~~~~~~
Marlin/src/gcode/bedlevel/G26.cpp:350:8: error: 'class MarlinUI' has no member named 'completion_feedback'
  350 |     ui.completion_feedback();
      |        ^~~~~~~~~~~~~~~~~~~
Marlin/src/gcode/bedlevel/G26.cpp: In member function 'bool<unnamed struct>::prime_nozzle()':
Marlin/src/gcode/bedlevel/G26.cpp:403:10: error: 'class MarlinUI' has no member named 'quick_feedback'
  403 |       ui.quick_feedback();
      |          ^~~~~~~~~~~~~~
*** [.pio/build/STM32H743Vx_btt/src/src/gcode/bedlevel/G26.cpp.o] Error 1
```

### Bug Timeline

around commit 9a1c025

### Expected behavior

Should compile as previously.

### Actual behavior

Not compiling!

### Steps to Reproduce

1. Update Marlin.
2. 2. Start to compile

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/commit/91cce2a7b39f1b64e37224fe6bef4ce00463fecf

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11288672/Configuration.zip)

ellensp, 2023-04-20:
code presumes UI.quick_feedback exists 

There are also config issue
eg
```
#if DISABLED(CLASSIC_JERK)
  #define JUNCTION_DEVIATION_MM 0.4 * sq(DEFAULT_XJERK) / DEFAULT_ACCELERATION // changed [0.4*Jerk²/Accel-print] 0.032 (mm) Distance from real junction edge

```
but DEFAULT_XJERK is only defined if CLASSIC_JERK is enabled...

ellensp, 2023-04-20:
This should fix the code

```DIFF
diff --git a/Marlin/src/gcode/bedlevel/G26.cpp b/Marlin/src/gcode/bedlevel/G26.cpp
index fe20423b8d..12b60e611a 100644
--- a/Marlin/src/gcode/bedlevel/G26.cpp
+++ b/Marlin/src/gcode/bedlevel/G26.cpp
@@ -322,7 +322,7 @@ typedef struct {
 
       if (bed_temp > 25) {
         LCD_MESSAGE_MAX(MSG_G26_HEATING_BED);
-        ui.quick_feedback();
+        TERN_(HAS_MARLINUI_MENU, ui.quick_feedback());
         TERN_(HAS_MARLINUI_MENU, ui.capture());
         thermalManager.setTargetBed(bed_temp);
 
@@ -339,7 +339,7 @@ typedef struct {
 
     // Start heating the active nozzle
     LCD_MESSAGE_MAX(MSG_G26_HEATING_NOZZLE);
-    ui.quick_feedback();
+    TERN_(HAS_MARLINUI_MENU,ui.quick_feedback());
     thermalManager.setTargetHotend(hotend_temp, active_extruder);
 
     // Wait for the temperature to stabilize
@@ -347,7 +347,7 @@ typedef struct {
       return G26_ERR;
 
     ui.reset_status();
-    ui.completion_feedback();
+    TERN_(HAS_MARLINUI_MENU,ui.completion_feedback());
 
     return G26_OK;
   }
@@ -400,7 +400,7 @@ typedef struct {
     #endif
     {
       LCD_MESSAGE_MAX(MSG_G26_FIXED_LENGTH);
-      ui.quick_feedback();
+      TERN_(HAS_MARLINUI_MENU,ui.quick_feedback());
       destination = current_position;
       destination.e += prime_length;
       prepare_internal_move_to_destination(fr_slow_e);

```

ellensp, 2023-04-20:
Or perhaps the intent to to make these two calls allways work, without or without a display, since it has this code, this seems more likely. and the code should be 

```DIFF
diff --git a/Marlin/src/lcd/marlinui.h b/Marlin/src/lcd/marlinui.h
index 32d747da43..4c5f7c9bcc 100644
--- a/Marlin/src/lcd/marlinui.h
+++ b/Marlin/src/lcd/marlinui.h
@@ -494,11 +494,6 @@ public:
 
       static void status_screen();
 
-    #else
-
-      static void quick_feedback(const bool=true) {}
-      static void completion_feedback(const bool=true) {}
-
     #endif
 
     #if HAS_MARLINUI_U8GLIB
@@ -526,6 +521,8 @@ public:
 
     static void update() {}
     static void kill_screen(FSTR_P const, FSTR_P const) {}
+    static void quick_feedback(const bool=true) {}
+    static void completion_feedback(const bool=true) {}
 
   #endif
 ```

discip, 2023-04-20:
@ellensp
Thank you very much for the quick response! 👍🏻
1. > There is also a config issue
...
but DEFAULT_XJERK is only defined if CLASSIC_JERK is enabled...

   I know about that, but it was added later on for testing purposes.
2. > Or perhaps the intent to to make these two calls always work, with or without a display, since it has this code, this seems more likely. and the code should be

Are you planing on opening a PR for this?

discip, 2023-04-20:
Unfortunately it is still not working.
This is what I am getting now:

```c#
In file included from Marlin/src/feature/../module/stepper/../../feature/tmc_util.h:25,
                 from Marlin/src/feature/../module/stepper/trinamic.h:35,
                 from Marlin/src/feature/../module/stepper/indirection.h:40,
                 from Marlin/src/feature/../module/stepper.h:47,
                 from Marlin/src/feature/babystep.cpp:31:
Marlin/src/feature/../module/stepper/../../feature/../lcd/marlinui.h:524:1: error: expected unqualified-id before '+' token
  524 | +   static void quick_feedback(const bool=true) {}
      | ^
Marlin/src/feature/../module/stepper/../../feature/../lcd/marlinui.h:525:1: error: expected unqualified-id before '+' token
  525 | +   static void completion_feedback(const bool=true) {}
      | ^
*** [.pio/build/STM32H743Vx_btt/src/src/feature/babystep.cpp.o] Error 1
In file included from Marlin/src/module/stepper/../../feature/tmc_util.h:25,
                 from Marlin/src/module/stepper/trinamic.h:35,
                 from Marlin/src/module/stepper/indirection.h:40,
                 from Marlin/src/module/stepper.h:47,
                 from Marlin/src/MarlinCore.cpp:51:
Marlin/src/module/stepper/../../feature/../lcd/marlinui.h:524:1: error: expected unqualified-id before '+' token
  524 | +   static void quick_feedback(const bool=true) {}
      | ^
Marlin/src/module/stepper/../../feature/../lcd/marlinui.h:525:1: error: expected unqualified-id before '+' token
  525 | +   static void completion_feedback(const bool=true) {}
      | ^
*** [.pio/build/STM32H743Vx_btt/src/src/MarlinCore.cpp.o] Error 1
```

ThomasToka, 2023-04-20:
thats what i reported as i knew it would affect not only my fork https://github.com/MarlinFirmware/Marlin/issues/25692

ellensp, 2023-04-21:
A PR has been created

closed 2023-04-21 by ellensp