MarlinFirmware/Marlin #25721: [BUG] Unable to build the latest Bugfix 2.1.x for Anycubic Vyper

Mangers95, 2023-04-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello Guys , 
Auto Build Marlin alway run into Error at dgus_tft.cpp .

[image]

### Bug Timeline

I think its a older bud? idk

### Expected behavior

i hoped it would build without an error.

### Actual behavior

Therer are More Issues like in the FileNavigator.cpp but i think ived solved this problem by adding +1 to line 44 like this ->[MAX_PATH_LEN + 1];

but im unable to find or solve the issue in the Dgus_tft.cpp

### Steps to Reproduce

First downloaded the latest Bugfix with changes winthin the last 8 Hours and the matching Configurations.
Then put the Configs to the Marlin Directory and load it into VS Code .

Now ive tried to Build via Auto Build Marlin and driect via Platorm.io/boards.

Both ways run into Error at the Point dgus_tdt.cpp

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Apr  3 2023 15:12:32)

### Printer model

Anycubic Vyper

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-21:
The function PlayTune originally took 3 parameters, the first being beeperPin was not used so it was removed from the code, but not the function definition was not updated. 
```DIFF
diff --git a/Marlin/src/lcd/extui/anycubic/Tunes.h b/Marlin/src/lcd/extui/anycubic/Tunes.h
index a05107738d..9f7bf6e1ca 100644
--- a/Marlin/src/lcd/extui/anycubic/Tunes.h
+++ b/Marlin/src/lcd/extui/anycubic/Tunes.h
@@ -61,7 +61,7 @@ n_END=10000 // end of tune marker
 
 namespace Anycubic {
 
-  void PlayTune(const uint8_t beeperPin, const uint16_t *tune, const uint8_t speed);
+  void PlayTune( const uint16_t *tune, const uint8_t speed);
 
   // Only uncomment the tunes you are using to save memory
   // This will help you write tunes!

```

fixes MAX_PATH_LEN, MAX_PATH_LEN+1 issues

```DIFF
diff --git a/Marlin/src/lcd/extui/anycubic_vyper/FileNavigator.cpp b/Marlin/src/lcd/extui/anycubic_vyper/FileNavigator.cpp
index 38603f5679..9006c38996 100644
--- a/Marlin/src/lcd/extui/anycubic_vyper/FileNavigator.cpp
+++ b/Marlin/src/lcd/extui/anycubic_vyper/FileNavigator.cpp
@@ -41,7 +41,7 @@ using namespace ExtUI;
 namespace Anycubic {
 
   FileList  FileNavigator::filelist;                          // Instance of the Marlin file API
-  char      FileNavigator::currentfoldername[MAX_PATH_LEN];   // Current folder path
+  char      FileNavigator::currentfoldername[];               // Current folder path
   uint16_t  FileNavigator::lastindex;
   uint8_t   FileNavigator::folderdepth;
   uint16_t  FileNavigator::currentindex;                      // override the panel request
```
This compiles, but I cannot test 

Please test this.

Mangers95, 2023-04-21:
i have tried it a few minutes ago. 
Its working fine and im  able to build this now via Auto Build marlin.

**### Great work and a big thank u !!!!**

thisiskeithb, 2023-04-21:
This still needs a PR created to patch the bug, so let's leave it open for now.

Mangers95, 2023-04-21:
Im sorry , i dont know the github system for very long 🙃

ellensp, 2023-04-22:
Pr created, closing.

closed 2023-04-22 by ellensp

Mangers95, 2023-04-23:
Hello Guys . 
Im now able to compillie the file .
But when i try to flash the bin file the Printer doesnt detect the file and start up without flashing

Heres the name of the bin file firmware-20230423-195008.bin ... is this right? 
if i rename the file to something like main_board_20xxxxxx.bin the printer reconize it but fails to flash and the display is dead.
After restart the printer starts makinge stange noises and display goes on ... but dont show any temeratures.

i hope u guys understand my problem and my english :D