MarlinFirmware/Marlin #25722: [BUG] Odd Z-homing behavior if Z is moved to negative value before homing

rondlh, 2023-04-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Not sure if it's a bug, but seems odd and a waste of time to me...
The Z-homing seems a bit odd to when you move the Z-axis to the negative direction after a reset (Z-position reset to 0), then the Z-position becomes negative (let's say -50mm). Then running a G28Z will first move the Z-axis back to 0 (50 mm further away from the probe) and then run the homing. This seems to be a waste of time.

I solved this by adding the following line in G28.cpp:
`if (current_position.z < Z_MIN_POS) current_position.z = Z_MIN_POS;`

### Bug Timeline

Don't know

### Expected behavior

Just execute the homing assuming you are at the minimum Z-position already.
No need to go so far away from the probe, because the "unknown" Z-position is out of range anyway.

### Actual behavior

The Z-axis moves away from the probe further than needed before homing, wasting time.

### Steps to Reproduce

1. Reset the motherboard
2. Move the Z-axis in the negative direction (like Z=-50)
3. Execute a Z-homing G28Z

### Version of Marlin Firmware

Latest Marling 2.1.x bugfix 2023-04-21

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BL Touch V3.1, BTT TFT35 V3.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11294875/Configuration_adv.zip)

DerAndere1, 2023-04-26:
I think it is possible to move below Z_MIN_POS if DEFAULT_ENDSTOPS_ALWAYS_ON is disabled and software endstops are disabled. So your proposal would not work in that situation

rondlh, 2023-04-27:
> I think it is possible to move below Z_MIN_POS if DEFAULT_ENDSTOPS_ALWAYS_ON is disabled and software endstops are disabled. So your proposal would not work in that situation

I don't fully understand this feedback. Could you paint the picture a bit clearer?
Before homing (G28) the printer doesn't know where the carriage is, and in my case I'm free to move below Z_MIN_POS, like Z = -50. But physically it's not "possible" to actually go beyond Z_MIN_POS. So it's useful to not assume the carriage is beyond Z_MIN_POS, but just in Z_MIN_POS. It saves time for homing.

DerAndere1, 2023-04-30:
Currently, Marlin lacks a clear separation between hard limits and soft limits. I think *_MIN_POS is currently seen more as soft limit. To allow for bed leveling, I believe there are two valid configurations: a) have a negative Z_MIN_POS or b) have Z_MIN_POS 0 and disable software endstops before probing. In case a user chooses option b), the print head may be below Z_MIN_POS during probing. Please correct me if I am wrong

closed 2023-07-11 by github-actions[bot]