MarlinFirmware/Marlin #25727: [FR] XY babystepping writes the dual extruder tool offset parameters.

Voxel3DPrinting, 2023-04-21:
### Is your feature request related to a problem? Please describe.

kinda, I'm on a quest to make dual extrusion 3D printing as hassle-free as possible, and I thought it would be really useful if the XY babystepping (when the second tool is enabled) actually overwrites the XY tool offset parameters. Just like the Z babystepping with a probe can do..

Then you could live adjust the offset and save the setting for next time. Now you have to print, measure, adjust, verify, measure, readjust, etcetera

### Describe the feature you want

Basically BABYSTEP_ZPROBE_OFFSET for X and Y features if the machine has multiple extruders.

### Additional context

I am creating a program that cleverly generates a gcode macro, based on the printer and nozzle size, that enables the user to first live-adjust Z, X, and Y during printing, after which it will print 3 very small verifications. 

Basically, 1 semi-assisted print to set and calibrate all offsets.

rondlh, 2023-04-23:
That's a very interesting idea, IDEX live alignment would be quite useful. 
M290 seems to support babystepping for X and Y already, and it considers the active extruder.
 https://marlinfw.org/docs/gcode/M290.html

Voxel3DPrinting, 2023-04-23:
Indeed, only if it would be possible to save the offset! that would be great!

rondlh, 2023-04-23:
> Indeed, only if it would be possible to save the offset! that would be great!

Understood, but the values of M218 (set hotend offset) are saved.
So when you do the live control you need to update the M218 values with your M290 findings, and save the settings, then it should work.
It is done in a similar way with the Z-baby stepping, you need to adjust M851Zxxx with your baby stepping findings (M290) and save the settings.
In my case the Z-baby stepping is supported by a TFT (BTT TFT35 V3.0), so I don't need to do the math and enter the commands manually. The TFT is opensource, so you could add it in there, or do a feature request. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

It seems this is not something for Marlin to handle. Marlin can already do what you want, you just have to issue the correct commands.

Voxel3DPrinting, 2023-04-24:
So workflow would be;

- babystep XY
- note the XY offsets
- recalculate new offsets
- set new offsets

Is that correct?

rondlh, 2023-04-24:
> So workflow would be;
>     * babystep XY
>     * note the XY offsets
>     * recalculate new offsets
>     * set new offsets

> Is that correct?

Yes, M290, M218 then M500

DerAndere1, 2023-04-26:
Would it help with this issue to have the command G10 L10 as known from LinuxCNC implemented in Marlin ? G10 L10 changes the hotend offset for a given hotend so that with the machine in its current position and with the current G5x and G92 offsets active, the current coordinates for the given axes will become the given values. This might eliminate the steop of recalculating new offsets manually.