MarlinFirmware/Marlin #25729: [BUG] Unable to build the latest Bugfix 2.1.x for Anet ET5X/ET4 board

darcyb, 2023-04-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have tried compiling 2 flavours of anet4 board, et4 and et5x. Both return this error in menu_bed_corners.cpp.o. I'm using vscode, platformio, autobuildmarlin. 
```
PS C:\Users\default.DESKTOP-A3KV4JI\Downloads\Marlin-bugfix-2.1.x> echo "done" >"C:\Users\DEFAUL~1.DES\AppData\Local\Temp\ipc"
PS C:\Users\default.DESKTOP-A3KV4JI\Downloads\Marlin-bugfix-2.1.x> platformio run --silent -e Anet_ET4_no_bootloader
In file included from Marlin\src\lcd\menu\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\menu\menu_bed_corners.cpp:27:
Marlin\src\lcd\menu\menu_bed_corners.cpp: In lambda function:
Marlin\src\lcd\menu\menu_bed_corners.cpp:337:58: error: 'leveling_was_active' was not declared in this scope; did you mean 'menu_leveling_was_active'?
  337 |             TERN_(HAS_LEVELING, set_bed_leveling_enabled(leveling_was_active));
      |                                                          ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:627:26: note: in definition of macro 'THIRD'
  627 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\lcd\menu\../../inc/../core/macros.h:247:29: note: in expansion of macro '___TERN'
  247 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:246:29: note: in expansion of macro '__TERN'
  246 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:245:29: note: in expansion of macro '_TERN'
  245 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\lcd\menu\menu_bed_corners.cpp:337:13: note: in expansion of macro 'TERN_'
  337 |             TERN_(HAS_LEVELING, set_bed_leveling_enabled(leveling_was_active));
      |             ^~~~~
Marlin\src\lcd\menu\menu_bed_corners.cpp: In function 'void _lcd_level_bed_corners()':
  358 |     leveling_was_active = planner.leveling_active;
      |     ^~~~~~~~~~~~~~~~~~~
      |     menu_leveling_was_active
*** [.pio\build\Anet_ET4_no_bootloader\src\src\lcd\menu\menu_bed_corners.cpp.o] Error 1
================================================================================================== [FAILED] Took 89.03 seconds ==================================================================================================
----------------------  --------  ------------
Anet_ET4_no_bootloader  FAILED    00:01:29.027
```

### Bug Timeline

new the config compiled in the past

### Expected behavior

compile 

### Actual behavior

Returned error

### Steps to Reproduce

1. Use the latest bugfix 2.1
2. Use a config for an ANET4 mainboard (et4, et5x) from marlinconfiguration
3. Compile with vscode platformio, 

### Version of Marlin Firmware

bugfix2.1

### Printer model

Anet ET5X

### Electronics

stock anet4 board

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11300112/Marlin.zip)

ellensp, 2023-04-22:
Provided config is for a ramps

ellensp, 2023-04-22:
current bugfix 2.1.x with https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Anet/ET4  compiles without issues

ellensp, 2023-04-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-04-22 by ellensp

darcyb, 2023-04-22:
Sorry. I uploaded the wrong config. Thank you, I'll check my work.